awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #241: KVS audio and video

Th0ru5, 2019-07-31:
Hello,
I have problem with sending audio and video to KVS with gstreamer plugin.
Exactly i'm using this command to stream my desktop and microphone to KVS:
gst-launch-1.0 -v ximagonvert ! video/x-raw,framerate=30/1,format=I420 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name=new_straming_system storage-size=512 access-key=xxx secret-key=xxx aws-region=eu-central-1 alsasrc device=hw:0,0 ! audioconvert ! avenc_aac ! queue ! sink.

And i'm geting this warn:
2019-07-31 23:35:00 [139715216074496] WARN - putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x52000085 decoding timestamp: 35294331 presentation timestamp: 35294331

I was trying to find what 0x52000085 status means, but i can't find any information in AWS Documentation.

When i try the same command but without the audio part all is working fine.
The command without audio part:

gst-launch-1.0 -v ximagonvert ! video/x-raw,framerate=30/1,format=I420 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name=new_straming_system storage-size=512 access-key=xxx secret-key=xxx aws-region=eu-central-1

Can you help me with it?

unicornss, 2019-07-31:
Hi @Th0ru5

It seems like the _number of audio frames_ to order with the respective video frames are exceeding the default size of 60 Ref:  [#define DEFAULT_MAX_FRAME_QUEUE_SIZE_PER_TRACK              60](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/75087f5a90a02a47191c9278cfec329e09535e98/kinesis-video-pic/src/mkvgen/include/com/amazonaws/kinesis/video/mkvgen/Include.h#L134).

 Could you please try updating this to higher value e.g. 240, then compile & run to see if that fixes your issue? 

The status codes 0x52000082 to [0x52000085](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/424580b800087f05f80000d30b43b61e5a964c29/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L186) are not updated in the [Reference](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html). We will fix that next.

Th0ru5, 2019-08-03:
I updated this to 240, and nothing happened, next i set it to 480, and still nothing.
Maybe i did something wrong.

After updating the value I ran ./min-install-script next I ran ./gstreamer-plugin-install-script

Did I did it correctly?

chehefen, 2019-08-05:
you just need to run ./min-install-script. Can you do `export DEBUG_DUMP_FRAME_INFO=TRUE` first and then execute the pipeline and send us the log? Thanks

Th0ru5, 2019-08-06:
There are the logs.

> gst-launch-1.0 -v ximagesrc startx=1280 use-damage=0 ! videoconvert ! video/x-raw,framerate=30/1,format=I420 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name=new_straming_system storage-size=512 access-key=AKIA6Q7H7MK5CQT3RD2X secret-key=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 aws-region=eu-central-1 alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.
Ustawianie potoku na PAUZOWANIE...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2019-08-06 22:43:44 [139960720545600] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2019-08-06 22:43:44 [139960720545600] INFO - heapInitialize(): Creating AIV heap.
2019-08-06 22:43:44 [139960720545600] INFO - heapInitialize(): Heap is initialized OK
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
2019-08-06 22:43:44 [139960720545600] DEBUG - getSecurityTokenHandler invoked
2019-08-06 22:43:44 [139960720545600] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1565124224899314365 Expiration: 0
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
2019-08-06 22:43:44 [139960720545600] INFO - createDeviceResultEvent(): Create device result event.
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
2019-08-06 22:43:44 [139960720545600] DEBUG - clientReadyHandler invoked
2019-08-06 22:43:44 [139960720545600] INFO - try creating stream
2019-08-06 22:43:44 [139960720545600] INFO - Creating Kinesis Video Stream new_straming_system
2019-08-06 22:43:44 [139960720545600] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-06 22:43:44 [139960720545600] INFO - describeStreamResultEvent(): Describe stream result event.
2019-08-06 22:43:45 [139959651743488] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
2019-08-06 22:43:45 [139959651743488] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-06 22:43:45 [139959651743488] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2019-08-06 22:43:45 [139959643350784] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
2019-08-06 22:43:45 [139959643350784] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-06 22:43:45 [139959643350784] DEBUG - getStreamingTokenHandler invoked
2019-08-06 22:43:45 [139959643350784] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1565124225701549510 Expiration: 18446744073709551615
2019-08-06 22:43:45 [139959643350784] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2019-08-06 22:43:45 [139959643350784] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
2019-08-06 22:43:45 [139959643350784] DEBUG - defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94784104142473
2019-08-06 22:43:45 [139959643350784] DEBUG - streamReadyHandler invoked
2019-08-06 22:43:45 [139959643350784] Stream is ready
Potok jest na żywo i nie wymaga PRZEWINIĘCIA...
/GstPipeline:pipeline0/GstXImageSrc:ximagesrc0.GstPad:src: caps = video/x-raw, format=(string)BGRx, width=(int)2000, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1
Ustawianie potoku na ODTWARZANIE...
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)2000, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, format=(string)I420
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, width=(int)2000, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, format=(string)I420
New clock: GstAudioSrcClock
/GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: actual-buffer-time = 200000
/GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: actual-latency-time = 10000
Rozprzestrzenianie opóźnień...
/GstPipeline:pipeline0/GstAlsaSrc:alsasrc0.GstPad:src: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:src: caps = audio/x-raw, rate=(int)44100, channels=(int)2, layout=(string)interleaved, format=(string)F32LE, channel-mask=(bitmask)0x0000000000000003
/GstPipeline:pipeline0/avenc_aac:avenc_aac0.GstPad:sink: caps = audio/x-raw, rate=(int)44100, channels=(int)2, layout=(string)interleaved, format=(string)F32LE, channel-mask=(bitmask)0x0000000000000003
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:sink: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
[... cut ...]
Przerwanie: Zatrzymywanie potoku...
Execution ended after 0:00:14.375877229
Ustawianie potoku na PAUZOWANIE...
Ustawianie potoku na GOTOWOŚĆ...
Ustawianie potoku na NULL...
Zwalnianie potoku...
INFO - Freeing Kinesis Video Stream new_straming_system
2019-08-06 22:44:00 [139960720545600] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2019-08-06 22:44:00 [139960720545600] DEBUG - defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 94784104142473
2019-08-06 22:44:00 [139960720545600]

Th0ru5, 2019-08-06:
There are the logs.

> gst-launch-1.0 -v ximagesrc startx=1280 use-damage=0 ! videoconvert ! video/x-raw,framerate=30/1,format=I420 ! x264enc  bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name=new_straming_system storage-size=512 access-key=XXX secret-key=XXX aws-region=eu-central-1 alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.
Ustawianie potoku na PAUZOWANIE...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2019-08-06 22:43:44 [139960720545600] INFO - heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
2019-08-06 22:43:44 [139960720545600] INFO - heapInitialize(): Creating AIV heap.
2019-08-06 22:43:44 [139960720545600] INFO - heapInitialize(): Heap is initialized OK
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
2019-08-06 22:43:44 [139960720545600] DEBUG - getSecurityTokenHandler invoked
2019-08-06 22:43:44 [139960720545600] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1565124224899314365 Expiration: 0
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
2019-08-06 22:43:44 [139960720545600] INFO - createDeviceResultEvent(): Create device result event.
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
2019-08-06 22:43:44 [139960720545600] DEBUG - clientReadyHandler invoked
2019-08-06 22:43:44 [139960720545600] INFO - try creating stream
2019-08-06 22:43:44 [139960720545600] INFO - Creating Kinesis Video Stream new_straming_system
2019-08-06 22:43:44 [139960720545600] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
2019-08-06 22:43:44 [139960720545600] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-06 22:43:44 [139960720545600] INFO - describeStreamResultEvent(): Describe stream result event.
2019-08-06 22:43:45 [139959651743488] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
2019-08-06 22:43:45 [139959651743488] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-06 22:43:45 [139959651743488] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2019-08-06 22:43:45 [139959643350784] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
2019-08-06 22:43:45 [139959643350784] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-06 22:43:45 [139959643350784] DEBUG - getStreamingTokenHandler invoked
2019-08-06 22:43:45 [139959643350784] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1565124225701549510 Expiration: 18446744073709551615
2019-08-06 22:43:45 [139959643350784] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2019-08-06 22:43:45 [139959643350784] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
2019-08-06 22:43:45 [139959643350784] DEBUG - defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94784104142473
2019-08-06 22:43:45 [139959643350784] DEBUG - streamReadyHandler invoked
2019-08-06 22:43:45 [139959643350784] Stream is ready
Potok jest na żywo i nie wymaga PRZEWINIĘCIA...
/GstPipeline:pipeline0/GstXImageSrc:ximagesrc0.GstPad:src: caps = video/x-raw, format=(string)BGRx, width=(int)2000, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1
Ustawianie potoku na ODTWARZANIE...
/GstPipeline:pipeline0/GstVideoConvert:videoconvert0.GstPad:src: caps = video/x-raw, width=(int)2000, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, format=(string)I420
/GstPipeline:pipeline0/GstCapsFilter:capsfilter0.GstPad:src: caps = video/x-raw, width=(int)2000, height=(int)1080, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1, format=(string)I420
New clock: GstAudioSrcClock
/GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: actual-buffer-time = 200000
/GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: actual-latency-time = 10000
Rozprzestrzenianie opóźnień...
/GstPipeline:pipeline0/GstAlsaSrc:alsasrc0.GstPad:src: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:src: caps = audio/x-raw, rate=(int)44100, channels=(int)2, layout=(string)interleaved, format=(string)F32LE, channel-mask=(bitmask)0x0000000000000003
/GstPipeline:pipeline0/avenc_aac:avenc_aac0.GstPad:sink: caps = audio/x-raw, rate=(int)44100, channels=(int)2, layout=(string)interleaved, format=(string)F32LE, channel-mask=(bitmask)0x0000000000000003
/GstPipeline:pipeline0/GstAudioConvert:audioconvert0.GstPad:sink: caps = audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)44100, channels=(int)2, channel-mask=(bitmask)0x0000000000000003
Rozprzestrzenianie opóźnień...
[... cut ...]
Przerwanie: Zatrzymywanie potoku...
Execution ended after 0:00:14.375877229
Ustawianie potoku na PAUZOWANIE...
Ustawianie potoku na GOTOWOŚĆ...
Ustawianie potoku na NULL...
Zwalnianie potoku...
INFO - Freeing Kinesis Video Stream new_straming_system
2019-08-06 22:44:00 [139960720545600] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2019-08-06 22:44:00 [139960720545600] DEBUG - defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 94784104142473
2019-08-06 22:44:00 [139960720545600]

MushMal, 2019-08-06:
@Th0ru5 it seems that your pipeline produces just audio frames only in the snippet of logs attached. I couldn't find any information on the ximagonvert element so I assume this is a custom element or something. Could you please look into the frame rate perhaps? You need to ensure there is no large "drift" between audio and video.

Th0ru5, 2019-08-06:
@MushMal 
God, I copied bad commands to first post.
This is the command I'm using:

gst-launch-1.0 -v ximagesrc startx=1280 use-damage=0 ! videoconvert ! video/x-raw,framerate=30/1,format=I420 ! x264enc bframes=0 key-int-max=45 bitrate=500 tune=zerolatency ! h264parse ! video/x-h264,stream-format=avc,alignment=au,profile=baseline ! kvssink name=sink stream-name=new_straming_system storage-size=512 access-key=XXX secret-key=XXX aws-region=eu-central-1 alsasrc device=hw:1,0 ! audioconvert ! avenc_aac ! queue ! sink.

I will edit the first post in the issue and i will paste the commands I'm using.
So sorry for this mistake.

The logs are from running the command above.

MushMal, 2019-08-08:
@Th0ru5 sorry for the delay. Not yet sure why the source doesn't produce the frames or whether it's the encoder that doesn't produce the encoded frames.

MushMal, 2019-08-08:
One thing that might help - would you please use fakesource or an actual video source instead of ximagesrc to eliminate the encoders?

unicornss, 2019-08-08:
Hi @Th0ru5 

Can you please confirm if you are able to see the video with the following pipeline?

`gst-launch-1.0 ximagesrc ! videoconvert ! autovideosink` 
and
`gst-launch-1.0 ximagesrc use-damage=0  ! videoconvert ! videoscale ! video/x-raw,width=1280,height=720 ! autovideosink`

Thanks

chehefen, 2019-08-08:
ximagesrc produces frames very slowly even if you configure framerate=30/1 in the caps filter. It seems like if you limit the area that ximagesrc captures by doing `gst-launch-1.0 ximagesrc use-damage=0 endx=640 endy=480` then it produces frames slightly faster. You can use these two pipeline to get a sense of the rate at which audio and video frames are produced 
```
gst-launch-1.0 -v ximagesrc use-damage=FALSE endx=640 endy=480 ! video/x-raw,framerate=30/1 ! queue ! fakesink silent=FALSE
gst-launch-1.0 -v alsasrc device=hw:1,0 ! fakesink silent=FALSE
```
On my device, in default setting ximagesrc produces a frame every 10s, while for audio hundreds of frames are produced. Kvssink is not expecting such big gap between audio and video. You need to tune your pipeline to increase the video framerate as well as reduce the audio framerate to reduce the gap between them.

Th0ru5, 2019-08-08:
@MushMal 
I tried to use videotestsrc and audiotestsrc and i have still the same WARN's.
@unicornss 
Yes i'm able to see the video.
@chehefen 
I'm trying to do what You said but i can not make it working.
Can you give me some example that is working on your device ?
I will try it on my PC. Maybe is something wrong with my PC, maybe bad drivers or something.
I'm on Ubuntu 18.04.

The weirdest thing is that.
With using the same sources, encoders, decoders, etc. I can stream to YouTube with RTMP and to MP4 file without problem.

chehefen, 2019-08-14:
can you show us what your pipeline is like for RTMP streaming and MP4 file generation so that we can try it in our own device?

shubb, 2019-08-17:
I am also struggling with this in a different context (https://stackoverflow.com/questions/57532541/how-to-merge-audio-and-video-from-different-sources-for-kinesis-video). 

Is there an easy way to force video and audio into sync in gstreamer?

MushMal, 2019-08-19:
@shubb the stackoverflow suggestion is correct in terms that the sink has 60 frame buffer which should be enough to buffer the "jitter" between the tracks in real-time streaming mode. If you have two distinct sources then you might need to adjust the timestamps based on the start timestamp and ensure the audio and video frames are produced at the correct cadence.

chehefen, 2019-09-16:
@Th0ru5 can you try apply this patch in the root directory of sdk?
[patch.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3618197/patch.txt)

Th0ru5, 2019-10-01:
@chehefen Ok i did what u asked me, and it helped me with XIMAGESRC.
But now i'm trying to send video and audio from IP camera to KVS.
Before the patch i had the same issue as with XIMAGESRC.

Without audio  all is working perfectly, but when i try add audio to stream I'm getting something like this:
`[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 37884632 presentation timestamp: 37884632
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 39591084 presentation timestamp: 39591084
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 39804418 presentation timestamp: 39804418
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 39824184 presentation timestamp: 40624116
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40017751 presentation timestamp: 40017751
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40224150 presentation timestamp: 42223960
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40231084 presentation timestamp: 40231084
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40657624 presentation timestamp: 40657624
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42364169 presentation timestamp: 42364169
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42577385 presentation timestamp: 42577385
[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42623919 presentation timestamp: 42623919
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42790718 presentation timestamp: 42790718
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 43857272 presentation timestamp: 43857272
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44070606 presentation timestamp: 44070606
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44223761 presentation timestamp: 45023684
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44283939 presentation timestamp: 44283939
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44497272 presentation timestamp: 44497272
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44710497 presentation timestamp: 44710497
[WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 45137164 presentation timestamp: 45137164
[DEBUG][2019-09-30 23:45:00] postReadCallback(): Wrote 9270 bytes to Kinesis Video. Upload stream handle: 0
[DEBUG][2019-09-30 23:45:00] postReadCallback(): Pausing CURL read for upload handle: 0`

In Your documentation on https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html is nothing about audio.

Now I'm not using commands, but i'm trying to create Java application with https://github.com/gstreamer-java/gst1-java-core.

The pipeline for KVS looks like this:
`appsrc name=videoAppSrc "+
				"! rawvideoparse name=videoparser use-sink-caps=false format=8 "+
				"! videoconvert ! video/x-raw,format=I420 "+
				"! x264enc ! h264parse ! video/x-h264,stream-format=avc,alignment=au "+
				"! kvssink name=kvs stream-name=new_straming_system storage-size=512 access-key=xxx secret-key=xxx aws-region=eu-central-1 "+
				"appsrc name=audioAppSrc "+
				"! rawaudioparse use-sink-caps=true "+
				"! audioconvert ! avenc_aac ! kvs.`
Caps for audio looks like this
`audio/x-raw, format=(string)F32LE, layout=(string)interleaved, rate=(int)48000, channels=(int)1`

I was trying to implement simple function to check if DTS(n) < DTS(n-1) + Duration(n-1) and i was eliminating those audio samples but this didn't help.

When I'm trying to send the stream with video and audio to YouTube from the same application but using different pipeline all is working.
I was trying to deal with it alone but after 6 hours i give up.

chehefen, 2019-10-01:
Can you close this issue since it has been fixed and open up a new issue? Thanks

Th0ru5, 2019-10-01:
Yes i can close it.
I will open new one right now.

closed 2019-10-01 by Th0ru5