awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #242: windows native build x64 compilation error

ioxFR, 2019-08-02:
Hello,

I trying to to a ./windows-install.bat 64 as described in the documentation.
All stuff are installed but when the script start to compile differents files i get an error as follow:

>  "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\pic_test.vcxproj" (cible par défaut) (1) ->
       (ClCompile cible) ->
         c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: indice négatif [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-vid
       eo-native-build\pic_test.vcxproj]
         c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: indice négatif [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-vid
       eo-native-build\pic_test.vcxproj]
         c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: indice négatif [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-vid
       eo-native-build\pic_test.vcxproj]
         c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: indice négatif [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-vid
       eo-native-build\pic_test.vcxproj]
         c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: indice négatif [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-vid
       eo-native-build\pic_test.vcxproj]

"Negative index error"

I tried multiple time to clean the directory and launch again but i still get the same issue. What i am doing wrong?
Someone already got the same issue?

Thanks for help

zhiyua-git, 2019-08-02:
Hi @ioxFR ,

From your compilation error message, it seems you are installing 64 bit but the your path was pointing to 32 bit library (c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487)). You will need to make sure there is no 32 bit library path mixed in when you are building 64 bit application.

From your latest update, seems you are able to get it compiled. Please open a new issue if you run into any.

Thanks,
-ZH

closed 2019-08-02 by zhiyua-git

Ulyssez, 2019-09-25:
I have removed 18362, still build fails with errors below ... Could you please advise what am I doing wrong? 

         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(320): error C2371: 'PSSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk
       -cpp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(319): error C2371: 'SIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-c
       pp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(319): error C2371: 'PSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
       cpp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(320): error C2371: 'SSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
       cpp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(320): error C2371: 'PSSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk
       -cpp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(319): error C2371: 'SIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-c
       pp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(319): error C2371: 'PSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
       cpp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(320): error C2371: 'SSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
       cpp\kinesis-video-native-build\pic_test.vcxproj]
         c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
       ommondefs.h(320): error C2371: 'PSSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk
       -cpp\kinesis-video-native-build\pic_test.vcxproj]

    249 Warning(s)
    176 Error(s)

Time Elapsed 00:00:48.24
Build error occurred! Please delete related directories in downloads\ to start a clean build.