awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #243: install-script not working in macOS

MizukiSonoko, 2019-08-05:
Hello, I try to install SDK on macOS, but I get an error about building libcurl.

## Environment
```
OS: Mojave 10.14.5
cmake version 3.14.4
autoconf (GNU Autoconf) 2.69
bison (GNU Bison) 2.3
```

## Precondition
I ran 
```
brew install pkg-config openssl cmake gstreamer gst-plugins-base gst-plugins-good gst-plugins-bad gst-plugins-ugly log4cplus gst-libav
```

## Error 
I ran
```
cd kinesis-video-native-build;
./install-script
```
I got
```
configure: Configured to build curl/libcurl:

  curl version:     7.57.0
  Host setup:       x86_64-apple-darwin18.6.0
  Install prefix:   /Users/mizuki/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local
  Compiler:         gcc
  SSL support:      enabled (OpenSSL)
  SSH support:      no      (--with-libssh2)
  zlib support:     no      (--with-zlib)
  brotli support:   enabled (libbrotlidec)
  GSS-API support:  no      (--with-gssapi)
  TLS-SRP support:  enabled
  resolver:         POSIX threaded
  IPv6 support:     enabled
  Unix sockets support: enabled
  IDN support:      enabled (libidn2)
  Build libcurl:    Shared=yes, Static=yes
  Built-in manual:  enabled
  --libcurl option: enabled (--disable-libcurl-option)
  Verbose errors:   enabled (--disable-verbose)
  SSPI support:     no      (--enable-sspi)
  ca cert bundle:   /Users/mizuki/workspace/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/certs/cert.pem
  ca cert path:     no
  ca fallback:      no
  LDAP support:     no      (--enable-ldap / --with-ldap-lib / --with-lber-lib)
  LDAPS support:    no      (--enable-ldaps)
  RTSP support:     no      (--enable-rtsp)
  RTMP support:     enabled (librtmp)
  metalink support: no      (--with-libmetalink)
  PSL support:      yes
  HTTP2 support:    disabled (--with-nghttp2)
  Protocols:        DICT FILE FTP FTPS GOPHER HTTP HTTPS IMAP IMAPS POP3 POP3S RTMP SMB SMBS SMTP SMTPS TELNET TFTP

Making all in lib
/Applications/Xcode.app/Contents/Developer/usr/bin/make  all-am
  CC       libcurl_la-file.lo
  CC       libcurl_la-timeval.lo
  CC       libcurl_la-base64.lo
  CC       libcurl_la-hostip.lo
  CC       libcurl_la-progress.lo
  CC       libcurl_la-formdata.lo
  CC       libcurl_la-cookie.lo
  CC       libcurl_la-http.lo
  CC       libcurl_la-sendf.lo
  CC       libcurl_la-ftp.lo
  CC       libcurl_la-url.lo
  CC       libcurl_la-dict.lo
  CC       libcurl_la-if2ip.lo
  CC       libcurl_la-speedcheck.lo
  CC       libcurl_la-ldap.lo
  CC       libcurl_la-version.lo
  CC       libcurl_la-getenv.lo
  CC       libcurl_la-escape.lo
  CC       libcurl_la-mprintf.lo
  CC       libcurl_la-telnet.lo
  CC       libcurl_la-netrc.lo
  CC       libcurl_la-getinfo.lo
  CC       libcurl_la-transfer.lo
  CC       libcurl_la-strcase.lo
  CC       libcurl_la-easy.lo
  CC       libcurl_la-security.lo
  CC       libcurl_la-curl_fnmatch.lo
  CC       libcurl_la-fileinfo.lo
  CC       libcurl_la-ftplistparser.lo
  CC       libcurl_la-wildcard.lo
  CC       libcurl_la-krb5.lo
  CC       libcurl_la-memdebug.lo
  CC       libcurl_la-http_chunks.lo
  CC       libcurl_la-strtok.lo
  CC       libcurl_la-connect.lo
  CC       libcurl_la-llist.lo
  CC       libcurl_la-hash.lo
  CC       libcurl_la-multi.lo
  CC       libcurl_la-content_encoding.lo
content_encoding.c:549:8: error: use of undeclared identifier 'BROTLI_DECODER_ERROR_COMPOUND_DICTIONARY'; did you mean 'BROTLI_DECODER_ERROR_FORMAT_DICTIONARY'?
  case BROTLI_DECODER_ERROR_COMPOUND_DICTIONARY:
       ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
       BROTLI_DECODER_ERROR_FORMAT_DICTIONARY
/usr/local/include/brotli/decode.h:116:35: note: 'BROTLI_DECODER_ERROR_FORMAT_DICTIONARY' declared here
  BROTLI_DECODER_ERROR_CODES_LIST(BROTLI_ERROR_CODE_ENUM_ITEM_, BROTLI_COMMA_)
                                  ^
content_encoding.c:549:8: error: duplicate case value 'BROTLI_DECODER_ERROR_FORMAT_DICTIONARY'
  case BROTLI_DECODER_ERROR_COMPOUND_DICTIONARY:
       ^
content_encoding.c:545:8: note: previous case defined here
  case BROTLI_DECODER_ERROR_FORMAT_DICTIONARY:
       ^
2 errors generated.
make[2]: *** [libcurl_la-content_encoding.lo] Error 1
make[2]: *** Waiting for unfinished jobs....
make[1]: *** [all] Error 2
make: *** [all-recursive] Error 1
```

So what I should do?

MizukiSonoko, 2019-08-05:
I update curl version `7.57.0` -> `7.58.0` 
and it works
```
$ git diff 
diff --git a/kinesis-video-native-build/install-script b/kinesis-video-native-build/install-script
index 79cb7bf..5bfc61c 100755
--- a/kinesis-video-native-build/install-script
+++ b/kinesis-video-native-build/install-script
@@ -121,11 +121,11 @@ if [[ $build_producer == TRUE ]]; then
     echo "Curl lib not found. Installing"
     if [ ! -f $DOWNLOADS/curl-7.57.0.tar.xz ]; then
       cd $DOWNLOADS
-      $SYS_CURL -L "https://curl.haxx.se/download/curl-7.57.0.tar.xz" -o "curl-7.57.0.tar.xz"
+      $SYS_CURL -L "https://curl.haxx.se/download/curl-7.58.0.tar.xz" -o "curl-7.58.0.tar.xz"
     fi
     cd $DOWNLOADS
-    tar -xvf curl-7.57.0.tar.xz
-    cd $DOWNLOADS/curl-7.57.0
+    tar -xvf curl-7.58.0.tar.xz
+    cd $DOWNLOADS/curl-7.58.0

     if [ -f "$CERT_PEM_FILE" ] ;then
       ./configure --prefix=$DOWNLOADS/local/ --enable-dynamic --disable-rtsp --disable-ldap --without-zlib --with-ssl=$DOWNLOADS/local/ --with-ca-bundle=$CERT_PEM_FILE
```

unicornss, 2019-08-05:
Hi @MizukiSonoko 

Thanks for checking the latest curl for Mojave. Could you please update the issue description specific to the Mojave.

Currently, the SDK has been tested in the  following OS versions.
[Kinesis Video Streams Producer SDK](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp) was tested on following platforms:

    Ubuntu 16
    Ubuntu 17
    Ubuntu 18
    Raspberry Raspbian
    MacOS El Captian
    MacOS Sierra
    MacOS High Sierra
    Windows 7 Enterprise
    Windows 10 Enterprise
    Windows Server 2012 R2
    Windows Server 2016

Thanks again.

chehefen, 2019-08-05:
btw you dont need to run ./install-script since you have installed all the dependencies using homebrew. You can use ./min-install-script instead to save time.

unicornss, 2019-08-08:
As @chehefen  mentioned run `./min-install-script ` if dependencies are already installed using `brew install` .  

However, just verified that `./install-script` works in Mojave 10.14.6 without any curl install issue for `7.57`.  Updated the issue description as **Mojave** specific in case someone find similar issue.

Closing the issue for now. Please reopen or create a new one if you have further questions.

closed 2019-08-08 by unicornss