awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #244: How to config buffer size in offline mode?

jiangxiaorong, 2019-08-06:
Hi:
   As my understand, in offline mode, if network traffic is bad, kvs sdk will save the frames in buffer, can you tell me how to config this buffer's size?

chehefen, 2019-08-06:
if you are using the c sdk, it's configured through pDeviceInfo->storageInfo.storageSize [here](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/75087f5a90a02a47191c9278cfec329e09535e98/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c#L90). if you are using the cpp sdk, you can refer to this example: [cpp_sample](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/75087f5a90a02a47191c9278cfec329e09535e98/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L215)

chehefen, 2019-08-16:
customer question has already been answered. They can open another issue if they have more question.

closed 2019-08-16 by chehefen