awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #245: How to sync up audio and video

jiangxiaorong, 2019-08-06:
Dear:
I have put audio and video frames to kvs server?  I have put audio and video to kvs server but when I get them from kvs server, I find video and audio frames are not sync.
Attach download file from kvs server.
[kvs_getvideo.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3471899/kvs_getvideo.zip)

chehefen, 2019-08-06:
Can you tell us how did you put audio and video frames to kvs server?

jiangxiaorong, 2019-08-07:
when create stream I call addTrack to add an audio track, and used putFrame api to put audio/video frames, if the frame is audio set frame's track id to audio track, if the frame is video frame, set the track is to video track.

jiangxiaorong, 2019-08-07:
@chehefen :Today, I used class PutFrameHelper to put audio and video frames to kvs server, I met below issuse:
1. when play the stream on kvs server, audio and video are still not sync;
2. I find the current frame rate are not normally, sometimes it is very large, can reach above 1000+, like below, attach kvs_test_20190807.log, you can get detail test log
    [18:00:53]        >> Current frame rate (fps): 1899.56
    [18:00:53]        >> Current transfer rate (bps): 33554432 (32768 Kbps)
[kvs_test_20190807.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3478038/kvs_test_20190807.log)

chehefen, 2019-08-07:
can you do `export KVS_DEBUG_DUMP_DATA_FILE_DIR=/path/to/directory` and `export DEBUG_DUMP_FRAME_INFO=TRUE` and run the application again and send us the log? Exporting KVS_DEBUG_DUMP_DATA_FILE_DIR will store the mkv file in the directory. Can you check if you still observe audio video out of sync?

jiangxiaorong, 2019-08-09:
@chehefen below is dump data and test log:
[dump_frames.zip](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3484319/dump_frames.zip)
[kvs_debug.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3484320/kvs_debug.log)

chehefen, 2019-08-12:
I found this in the log:
```
2019-08-08 11:08:52 [1074373440] DEBUG - Putting frame for stream: camera-core-device-3960 TID: 0x4009a340, trackId: 1, Id: 3068, Key Frame: true, Size: 11954, Dts: 15652625223071785, Pts: 15652625223071785, Frame duration: 0, Total duration:8204
2019-08-08 11:08:52 [1074373440] DEBUG - Putting frame for stream: camera-core-device-3960 TID: 0x4009a340, trackId: 2, Id: 1133, Key Frame: false, Size: 253, Dts: 15652625196041142, Pts: 15652625196041142, Frame duration: 0, Total duration:8204
2019-08-08 11:08:52 [1074373440] DEBUG - Putting frame for stream: camera-core-device-3960 TID: 0x4009a340, trackId: 1, Id: 3206, Key Frame: true, Size: 6233, Dts: 15652625254631785, Pts: 15652625254631785, Frame duration: 0, Total duration:11360
```

do you know why you had two video key frames in such short duration?

jiangxiaorong, 2019-08-13:
I set evey I frame as a key frame, fps of my stream is 15, gov is 15

Xiaoron, 2019-08-15:
@chehefen it's my streaming server error, it jump some frames at the beginning.

chehefen, 2019-08-15:
﻿﻿﻿I sorted the frames by their pts in this file: 
[sortedFrame.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3507137/sortedFrame.txt). As you can see, between pts 15652625203051785 and 15652625223071785 there is no audio frames. The producer is expecting each fragment to contain both audio and video frames and would through error if fragment has only one type of frame. Can you try to tune your media pipeline to produce more evenly distributed frames? Also, are you doing realtime streaming or streaming clips?

jiangxiaorong, 2019-08-19:
@chehefen Thanks for your help.
Now audio and video can sync up with each other. I use kvs producer to upload real time streaming

unicornss, 2019-10-31:
Closing the issue for now as the question has been answered. Thanks.

closed 2019-10-31 by unicornss