awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #246: Install Script Issue - raspberry pi

Alex-fanfanfan, 2019-08-06:
[image]
I typed ./min-install-script and got this.

Another issue is that I can't run ./install-script because it said this is going to be more than ten hours and only allowed me to type ./min-install-script.

oily126, 2019-08-06:
Hi @Alex-fanfanfan , 

Have you installed the dependencies before you run min-install-script?

$ sudo apt-get update
$ sudo apt-get install libssl-dev libcurl4-openssl-dev **liblog4cplus-1.1-9 liblog4cplus-dev**

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-steps-for-ubuntu-17x-and-raspbian-stretch-using-apt-get

zhiyua-git, 2019-08-06:
Hi @Alex-fanfanfan ,

Have you installed the dependencies before you run min-install-script?

$ sudo apt-get update
$ sudo apt-get install libssl-dev libcurl4-openssl-dev **liblog4cplus-1.1-9 liblog4cplus-dev**

https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-steps-for-ubuntu-17x-and-raspbian-stretch-using-apt-get

Alex-fanfanfan, 2019-08-06:
`pi@raspberrypi:~/Downloads/log4cplus-1.2.1 $ sudo apt-get install libssl-dev libcurl4-openssl-dev liblog4cplus-1.1-9 liblog4cplus-dev

Reading package lists... Done
Building dependency tree       
Reading state information... Done
E: Unable to locate package liblog4cplus-1.1-9
E: Couldn't find any package by regex 'liblog4cplus-1.1-9'
`

It showed this. Is that because I did not download the package?

chehefen, 2019-08-06:
did you run sudo apt-get update?

Alex-fanfanfan, 2019-08-06:
I did.

chehefen, 2019-08-06:
can you tell us about what's your raspberry pi model and the os version?

Alex-fanfanfan, 2019-08-06:
Raspberry Pi 3 Model B
OS version 8

chehefen, 2019-08-06:
you can try do `sudo apt-get install liblog4cplus-dev` and run `min-install-script` to see if the producer builds. If not you have to upgrade to OS version 9. Dont forget to install all gstreamer dependencies too.

chehefen, 2019-08-07:
customer managed to build the plugin. closing the issue.

closed 2019-08-07 by chehefen

str3tch, 2019-08-29:
I have the exact same issue. What was the resolution? I have done `sudo apt-get install liblog4cplus-dev` but I still get the same error:

```
/home/pi/git/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.cpp: In static member function ‘static void com::amazonaws::kinesis::video::DefaultCallbackProvider::logPrintHandler(UINT32, PCHAR, PCHAR, ...)’:
/home/pi/git/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.cpp:308:40: error: ‘LOG4CPLUS_SUPPRESS_DOWHILE_WARNING’ was not declared in this scope
     LOG4CPLUS_SUPPRESS_DOWHILE_WARNING()
                                        ^
/home/pi/git/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-producer/src/DefaultCallbackProvider.cpp:321:39: error: ‘LOG4CPLUS_RESTORE_DOWHILE_WARNING’ was not declared in this scope
     LOG4CPLUS_RESTORE_DOWHILE_WARNING()
```

fideliocc, 2019-09-18:
Please, can you share the solution? I'm having the same issue here and isn't related to the OS version, I tested it with Raspbian Stretch and Buster (the last OS release). Worth to mention that a few months ago I was able to perform build with `./install-script` (yes, it takes a lot of time to perform). But on this occasion, I find this new message suggesting to use `./min-install-script` instead. 

When I try to run the gstreamer_sample_app after this suggested installation, the stream is created but stopped with the error: 

`log4cplus:ERROR Failed to open file ./log/kvs.log`

I also found that this installation doesn't create the `/bin` directory at `amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local`

Help is appreciated.