awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #247: Code running but no data streams

Alex-fanfanfan, 2019-08-07:
I can run this successfully.
`gst-launch-1.0 -v avfvideosrc device-index=1 ! h264parse ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.
`
But these is nothing showed up on the aws console video streams preview.

MushMal, 2019-08-07:
@Alex-fanfanfan by default, the streaming will use us-west-2 region. Can you make sure you are in us-west-2?

If that's not the case, could you please capture the logs so we can see whether there is a networking issue or some other sort of issue.

Alex-fanfanfan, 2019-08-07:
I used us-east-1 on this streaming and the same on the aws configure, is this correct?

This is the log.

`2019-08-07 11:48:33 [0x70000943c000] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Wrote 1737 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: YourStreamName and upload handle: 0 returned: {"EventType":"RECEIVED","FragmentTimecode":26970,"FragmentNumber":"91343852333181442385334059733654961226170394198"}

2019-08-07 11:48:34 [0x70000974e000] DEBUG - fragmentAckReceivedHandler invoked
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Wrote 4246 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: YourStreamName and upload handle: 0 returned: {"EventType":"BUFFERING","FragmentTimecode":28467,"FragmentNumber":"91343852333181442390285819890796482644972776312"}

2019-08-07 11:48:34 [0x70000974e000] DEBUG - fragmentAckReceivedHandler invoked
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postWriteCallback(): Curl post body write function for stream with handle: YourStreamName and upload handle: 0 returned: {"EventType":"PERSISTED","FragmentTimecode":26970,"FragmentNumber":"91343852333181442385334059733654961226170394198"}

2019-08-07 11:48:34 [0x70000974e000] DEBUG - fragmentAckReceivedHandler invoked
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Wrote 7255 bytes to Kinesis Video. Upload stream handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - postReadCallback(): Pausing CURL read for upload handle: 0
2019-08-07 11:48:34 [0x70000974e000] DEBUG - Kinesis Video client and stream metrics
	>> Overall storage byte size: 536870912
	>> Available storage byte size: 536765699
	>> Allocated storage byte size: 105213
	>> Total view allocation byte size: 144072
	>> Total streams frame rate (fps): 30
	>> Total streams transfer rate (bps): 5260616 (5137 Kbps)
	>> Current view duration (ms): 864
	>> Overall view duration (ms): 1496
	>> Current view byte size: 57845
	>> Overall view byte size: 102918
	>> Current frame rate (fps): 30.477
	>> Current transfer rate (bps): 5260616 (5137 Kbps)`

zhiyua-git, 2019-08-07:
To use it in us-east-1, you need to add aws-region=us-east-1 to your gst-launch  command.

gst-launch-1.0 -v avfvideosrc device-index=1 ! h264parse ! kvssink name=sink stream-name="my_stream_name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey" **aws-region=us-east-1** osxaudiosrc ! audioconvert ! avenc_aac ! queue ! sink.

Alex-fanfanfan, 2019-08-07:
Still not working.

Setting pipeline to PAUSED ...
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - heapInitialize(): Creating AIV heap.
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - heapInitialize(): Heap is initialized OK
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - getSecurityTokenHandler invoked
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1565204308001464 Expiration: 0
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - createDeviceResultEvent(): Create device result event.
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - clientReadyHandler invoked
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - try creating stream
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - Creating Kinesis Video Stream YourStreamName
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
2019-08-07 11:58:28 [0x7fffa9c7b380] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-07 11:58:28 [0x7fffa9c7b380] INFO - writeHeaderCallback(): RequestId: 1db6d756-976c-40cd-9bb1-9491a934509b
2019-08-07 11:58:28 [0x70000de4d000] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.565202306662E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-1:746510367092:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:746510367092:stream/YourStreamName/1565202306662","StreamName":"YourStreamName","Version":"gO0jlcpDHPt4VZKhl9we"}}
2019-08-07 11:58:28 [0x70000de4d000] INFO - describeStreamResultEvent(): Describe stream result event.
2019-08-07 11:58:28 [0x70000de4d000] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2019-08-07 11:58:28 [0x70000de4d000] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
2019-08-07 11:58:28 [0x70000de4d000] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-07 11:58:28 [0x70000de4d000] INFO - writeHeaderCallback(): RequestId: fe77cfdd-6ca4-4233-ac45-9b20736d50f1
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
2019-08-07 11:58:29 [0x70000ded0000] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - getStreamingTokenHandler invoked
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1565204309325690 Expiration: 18446744073709551615
2019-08-07 11:58:29 [0x70000ded0000] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - defaultStreamReadyCallback(): Reported streamReady callback for stream handle 140634653080873
2019-08-07 11:58:29 [0x70000ded0000] DEBUG - streamReadyHandler invoked
2019-08-07 11:58:29 [0x70000ded0000] Stream is ready
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0: Invalid video capture device index
Additional debug info:
avfvideosrc.m(360): -[GstAVFVideoSrcImpl openDeviceInput] (): /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0
Setting pipeline to NULL ...
Freeing pipeline ...
INFO - Freeing Kinesis Video Stream YourStreamName
2019-08-07 11:58:50 [0x7fffa9c7b380] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2019-08-07 11:58:50 [0x7fffa9c7b380] DEBUG - defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 140634653080873

Alex-fanfanfan, 2019-08-07:
I fixed the kvs_log_configuration but get another issue.

> Setting pipeline to PAUSED ...
[INFO ][2019-08-07 19:11:30] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-08-07 19:11:30] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-08-07 19:11:30] heapInitialize(): Creating AIV heap.
[INFO ][2019-08-07 19:11:30] heapInitialize(): Heap is initialized OK
[DEBUG][2019-08-07 19:11:30] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-08-07 19:11:30] getSecurityTokenHandler invoked
[DEBUG][2019-08-07 19:11:30] Refreshing credentials. Force refreshing: 0 Now time is: 1565205090640459 Expiration: 0
[DEBUG][2019-08-07 19:11:30] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-08-07 19:11:30] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-08-07 19:11:30] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-08-07 19:11:30] clientReadyHandler invoked
[INFO ][2019-08-07 19:11:30] try creating stream
[INFO ][2019-08-07 19:11:30] Creating Kinesis Video Stream YourStreamName
[INFO ][2019-08-07 19:11:30] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-08-07 19:11:30] stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-08-07 19:11:30] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-08-07 19:11:31] writeHeaderCallback(): RequestId: 6a0cecca-0629-4f3d-a6f6-205f5401f1bd
[DEBUG][2019-08-07 19:11:31] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.565202306662E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-1:746510367092:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-1:746510367092:stream/YourStreamName/1565202306662","StreamName":"YourStreamName","Version":"gO0jlcpDHPt4VZKhl9we"}}
[INFO ][2019-08-07 19:11:31] describeStreamResultEvent(): Describe stream result event.
[WARN ][2019-08-07 19:11:31] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG][2019-08-07 19:11:31] stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
[DEBUG][2019-08-07 19:11:31] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-08-07 19:11:32] writeHeaderCallback(): RequestId: 5f122cc1-2dd4-43bd-b9a0-593c5e1b97f3
[DEBUG][2019-08-07 19:11:32] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-1e415f8b.kinesisvideo.us-east-1.amazonaws.com"}
[INFO ][2019-08-07 19:11:32] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-08-07 19:11:32] stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
[DEBUG][2019-08-07 19:11:32] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[DEBUG][2019-08-07 19:11:32] getStreamingTokenHandler invoked
[DEBUG][2019-08-07 19:11:32] Refreshing credentials. Force refreshing: 1 Now time is: 1565205092141098 Expiration: 18446744073709551615
[INFO ][2019-08-07 19:11:32] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-08-07 19:11:32] stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-08-07 19:11:32] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 140684875552041
[DEBUG][2019-08-07 19:11:32] streamReadyHandler invoked
Stream is ready
ERROR: Pipeline doesn't want to pause.
ERROR: from element /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0: Invalid video capture device index
Additional debug info:
avfvideosrc.m(360): -[GstAVFVideoSrcImpl openDeviceInput] (): /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0
Setting pipeline to NULL ...
Freeing pipeline ...
[INFO ][2019-08-07 19:11:54] Freeing Kinesis Video Stream YourStreamName
[INFO ][2019-08-07 19:11:54] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-08-07 19:11:54] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 140684875552041

zhiyua-git, 2019-08-07:
Have you exit the previous gst-launch sending to us-west-2? It will occupy the device and you cannot run two gst-launch on the source at the same time.

Alex-fanfanfan, 2019-08-07:
How to exit the previous gst-launch?

zhiyua-git, 2019-08-07:
You can use ps -ef | grep gst-launch-1.0 to find all running gst-launch-1.0 command and kill them.

Alex-fanfanfan, 2019-08-07:
I killed all of them but still get the same error.

This is what I got entering ps-ef | grep gst-launch-1.0.

> $ ps -ef | grep gst-launch-1.0
  501  1171  1138   0 12:49PM ttys000    0:00.01 grep gst-launch-1.0

chehefen, 2019-08-07:
can you remove `device-index=1` from the pipeline and try again?

Alex-fanfanfan, 2019-08-07:
Thank you for your reply,
I got another error message:
```
Stream is ready
Pipeline is live and does not need PREROLL ...
/GstPipeline:pipeline0/GstOsxAudioSrc:osxaudiosrc0: device = 56
Setting pipeline to PLAYING ...
New clock: GstAudioSrcClock
ERROR: from element /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(2939): void gst_base_src_loop(GstPad *) (): /GstPipeline:pipeline0/GstAVFVideoSrc:avfvideosrc0:
streaming stopped, reason not-negotiated (-4)
Execution ended after 0:00:00.001433000
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
[INFO ][2019-08-07 20:44:03] Freeing Kinesis Video Stream YourStreamName
[INFO ][2019-08-07 20:44:03] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-08-07 20:44:04] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 140568400203561

```

closed 2019-08-08 by Alex-fanfanfan

iuprade, 2020-10-13:
what was the solution to this issue ?