awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #248: WARNING: erroneous pipeline: no element "kvssink"

ac4289, 2019-08-07:
Hi,
I am trying to use KVS on Jetson TX2 with JetPack 4.2. I compiled KVS producer library as instructed and copied files to `/usr/lib/aarch64-linux-gnu/`.
Here is my command.
```
sudo gst-launch-1.0 nvarguscamerasrc ! 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, format=(string)NV12, framerate=30/1' ! nvtee ! nvvidconv flip-method=0  ! 'video/x-raw, format=(string)NV12, width=640, height=480, framerate=30/1' ! omxh264enc  ! video/x-h264, stream-format=avc, alignment=au ! kvssink stream-name=*********** storage-size=512 access-key=************ secret-key=*********************
```
It worked right after compilation finished. Then I rebooted TX2 and it does not work anymore.
The only line of error message is
```
WARNING: erroneous pipeline: no element "kvssink"
```
But libgstkvssink.so exists under `/usr/lib/aarch64-linux-gnu/gstreamer-1.0`.
Here is the gst-inspect-1.0 output.
```
$ gst-inspect-1.0 kvssink

(gst-plugin-scanner:7820): GStreamer-WARNING **: 16:41:50.069: Failed to load plugin '/usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstkvssink.so': /usr/lib/aarch64-linux-gnu/gstreamer-1.0/libgstkvssink.so: undefined symbol: _ZN3com9amazonaws7kinesis5video18CredentialProvider12getCallbacksEP17__ClientCallbacks
No such element or plugin 'kvssink'
```
How can i fix this error?
Thank you!

zhiyua-git, 2019-08-08:
Hi @ac4289 ,

You need to add directory of libgstkvssink.so to environment variable GST_PLUGIN_PATH or move it to the directory where gstreamer plugins are in.

chehefen, 2019-08-08:
It's complaining about not be able to find symbol in libproducer.so. the plugin is dynamically linked to the libproducer.so. Can you do `export LD_LIBRARY_PATH=/path/to/dir` for the directory that has libproducer.so?

ac4289, 2019-08-08:
> It's complaining about not be able to find symbol in libproducer.so. the plugin is dynamically linked to the libproducer.so. Can you do `export LD_LIBRARY_PATH=/path/to/dir` for the directory that has libproducer.so?

Thank you for helping!
I tried this:
```
export LD_LIBRARY_PATH=/usr/lib/aarch64-linux-gnu/gstreamer-1.0
```
Still does not work.

chehefen, 2019-08-08:
is that where your libproducer.so is?

ac4289, 2019-08-08:
> is that where your libproducer.so is?

Yes

chehefen, 2019-08-08:
what about libcproducer.so?

ac4289, 2019-08-08:
libcproducer.so is under `/usr/lib/aarch64-linux-gnu/`

chehefen, 2019-08-08:
can you move it to /usr/lib/aarch64-linux-gnu/gstreamer-1.0 too? and do `export LD_LIBRARY_PATH=/usr/lib/aarch64-linux-gnu/gstreamer-1.0`

ac4289, 2019-08-08:
> LD_LIBRARY_PATH=/usr/lib/aarch64-linux-gnu/gstreamer-1.0

Does not work. T_T

chehefen, 2019-08-08:
just wanna confirm libcproducer.so and libproducer.so are all in /usr/lib/aarch64-linux-gnu/gstreamer-1.0?

ac4289, 2019-08-08:
> just wanna confirm libcproducer.so and libproducer.so are all in /usr/lib/aarch64-linux-gnu/gstreamer-1.0?

Yes

unicornss, 2019-08-08:
Hi @ac4289  ,

> It worked right after compilation finished. 

Given that it worked first time, could you please run the [`source set_kvs_sdk_env.sh`](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/install-script#L692) and see if that helps.

ac4289, 2019-08-09:
> Hi @ac4289 ,
>
>
> Given that it worked first time, could you please run the [`source [...]

No I already tried that. I just reflashed TX2 with JetPack 4.2. I'll see how it goes.

ac4289, 2019-08-10:
Ok, reflashed TX2 works with KVS. The difference is that the old OS had TVM installed which requires a change of Python path in .bashrc. 
Thank you again for your help!

closed 2019-08-10 by ac4289

aimroboticsbying, 2019-09-24:
I'm getting the same error: no element "kvssink"
plugin scanner says: Failed to load libgstx264.so, undefined symbol x264_deblock_h_chroma_422_intra_neon
Did you have this issue? Do you know how to fix it? Thanks.

MushMal, 2019-09-24:
Once you've successfully built the plugin, you should set the paths per: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-macos.md#setting-the-environment-variables-for-library-path