awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #249: KinesisVideoRendererExampleTest caused Codec changed

Alex-fanfanfan, 2019-08-09:
I can watch the video instream in AWS console. When I ran KinesisVideoRendererExampleTest, the video stream of Java is good but the AWS console's showed this error.
The java's video stream worked well. What's more, some bad error happened when I tried to run RekognitionStreamProcessorTest. 
All code is from this link. 
https://github.com/aws/amazon-kinesis-video-streams-parser-library
> Codec changed
The detected codec changed between fragments. The codec must be consistent across fragments.
Previous Codec ID:
avc1.4d0020 (H.264 Main@3.2)
mp4a.40.2 (AAC LC)
Codec ID:
avc1.42c028 (H.264 Baseline@4.0)
mp4a.40.2 (AAC LC)
Fragment number: 91343852333181620559568033999917556532668381367
Producer timestamp: August 9, 2019 3:57 PM PDT
Server timestamp: August 9, 2019 3:57 PM PDT
Size: 74.1 KB
This may be an intermittent issue. Retry the same action, or try refreshing your browser tab.

MushMal, 2019-08-09:
@Alex-fanfanfan ideally, you should post this question to the appropriate GitHub projects issue list. This project is tracking the C++/Native KVS SDK. 

What you are using is a simple test which streams various types of previously generated MKVs. The example file contains different tests which upload different types of files into the same stream. These files contain different CPDs (Codec Private Data) which makes the continuous playback impossible.

I would suggest to modify your application to continuously upload a single file or playback the same set of frames like the examples in the Java SDK https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java

unicornss, 2019-11-05:
Closing the issue as it has been answered.

closed 2019-11-05 by unicornss