awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #250: kinesis_video_gstreamer_sample_app: RTSP source; specify width, height, bitrate, framerate error

claw68, 2019-08-14:
Hi, I want to compress RTSP stream because it carries too much load over the network if it streams multiple cameras.

I tried running:

 `./kinesis_video_gstreamer_sample_app stream-name rtsp://url`

And it works just fine. 

But when i run:

`./kinesis_video_gstreamer_sample_app stream-name -w 1280 -h 720 -f 30 -b 2048 rtsp://url`

it doesn't run as expected and returns:

`(kinesis_video_gstreamer_sample_app.exe:8860): GStreamer-WARNING **: 11:16:06.884: Failed to load plugin 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstopencv.dll': 'C:\msys64\mingw64\lib\gstreamer-1.0\libgstopencv.dll': The specified module could not be found.

(kinesis_video_gstreamer_sample_app.exe:8860): GLib-GObject-WARNING **: 11:16:06.931: g_object_set_is_valid_property: object class 'GstKsVideoSrc' has no property named 'device'
Resolution 1280x720 not supported by video source
[INFO ][2019-08-14 03:16:05] Using region: region
[INFO ][2019-08-14 03:16:05] Using aws credentials for Kinesis Video Streams
[INFO ][2019-08-14 03:16:05] No session token was detected.
[INFO ][2019-08-14 03:16:05] Initializing curl.
[INFO ][2019-08-14 03:16:05]
createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-08-14 03:16:05]
heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-08-14 03:16:05]
heapInitialize(): Creating AIV heap.
[INFO ][2019-08-14 03:16:05]
heapInitialize(): Heap is initialized OK
[DEBUG][2019-08-14 03:16:05] Refreshing credentials. Force refreshing: 0 Now time is: 1565752565472264100 Expiration: 0
[INFO ][2019-08-14 03:16:05] New credentials expiration is 1565754965
[DEBUG][2019-08-14 03:16:05] createDeviceHandler invoked
[INFO ][2019-08-14 03:16:05]
createDeviceResultEvent(): Create device result event.
[DEBUG][2019-08-14 03:16:05] clientReadyHandler invoked
[DEBUG][2019-08-14 03:16:05] Awaiting for the producer to become ready...
[DEBUG][2019-08-14 03:16:05] Kinesis Video producer is Ready.
[DEBUG][2019-08-14 03:16:05] Client is ready
[INFO ][2019-08-14 03:16:05] Creating Kinesis Video Stream stream-name
[INFO ][2019-08-14 03:16:05]
createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-08-14 03:16:05] describeStreamHandler invoked
[DEBUG][2019-08-14 03:16:05] Awaiting for the stream to become ready...
[DEBUG][2019-08-14 03:16:05] Refreshing credentials. Force refreshing: 0 Now time is: 1565752565473834000 Expiration: 0
[DEBUG][2019-08-14 03:16:06] describeStream response: {"StreamInfo":{"CreationTime":1.565171626561E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn","StreamName":"stream-name","Version":"Qr6RuGUnmgYDAxhYPpkB"}}
[DEBUG][2019-08-14 03:16:06] Discovered existing Kinesis Video stream: arn
[DEBUG][2019-08-14 03:16:06] stream arn in stream_info struct: arn
[INFO ][2019-08-14 03:16:06]
describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-08-14 03:16:06] streamingEndpointHandler invoked
[DEBUG][2019-08-14 03:16:06] Refreshing credentials. Force refreshing: 0 Now time is: 1565752566175859500 Expiration: 0
[DEBUG][2019-08-14 03:16:06] getStreamingEndpoint response: {"DataEndpoint":"endpoint"}
[INFO ][2019-08-14 03:16:06] streaming to endpoint: endpoint
[INFO ][2019-08-14 03:16:06]
getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-08-14 03:16:06] streamingTokenHandler invoked
[DEBUG][2019-08-14 03:16:06] Refreshing credentials. Force refreshing: 1 Now time is: 1565752566783437500 Expiration: 1565754965
[INFO ][2019-08-14 03:16:06] New credentials expiration is 1565754966
[INFO ][2019-08-14 03:16:06]
getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-08-14 03:16:06] streamReadyHandler invoked
[DEBUG][2019-08-14 03:16:06] Kinesis Video stream stream-name is Ready.
[DEBUG][2019-08-14 03:16:06] Stream is ready
[INFO ][2019-08-14 03:16:06] Streaming from live source
[DEBUG][2019-08-14 03:16:06] Streaming with live source and width: 1280, height: 720, fps: 30, bitrateInKBPS2048
[INFO ][2019-08-14 03:16:06] Freeing Kinesis Video Stream stream-name
[INFO ][2019-08-14 03:16:06]
freeKinesisVideoStream(): Freeing Kinesis Video stream.
[INFO ][2019-08-14 03:16:06]
freeKinesisVideoClient(): Freeing Kinesis Video Client
[INFO ][2019-08-14 03:16:06]
commonHeapDebugCheckAllocator(): Heap is initialized
[INFO ][2019-08-14 03:16:06]
commonHeapDebugCheckAllocator(): Heap limit:                            134217728
[DEBUG][2019-08-14 03:16:06] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ][2019-08-14 03:16:06]
commonHeapDebugCheckAllocator(): Heap size:                             0
[INFO ][2019-08-14 03:16:06]
commonHeapDebugCheckAllocator(): Number of allocations:                                 0
[INFO ][2019-08-14 03:16:06]
aivHeapDebugCheckAllocator(): Allocated blocks pointer:                                 0000000000000000
[INFO ][2019-08-14 03:16:06]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-08-14 03:16:07]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-08-14 03:16:07]
aivHeapDebugCheckAllocator(): Free blocks pointer:                              000000000381c040
[INFO ][2019-08-14 03:16:07]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-08-14 03:16:07]
aivHeapDebugCheckAllocator(): Block:    000000000381c040                size:   134217679
[INFO ][2019-08-14 03:16:07]
aivHeapDebugCheckAllocator(): *******************************************
[INFO ][2019-08-14 03:16:07]
heapRelease(): Freeing native heap.
`

chehefen, 2019-08-16:
you cant control the bitrate, resolution and framerate of a rtsp stream since it has already been encoded. Thus the sample app doesnt support -w 1280 -h 720 -f 30 -b 2048 for rtsp stream. You have to configure them in your rtsp source.

unicornss, 2019-11-05:
Closing the issue as it has been answered (RTSP source params are to be controlled in the camera for best performance).

closed 2019-11-05 by unicornss