awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #251: Failed to create Kinesis Video Stream - timed out.

Xiaoron, 2019-08-14:
I meet an issue that sometimes can not create Kinesis Video Stream,
Below is my test log, can u help me check?
[kvs_2019081309.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3501207/kvs_2019081309.log)

aws-kinesis-video, 2019-08-14:
What is the application you are running? What region are you in? By default sample applications are using us-west-2 as endpoint, if you have high network latency against us-west-2 endpoint, it is possible to timeout the operation and fail the createStream.

zhiyua-git, 2019-08-14:
What is the application you are running? What region are you in? By default sample applications are using us-west-2 as endpoint, if you have high network latency against us-west-2 endpoint, it is possible to timeout the operation and fail the createStream.

jiangxiaorong, 2019-08-15:
@zhiyua-git I develop my producer application base on test sample code under kinesis-video-producer. My camera is set up in China.

zhiyua-git, 2019-08-15:
You can try to set region as ap-northeast-1 or ap-southeast-2 to get lower latency.

jiangxiaorong, 2019-08-19:
@zhiyua-git Thanks for your suggestion, actually when I set to ap-southeast-2 I also meet the same issue.

zhiyua-git, 2019-08-21:
@jiangxiaorong Can you paste a test log with region set to ap-southeast-2? I saw some 2-second delays in the log when state machine is doing state transition and would like to double check if that latency is caused by cross-region access or other source.

MushMal, 2019-09-17:
@jiangxiaorong do you have more information on the timeout?

jiangxiaorong, 2019-09-18:
@MushMal thanks you. I think it's because my network is not stable, and add retry when create stream failed. now I can create stream. Thank you again.

closed 2019-09-18 by MushMal