awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #252: Free stream take a lot of time

Xiaoron, 2019-08-14:


MushMal, 2019-08-14:
@Xiaoron freeKinesisVideoStream() is a "prompt" operation - meaning it's almost instantaneous. What's likely happened is something crashes the app. Can you provide more info/

jiangxiaorong, 2019-08-15:
@MushMal , can you kindly tell me what information you want?
My producer application is used to upload event clip to server. Firstly, create a kvs stream, if an event happened, call resetStream api, and then get frame from my streaming server and call putFrame api to put frame to kvs stream, after finishing put all frames, will call api stopSync to stop the stream. if resetStream failed, or stopSync failed, will call freeStream api.

I meet cases that sometimes freeStream api take a lot of times, and resetStream sometimes also take a lot of time.

MushMal, 2019-08-15:
Not sure why you would need to resetStream API after creating it to put frames into it. You can simply createStreamSync to create the stream, put frames into it, stopStreamSync and the freeStream.

Looking at the log you've attached, there are two interesting things

1) You are not producing an I-frame into the stream 
2) There are no logs after freeStream.

This is what made me think that your application simply crashed after the free stream returned or something similar.

Can you please re-run the scenario and investigate whether you are producing I-frames or not and what happens after the stopSync and freeStream is called. Please run this under a debugger so it can give you more info on the stack trace perhaps in case of the crash.

jiangxiaorong, 2019-08-15:
@MushMal thanks for your help, I will do the test.

MushMal, 2019-08-15:
stopStreamSync is a rather complex API - it will attempt to stream out the remaining bits from the buffer and properly receive the ACK from the backend before closing the connection and being done. It can, in fact, timeout - for example, if there is networking outage or something similar. 

FreeStream is basically unconditional freeing - it will remove and delete any data in the buffers and close the connection.

Many customers that need immediate termination do not call stopStreamSync but simply call stopStream and follow up with freeStream

jiangxiaorong, 2019-08-15:
@MushMal 
1. The reason why I use resetStream API is I want trigger the stream into a state that can receive frame again after stop it.
2. I have did a test that use start() to set the stream into a that ready to receive frames, but I miss error 0x52000052, the error means stream is in stop state, but still try to put frame to it. 
My test code flow is :
createStream()->start()->putFrames()...->stopSync()......->start()->putFrames()->stopSync().

What API should I use if I want set the stream into state that ready to receive frames again after stop it?

MushMal, 2019-08-16:
Sorry for the delay. 0x52000052 status code indeed means that you are attempting to put frame or metadata into the stream after it's been stopped. Check your code and ensure no more frames are put into the stream after stop or stopSync.

Generally, your flow should be either:

createStream<Sync>()->start()->putFrames()...-stopStream<Sync>()...resetStream()....->start()->putFrames()....

or

createStream<Sync>()->start()->putFrames()...-stopStream<Sync>()...freeStream()....createStream<Sync>()->start()->putFrames()....

Either way, freeStream should return immediately unless it crashes. Please run your scenario under a debugger to catch any faults/exceptions which can help to identify what's going on.

Please note that stopStreamSync() can take multiple seconds to finish or timeout by default.

Xiaoron, 2019-08-19:
@MushMal I am trying debug on debugger, my enviroment is and embed linux system, so, it need some time.

Today, I try to freeStream if stopsync failed, the freeStream process take about 10 minutes. you can find at 2019-8-19 04:13:14 stop stream failed, at 2019-8-19 18:14 start freeStream, at 2019-8-19 04:28:45 begin drop fragment
### "
2019-08-19 04:13:14 [1108182192] ERROR - stopStreamSync(): Failed to stop Kinesis Video Stream - timed out.
2019-08-19 04:18:14 [1074467648] ERROR - Failed to stop the stream with: 15
2019-08-19 04:18:14 [1074467648] INFO - Freeing Kinesis Video Stream camera-core-device-4189
2019-08-19 04:18:14 [1074467648] INFO - freeKinesisVideoStream(): Freeing Kinesis Video stream.
2019-08-19 04:18:14 [1074467648] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east-1.amazonaws.com/getDataEndpoint with result Error: Unknown SSL protocol error in connection to kinesisvideo.us-east-1.amazonaws.com:443 
2019-08-19 04:28:44 [1125799088] WARN - curlCompleteSync(): HTTP Error 0 : Response: (null)
Request URL: https://kinesisvideo.us-east-1.amazonaws.com/getDataEndpoint
Request Headers:
    Authorization: AWS4-HMAC-SHA256 Credential=ASIAT4XQRRLSUL67NAWW/20190819/us-east-1/kinesisvideo/aws4_request, SignedHeaders=content-type;host;user-agent;x-amz-date, Signature=a7eed5174bea56ca14b4d7cb27566b7502460d3f46eda9997e18c55bd8fb2135
    content-type: application/json
    host: kinesisvideo.us-east-1.amazonaws.com
    user-agent: AWS-SDK-KVS/2.0.1 GCC/4.9
2019-08-19 04:28:44 [1125799088] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: 
2019-08-19 04:28:45 [1125799088] TRACE - terminateCurlSession(): Force stopping the curl connection
2019-08-19 04:28:45 [1125799088] DEBUG - defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 681749
2019-08-19 04:28:45 [1074467648] WARN - viewItemRemoved(): Reporting a dropped frame/fragment.
"

MushMal, 2019-08-19:
Something is pretty weird going on. It seems that your observed bandwidth is fluctuating between a few Mbps to up-to 32Mbps. There were a few times stopsync/reset calls worked in the log. 
At the time of the free stream you call stop sync you there is about 16 seconds worth of content in the buffer. Stopsync fails with a timeout error in 5 mins (this is odd as by default the timeout is 2 minutes). Next, your free stream gets blocked trying to terminate CURL for 10 minutes:

> 2019-08-19 04:18:14 [1074467648] WARN - curlCompleteSync(): curl perform failed for url https://kinesisvideo.us-east-1.amazonaws.com/getDataEndpoint with result Error: Unknown SSL protocol error in [...]

Log4cplus has a known issue with logging a line of text with a previous timestamp.

The freeStream path calls shutdown callback which attempts to kill in-flight requests. In your case, there is a getDataEndpoint request that takes 10 minutes to kill. This is where I am not sure what's happening.

Are you using your platform built curl or you are building CURL library using install-script?

Could you also add the following line in your main CMakeLists.txt file to get extra verbose logging

`add_definitions(-DLOG_STREAMING)`

MushMal, 2019-12-04:
I am going to resolve this issue for now. Please re-open after attempting to stream on the latest release version

closed 2019-12-04 by MushMal