awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #253: MAX_FRAGMENT_DURATION_REACHED

arnamoy10, 2019-08-14:
Hello;

I am trying to stream an mp4 file [file link](https://drive.google.com/file/d/1jdLYrGRegEYqMYpOFDCDGY-s97OLyYg6/view?usp=sharing) a kinesis stream using the gstreamer_sample_app that comes with the git repo.

But I am getting the following error:

`Curl post body write function for stream: evastream1 and upload handle: 13 returned: {"EventType":"ERROR","FragmentTimecode":1565810137603,"FragmentNumber":"91343852338176208490308672115616860505404839548","ErrorCode":"MAX_FRAGMENT_DURATION_REACHED","ErrorId":4002}`

I have tried the fix #147 by chaning the key-int-max=10 and frame-timestamp=1 in the cpp file `( g_object_set() ).`  But none of them helped.

Could you please tell me what I might be doing wrong?

MushMal, 2019-08-14:
@arnamoy10 I looked at the file (opened in http://lvcdn.net/gop/) and it seems that 

1) File has multiple B-frames (OK if non-intrafragment referenced)
2) File has a single GOP which is 20 seconds long.

KVS has a restriction on a fragment duration - it's 10 seconds. You can try to configure the original encoder that produced this MPEG to produce smaller GOPs or re-encode.

arnamoy10, 2019-08-14:
Thank you so much for the reply.  Fixing the GOP of the video fixed the issue.  The video had around 28 fps.  I used the following command to lower the GOP:

`ffmpeg -i dummy2.mp4 -g 200 dummy3.mp4`

closed 2019-08-14 by arnamoy10