awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #254: Error - ./min-install-script: line 170: make: command not found cp: cannot stat 'libjsmn.a': No such file or directory

ylulloa, 2019-08-16:
I'm going over the step by step guide: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msys2.md

I can't pass step #5. Error:
./min-install-script: line 170: make: command not found cp: cannot stat 'libjsmn.a': No such file or directory

[image]

Running this as Admin in a Windows 10 Pro.

Any suggestion?

zhiyua-git, 2019-08-16:
Can you double check if you are running ./min-install-script in mingw32 or mingw64 shell? You cannot run it from msys2 shell.

ylulloa, 2019-08-16:
That was it! Thanks.

closed 2019-08-16 by ylulloa

hussamsayeed, 2019-09-25:
$ cd kinesis-video-native-build; ./min-install-script
bash: cd: kinesis-video-native-build: No such file or directory
./min-install-script: line 15: /home/Hussam: No such file or directory

this is the error raising while Building the Kinesis Video Streams Producer SDK using min-install-script.. Anyone please help me out!

surekha3079, 2020-09-15:
smanda@VSC-WS216 MINGW64 /c/AWS/amazon-kinesis-video-streams-producer-sdk-cpp/build (master)
$ ./min-install-script
bash: ./min-install-script: No such file or directory
Could u pls suggest me..

MushMal, 2020-09-15:
@surekha3079 please use the latest SDK commits. We have no longer scripts to run. Check out the readme in the root directory to get more info on how to build and get specifics on what prerequisites  to install for Windows.

Once you try and if you have further questions please cut a separate issue as this one is closed