awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #255: KVS misinterprets video frame resolution

ecolgan, 2019-08-20:


unicornss, 2019-08-20:
Hi @ecolgan 

Could you please clarify

> However, the kvs client player claims it is 256 x 528 and so decoding fails entirely. 

Are you seeing this during playback in the console? or downloading the ingested media fragments through the GetMedia API?

ecolgan, 2019-08-20:
We are seeing this in the console.

unicornss, 2019-08-30:
Could you please provide the screen shot from the console to see the details returned on the playback?

Also, can you use the KVS [HLS viewer](https://github.com/aws-samples/amazon-kinesis-video-streams-hls-viewer/blob/master/index.html ) to check the playback on this stream?
https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/hls-playback.html

If you are seeing an issue here, send us the log information (without credentials information) to  kinesis-video-support@amazon.com.

Thanks

MixMasterMitch, 2019-11-20:
This issue is now resolved.

closed 2019-11-20 by MixMasterMitch