awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #256: Streaming h.265 video from RTSP

NebiyouTen, 2019-08-21:
Hi, 

I am using the following pipeline to stream an RTSP source to kvs.  

`gst-launch-1.0 rtspsrc location=$1 ! rtph265depay ! h265parse ! video/x-h265, alignment=au ! kvssink stream-name=$2 storage-size=512 access-key=$3 secret-key=$4 aws-region=$5
`

But I get the following error in the console

> Fragment decoding error
> There was an error decoding the media. Verify that the stream contains valid content.
> RetryHide details
> Codec ID:
> hvc1.1.6.L0.0 (H.265 Main@L0.0)
> Fragment number: 91343852333181467074810203241608003385279896619
> Producer timestamp: August 21, 2019 11:05 PM UTC
> Server timestamp: August 21, 2019 11:05 PM UTC
> Size: 71.4 KB
> This may be an intermittent issue. Retry the same action, or try refreshing your browser tab

(I have tried retrying and refreshing)

Are h.265 streams supported by kvs? The same pipeline works fine for h264 streams, just by changing the parser and deplayloder to h.264.

unicornss, 2019-08-21:
Hi @NebiyouTen,

Can you please provide the logs from the producer (sending video data) without any security/credentials information? 

You can also send the log to the kinesis-video-support@amazon.com

Thanks

NebiyouTen, 2019-08-22:
I have sent the logs to kinesis-video-support@amazon.com. 

Thanks.

NebiyouTen, 2019-08-23:
Also, I was able to generate a `KVS_DEBUG_DUMP_DATA_FILE` and the MKV video looks good. The codec is **V_MPEGH/ISO/HEVC**, which I think should be supported. 

Any ideas?

MushMal, 2019-08-23:
Hi @NebiyouTen sorry for the delay. The logs didn't show anything immediately standing out - saw some RTP warnings but other than that it looks ok. Can we suggest to do the following:

1) Some browsers might not have H265 decoder support. Could you please try another browser like Safari?
2) Could you use the HLS player sample to attempt to playback HLS/MPEG-DASH? Save the content of this file: https://raw.githubusercontent.com/aws-samples/amazon-kinesis-video-streams-media-viewer/master/index.html Open in the browser, fill in the details and attempt to playback
3) Same with the playback but try to download the resulting MPEG fragment and attempt to playback in VLC
4) Could you send us a couple of the captured fragments from MKV dump and the MPEG fragment from HLS/MPEG-DASH if they don't playback?

NebiyouTen, 2019-08-23:
Thanks for the reply. I will try the suggested points and will get back to you.

NebiyouTen, 2019-08-23:
1. I was able to play it using Microsoft Edge. But It didn't work on Safari Version 12.1.2, Chrome both on Mac and on windows, and on IE. 

2- 3. The HLS player doesn't seem to play the video. I filled in all the required information but nothing shows up on the player and the logs field is empty. 

4. I have sent the MKV dump. 

Thanks.

MushMal, 2019-08-25:
We received the MKV file and it plays OK in VLC player. Also, the inspection of the MKV didn't show anything wrong with it. Could you confirm that you were able to playback the h265 stream using console on Microsoft Edge?

NebiyouTen, 2019-08-25:
Yes, I can confirm that I was able to playback the h265 stream (codec id: **hvc1.1.6.L0.0 (H.265 Main@L0.0)**) using Microsoft Edge.

MushMal, 2019-08-25:
Referencing https://caniuse.com/#feat=hevc seems that the support is pretty spotty - for example IE11 and Edge 18 support HEVC when hardware acceleration present, Safari on macOS supports it on 11-12 and 12.1 on macOS High Sierra and later. No support for Chrome it seems (there are some plugins though).

thinhlx1993, 2019-09-14:
> Hi,
>
> I am using the following pipeline to stream an RTSP source to kvs.
>
> `gst-launch-1.0 rtspsrc location=$1 ! rtph265depay ! h265parse ! video/x-h265, alignment=au ! kvssink stream-name=$2 [...]

@NebiyouTen 
Could you help me? When I start gst-launch I got this message. I think i missing something when installing stream library.
WARNING: erroneous pipeline: could not link h265parse0 to kvssink0, kvssink0 can't handle caps video/x-h265, alignment=(string)au

MushMal, 2019-09-14:
@NebiyouTen h265 is part of 'videoparsersbad' according to https://gstreamer.freedesktop.org/data/doc/gstreamer/head/gst-plugins-bad/html/gst-plugins-bad-plugins-h265parse.html

Could you make sure you have the bad plugins installed per readmes.

To verify, launch `gst-inspect-1.0 h265parse` and print out the result?

thinhlx1993, 2019-09-14:
Here is my result after run gst-inspect-1.0. Can you check it for me?

Factory Details:
  Rank                     secondary (128)
  Long-name                H.265 parser
  Klass                    Codec/Parser/Converter/Video
  Description              Parses H.265 streams
  Author                   Sreerenj Balachandran <sreerenj.balachandran@intel.com>

Plugin Details:
  Name                     videoparsersbad
  Description              videoparsers
  Filename                 /home/pi/kinesis-video-native-build/kinesis-video-native-build/downloads/local/lib/gstreamer-1.0/libgstvideoparsersbad.so
  Version                  1.12.3
  License                  LGPL
  Source module            gst-plugins-bad
  Source release date      2017-09-18
  Binary package           GStreamer Bad Plug-ins source release
  Origin URL               Unknown package origin

GObject
 +----GInitiallyUnowned
       +----GstObject
             +----GstElement
                   +----GstBaseParse
                         +----GstH265Parse

Pad Templates:
  SINK template: 'sink'
    Availability: Always
    Capabilities:
      video/x-h265

  SRC template: 'src'
    Availability: Always
    Capabilities:
      video/x-h265
                 parsed: true
          stream-format: { (string)hvc1, (string)hev1, (string)byte-stream }
              alignment: { (string)au, (string)nal }

Element Flags:
  no flags set

Element Implementation:
  Has change_state() function: gst_base_parse_change_state

Element has no clocking capabilities.
Element has no URI handling capabilities.

Pads:
  SINK: 'sink'
    Pad Template: 'sink'
  SRC: 'src'
    Pad Template: 'src'

Element Properties:
  name                : The name of the object
                        flags: readable, writable
                        String. Default: "h265parse0"
  parent              : The parent of the object
                        flags: readable, writable
                        Object of type "GstObject"
  disable-passthrough : Force processing (disables passthrough)
                        flags: readable, writable
                        Boolean. Default: false
  config-interval     : Send VPS, SPS and PPS Insertion Interval in seconds (sprop parameter sets will be multiplexed in the data stream when detected.) (0 = disabled)
                        flags: readable, writable
                        Unsigned Integer. Range: 0 - 3600 Default: 0

MushMal, 2019-09-16:
Seems like the plugin is present. I assume you've used install-script to build as it sets the correct paths to the GST? Could you try setting the GST_PLUGIN_PATH variable as described in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#setting-the-environment-variables-for-library-path