awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #257: Could NOT find Log4cplus (missing: LOG4CPLUS_LIBRARIES LOG4CPLUS_INCLUDE_DIR) Log4cplus is not found. Only c producer will be built.

norbyn, 2019-08-22:
I'm reproducing the steps of the guide: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msys2.md but step #6 is giving me this error:

 Could NOT find Log4cplus (missing: LOG4CPLUS_LIBRARIES LOG4CPLUS_INCLUDE_DIR) Log4cplus is not found. Only c producer will be built.

[image]

Running this as Admin in a Windows 10 Enterprise.

Any suggestion?

zhiyua-git, 2019-08-23:
Hi @norbyn ,

Can you double check if you are running ./min-install-script in mingw32 or mingw64 shell? You cannot run it from msys2 shell.

norbyn, 2019-08-23:
@zhiyua-git I ran the wrong shell msys2.exe, thanks.

closed 2019-08-23 by norbyn

BlueClouDragon, 2020-07-10:
I have same problem. I ran the command on mingw64 shell
but the same problem with @norbyn 
any solution?