awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #259: Install on MacOS - library not found for -lglib-2.0

js1972, 2019-08-29:
Following the install instructions for MacOS and have installed the dependencies with brew. When I run `./min-install-script` I get the error:

```
[  6%] Building C object CMakeFiles/cproducer.dir/Users/jason/dev/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-pic/src/client/src/ClientState.c.o
4 warnings generated.
[  7%] Linking CXX executable kvs_producer_plug?in_rtsp_demo
ld: library not found for -lglib-2.0
clang: error: linker command failed with exit code 1 (use -v to see invocation)
make[2]: *** [kvs_producer_plugin_rtsp_demo] Error 1
make[1]: *** [CMakeFiles/kvs_producer_plugin_rtsp_demo.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
```

Any ideas?

zhiyua-git, 2019-08-29:
Hi @js1972 ,

If you are running min-install-script on MacOS Mojave, then you will see this issue. It is not clear to us why cmake is not able to pick up the /usr/local/lib path on mac.

You can work around by adding **link_directories(/usr/local/lib)** at https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/CMakeLists.txt#L588

Then you can rerun ./min-install-script.

-ZH

js1972, 2019-08-30:
Thanks for the quick answer. I’ll try that tonight. My fallback would’ve been going to Docker. ;-)
I did try adding the link directories statement but I used the path to the libraries it could find instead of just /usr/local/lib.  Been a while since I’ve touched cmake.  ;-)

js1972, 2019-08-30:
That did the trick thanks. Interestingly using this method (./min-install-script) doesn't seem to create the gstreamer plugin as all those samples do not work. Only the cpp program examples are working. I also find that the preview on was kinesis always freezes up after a few seconds of footage. Will do some more testing and open up a different issue if necessary. Thx.

closed 2019-08-30 by js1972