awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #261: Failed to submit to kinesis video client errors 0x30000005 - stable encoder? Lots of issues with the sample app.

js1972, 2019-08-31:
I'm testing out kinesis and am getting some strange results. Running the 'kinesis_video_gstreamer_sample_app' against an RTSP camera (Hikvision).
I notice that the Amazon Kinesis dashboard preview works for only a few seconds then seems to freeze and the timestamp just starts increasing... minutes... 30 mins and so on. If I click the right-arrow to jump to the front of the stream the same happens - I see a couple of seconds of footage at best.
The Hikvision camera was on default settings of 2688*1520; full frame (which I think means the highest frame rate it can do).

I've turned these settings down to 1920x1080; 15fps. Still the kinesis preview is bad.
Btw. when I changed these settings it suggested I set the I-frame rate to the same as the fps so I did. Not sure what that means yet.  ;-)

Have also tried using the HLS website. It seems to keep playing better but I do have to keep hitting the play button every 5 seconds or so as it stops.

On these lower settings I noticed lots of "failed to submit to kinesis video client errors: 0x30000005". From some closed issues and the error code page I can see this is due to an unstable encoder... But what can I do about that if its an RTSP stream straight from the camera?

Additionally - the latency seems to be extreme! After I've been streaming for a while - if I walk in front of the camera it can be a good 4 minutes before I see myself in the HLS review!

Note: I have tested using the same camera stream in VLS player on my Mac and it stream perfectly with very minimal latency of maybe 1 second.

Am I maybe using far higher image sizes and fps than Kinesis is designed for? Surely not when even the most basic hone cameras and USB cameras are all full HD.

Below is a section of the console output:

	>> Current transfer rate (bps): 1731112 (1690 Kbps)
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3741554232 presentation timestamp: 3739365317
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3741555587 presentation timestamp: 3739764531
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 737120000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3741569112 presentation timestamp: 3740562978
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 737920000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3741571518 presentation timestamp: 3741361449
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 738720000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3743588327 presentation timestamp: 3742159945
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 739510000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3743593917 presentation timestamp: 3742958465
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3743596413 presentation timestamp: 3743357734
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 740710000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 741110000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3744694471 presentation timestamp: 3744555575
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 741910000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[DEBUG][2019-08-31 06:56:13] postWriteCallback(): Curl post body write function for stream with handle: test-stream and upload handle: 1 returned: {"EventType":"RECEIVED","FragmentTimecode":298135,"FragmentNumber":"91343852333181556260962393517760842976963690591"}

[DEBUG][2019-08-31 06:56:13] fragmentAckReceivedHandler invoked
[DEBUG][2019-08-31 06:56:13] Reporting fragment ack received. Ack timecode 298135
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] postWriteCallback(): Curl post body write function for stream with handle: test-stream and upload handle: 1 returned: {"EventType":"BUFFERING","FragmentTimecode":299136,"FragmentNumber":"91343852333181556265914153674902364765905690877"}

[DEBUG][2019-08-31 06:56:13] fragmentAckReceivedHandler invoked
[DEBUG][2019-08-31 06:56:13] Reporting fragment ack received. Ack timecode 299136
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 742310000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[DEBUG][2019-08-31 06:56:13] postWriteCallback(): Curl post body write function for stream with handle: test-stream and upload handle: 1 returned: {"EventType":"PERSISTED","FragmentTimecode":298135,"FragmentNumber":"91343852333181556260962393517760842976963690591"}

[DEBUG][2019-08-31 06:56:13] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 742710000
[DEBUG][2019-08-31 06:56:13] streamLatencyPressureHandler invoked
[DEBUG][2019-08-31 06:56:13] fragmentAckReceivedHandler invoked
[DEBUG][2019-08-31 06:56:13] Reporting fragment ack received. Ack timecode 298135
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3747704671 presentation timestamp: 3746152776
[WARN ][2019-08-31 06:56:13] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3747706811 presentation timestamp: 3746552091
[... 555 lines cut ...]
[DEBUG][2019-08-31 06:56:20] postReadCallback(): Wrote 16372 bytes to Kinesis Video. Upload stream handle: 1
[... cut ...]
[DEBUG][2019-08-31 06:56:20] streamLatencyPressureHandler invoked
[WARN ][2019-08-31 06:56:20] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 3819379590 presentation timestamp: 3818121697
[DEBUG][2019-08-31 06:56:20] defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601. Current buffer duration in 100ns: 745470000
[DEBUG][2019-08-31 06:56:20] streamLatencyPressureHandler invoked
^C

js1972, 2019-08-31:
The reference to VLS player above was meant to say VLC player.

MushMal, 2019-09-02:
@js1972 a couple of observations from the snippet of the logs:

* Your observed streaming throughput increases and stabilizes at around 2.6 Mbps.
* Your streaming has 74 seconds worth of data in the buffer - seems that with 2.6 Mbps it still falls behind.
* Your observed FPS is reported at 612 (frames per second).
* Due to high FPS, your frames are "clumped" together which makes the timestamps to overlap causing the 0x30000005 error. For example, the timestamp different between the frames at times is about 2 microseconds.

The KVS console playback as well as the HLS playback are fragment based playbacks - meaning, they are not realtime. The latency will include the fragment duration + the time it takes to download the fragment.

When you are streaming and viewing the RTSP stream you are putting 3x the bandwidth pressure on the network - first time to stream the frames from the camera to the device that streams the KVS, second time, the bandwidth of KVS streaming the frames and the third time is when your browser downloads the fragments for the local playback. Please ensure you have the bandwidth available.

Regarding the issue you are observing. It's possible that the RTSP stream is "bursty" - aka, it produces multiple frames in bursts, thus, causing the timestamps to collide.

Is there a way to set it up with a small resolution and high encoding quality to produce less dense stream to ensure there are no main issues with the bandwidth congestion first?

P.S. RE setting the GoP size - or I-frame interval. We recommend having 2-4 seconds fragments for optimal streaming. For 25fps streaming, this will evaluate to an I-frame every 50-100 frames. In order to lower the latency of HLS/Console playback, you can set it 1-second fragments which will be setting an I-frame every 25 frames in 25 fps video. Note: lowering the GoP size will generally increase the density (bitrate) of the stream.

js1972, 2019-09-02:
Thankyou @MushMal . Learning allot from your response here. I wondered about the fact that the console seems to report: "Total streams frame rate (fps): 612". Because my hikvision camera defaults to an fps of "full frame" (which means max frame rate it can handle) is this saying that it's actually managing to push out an fps of 612? That's seems extreme. I thought the maximum the camera would do was 25fps.

Your suggesting using a smaller image size. I will try with something like 720p. Maybe a small FPS like 10. So maybe set the I-frame to 30 to fit into your ideal of 2-4 second fragments.

You also mentioned about the amount of network load. If I'm seeing this in the log: "defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 140477886178601" - is that saying that the network us struggling?

MushMal, 2019-09-02:
the report in the log (the one you've attached) that comes from the SDK is about the "observed" fps - that is the rate the media source (RTSP GStreamer plugin) pushes frames into the KVS sink in the GStreamer sample application. If the RTSP source is "bursty" it will produce a set of frames in short time intervals which would result in high observed FPS.

The printout from "defaultStreamLatencyPressureCallback()" indicates that the KVS SDK core reports the buffer accumulated more than the max latency (from the StreamInfo) duration of frames. KVS plugin specifies 60 seconds by default. The beginning of the log there is 737120000 (100ns) duration in the buffer which is 73.7 seconds. This is likely indicative of a network throughput or very high latency issue in the network. There could be other issues like stale CURL handler but I couldn't see the whole log.

My suggestion would be to try a lot lower resolution first - SD resolution for example (480p). Capture the whole log from the producer and send it to kinesis-video-support@amazon.com

js1972, 2019-09-02:
Thankyou again @MushMal . I'll close this issue for now and continue testing with small resolutions and come back to that email with the full log output.

closed 2019-09-02 by js1972