awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #262: RTSP Fragments not found

jesssboxer, 2019-09-05:
I am on a Mac (OS Mojave) running the docker image as described in https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-rtsp.html.

I provided my keys and a public RSTP url in the docker command:
docker run -it 2e6c285722df MY_ACCESS_KEY MY_SECRET rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov KVStreamForAI

According to the attached [dockerKinesis.out.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3580064/dockerKinesis.out.txt) log, everything is sending and the cloud watch graph shows data is being received.

[image]

But I get an error when I try to get a fragment:
aws kinesis-video-archived-media get-media-for-fragment-list --stream-name KVStreamForAI --fragments='["91343852333181462128001806257567932306329308888"]' out.webm --endpoint-url https://b-604520a7.kinesisvideo.us-east-1.amazonaws.com

An error occurred (InvalidArgumentException) when calling the GetMediaForFragmentList operation: Fragment numbers are invalid : [91343852333181462128001806257567932306329308888]

I also tried to use the hosted viewer at https://aws-samples.github.io/amazon-kinesis-video-streams-media-viewer/ and it returned this error:

[ERROR] {
  "message": "No fragments found in the stream for the streaming request.",
  "code": "ResourceNotFoundException",
  "time": "2019-09-05T15:06:13.356Z",
  "requestId": "db33d135-ebc5-4364-9182-380ec2bf52b3",
  "statusCode": 404,
  "retryable": false,
  "retryDelay": 87.19487120417698
}

jesssboxer, 2019-09-05:
I figured out what happened. It was going to the west region. My local config is set to east so I assumed it would go there. How do I set the region when running through docker?

zhiyua-git, 2019-09-05:
Hi @jesssboxer ,

You can set environment variable to pass in the region.
docker run **-e AWS_DEFAULT_REGION=us-east-1** -it 7c99a34f9530 MY_ACCESS_KEY MY_SECRET  rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov KVStreamForAI

jesssboxer, 2019-09-05:
Perfect, thank you!

closed 2019-09-05 by jesssboxer