awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #263: Video stream is not working in AWS kinesis in RaspberryPi 3 B+

biranchi2018, 2019-09-08:
./kinesis_video_gstreamer_sample_app test_stream
log4cplus:ERROR Unable to open file: ./log/kvs.log
[INFO ][2019-09-08 10:01:00] Using region: ap-northeast-1
[INFO ][2019-09-08 10:01:00] Using aws credentials for Kinesis Video Streams
[INFO ][2019-09-08 10:01:00] No session token was detected.
[INFO ][2019-09-08 10:01:00] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-09-08 10:01:00] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-09-08 10:01:00] heapInitialize(): Creating AIV heap.
[INFO ][2019-09-08 10:01:00] heapInitialize(): Heap is initialized OK
[DEBUG][2019-09-08 10:01:00] stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x1
[DEBUG][2019-09-08 10:01:00] getSecurityTokenHandler invoked
[DEBUG][2019-09-08 10:01:00] Refreshing credentials. Force refreshing: 0 Now time is: 1567936860626350763 Expiration: 0
[INFO ][2019-09-08 10:01:00] New credentials expiration is 1567939260
[DEBUG][2019-09-08 10:01:00] stepStateMachine(): PIC Client State Machine - Current state: 0x44a890, Next state: 0x2
[INFO ][2019-09-08 10:01:00] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-09-08 10:01:00] stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x10
[DEBUG][2019-09-08 10:01:00] clientReadyHandler invoked
[DEBUG][2019-09-08 10:01:00] Client is ready
[INFO ][2019-09-08 10:01:00] Creating Kinesis Video Stream test_stream
[INFO ][2019-09-08 10:01:00] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-09-08 10:01:00] stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x1
[DEBUG][2019-09-08 10:01:00] stepStateMachine(): PIC Client State Machine - Current state: 0x203, Next state: 0x40
[INFO ][2019-09-08 10:01:01] writeHeaderCallback(): RequestId: 848063e1-7866-471d-9d47-a7f3fd58ee6f
[DEBUG][2019-09-08 10:01:01] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.567832999842E9,"DataRetentionInHours":1,"DeviceName":null,"KmsKeyId":"arn:aws:kms:ap-northeast-1:349125139315:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:ap-northeast-1:349125139315:stream/test_stream/1567832999842","StreamName":"test_stream","Version":"44igTCDGbbaq5BttrqDq"}}
[INFO ][2019-09-08 10:01:01] describeStreamResultEvent(): Describe stream result event.
[WARN ][2019-09-08 10:01:01] describeStreamResult(): Retention period returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[WARN ][2019-09-08 10:01:01] describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[DEBUG][2019-09-08 10:01:01] stepStateMachine(): PIC Stream State Machine - Current state: 0x6d968504, Next state: 0x2
[DEBUG][2019-09-08 10:01:01] stepStateMachine(): PIC Client State Machine - Current state: 0x0, Next state: 0x40
[INFO ][2019-09-08 10:01:02] writeHeaderCallback(): RequestId: 2d855e33-b61a-4c81-898c-d71ef240957f
[DEBUG][2019-09-08 10:01:02] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-96e4f3e9.kinesisvideo.ap-northeast-1.amazonaws.com"}
[INFO ][2019-09-08 10:01:02] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-09-08 10:01:02] stepStateMachine(): PIC Stream State Machine - Current state: 0x76826b24, Next state: 0x20
[DEBUG][2019-09-08 10:01:02] stepStateMachine(): PIC Client State Machine - Current state: 0x95, Next state: 0x40
[DEBUG][2019-09-08 10:01:02] getStreamingTokenHandler invoked
[DEBUG][2019-09-08 10:01:02] Refreshing credentials. Force refreshing: 1 Now time is: 1567936862271553582 Expiration: 1567939260
[INFO ][2019-09-08 10:01:02] New credentials expiration is 1567939262
[INFO ][2019-09-08 10:01:02] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-09-08 10:01:02] stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x10
[DEBUG][2019-09-08 10:01:02] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 4522245
[DEBUG][2019-09-08 10:01:02] streamReadyHandler invoked
[DEBUG][2019-09-08 10:01:02] Stream is ready
[INFO ][2019-09-08 10:01:02] Streaming from live source
[DEBUG][2019-09-08 10:01:02] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
Unable to set the source to ready state.
[INFO ][2019-09-08 10:01:02] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 0000000000450105.
[DEBUG][2019-09-08 10:01:02] defaultStreamClosedCallback(): Reported streamClosed callback for stream handle 4522245. Upload handle 18446744073709551615
[DEBUG][2019-09-08 10:01:02] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ][2019-09-08 10:01:02] Freeing Kinesis Video Stream test_stream
[INFO ][2019-09-08 10:01:02] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-09-08 10:01:02] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 4522245
[INFO ][2019-09-08 10:01:02] Freeing Kinesis Video Stream test_stream
[INFO ][2019-09-08 10:01:02] freeKinesisVideoClient(): Freeing Kinesis Video Client
log4cplus:ERROR Failed to open file ./log/kvs.log

MushMal, 2019-09-08:
Seems like the media pipeline couldn't be setup properly

[DEBUG][2019-09-08 10:01:02] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512
Unable to set the source to ready state.

Please provide more info on your usage such as what version of the SDK, what asset, what platform, details.

If you are using GStreamer plugin then please use GST_DEBUG=4 env var to enable verbose logging

biranchi2018, 2019-09-11:
Fixed it by resetting the Raspberry pi cam using "sudo modprobe bcm2835-v4l2" command.

closed 2019-09-11 by biranchi2018

tailer954, 2020-11-04:
I have similar problem in raspberry pi 4 with a pi camera v2. 

./kinesis_video_gstreamer_sample_app ExampleVideoStream
-----------------------------------------------------------------------------------------------------------------------------------------
[DEBUG] [04-11-2020 05:43:33:232.554 GMT] describeStreamCurlHandler(): DescribeStream API response: 
[INFO ] [04-11-2020 05:43:33:232.666 GMT] describeStreamResultEvent(): Describe stream result event.
[ERROR] [04-11-2020 05:43:35:067.018 GMT] createKinesisVideoStreamSync(): Failed to create Kinesis Video Stream - timed out.
[INFO ] [04-11-2020 05:43:35:067.197 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG] [04-11-2020 05:43:36:667.955 GMT] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 33775149
[ERROR] [04-11-2020 05:43:36:668.341 GMT]  Unable to create Kinesis Video stream. ExampleVideoStream Error status: 0xf
[INFO ] [04-11-2020 05:43:36:668.666 GMT] Freeing Kinesis Video Stream ExampleVideoStream
[INFO ] [04-11-2020 05:43:36:668.748 GMT] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[ERROR] [04-11-2020 05:43:36:668.889 GMT] Failed to initialize kinesis video with an exception:  Unable to create Kinesis Video stream. ExampleVideoStream Error status: 0xf
[INFO ] [04-11-2020 05:43:36:668.989 GMT] freeKinesisVideoClient(): Freeing Kinesis Video Client
[WARN ] [04-11-2020 05:43:36:669.059 GMT] curlApiCallbacksShutdownActiveRequests(): pActiveRequests hashtable is empty
[DEBUG] [04-11-2020 05:43:37:069.723 GMT] freeKinesisVideoClientInternal(): Total allocated memory 0
[WARN ] [04-11-2020 05:43:37:069.808 GMT] curlApiCallbacksShutdown(): curlApiCallbacksShutdown called when already in progress of shutting down
log4cplus:ERROR Failed to open file ./log/kvs.log

When I type in the command, "ls -al /dev/video*", I get eight devices. It seems to be strange. I use "sudo modprobe bcm2835-v4l2" to reset. However, it doesn't work.
-----------------------------------------------------------------------------------------------------------------------------------------
crw-rw----+ 1 root video 81, 0 11月  4 11:51 /dev/video0
crw-rw----+ 1 root video 81, 5 11月  4 11:51 /dev/video10
crw-rw----+ 1 root video 81, 6 11月  4 11:51 /dev/video11
crw-rw----+ 1 root video 81, 7 11月  4 11:51 /dev/video12
crw-rw----+ 1 root video 81, 1 11月  4 11:51 /dev/video13
crw-rw----+ 1 root video 81, 2 11月  4 11:51 /dev/video14
crw-rw----+ 1 root video 81, 3 11月  4 11:51 /dev/video15
crw-rw----+ 1 root video 81, 4 11月  4 11:51 /dev/video16