awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #264: Sending Frames to Kinesis Video Stream

arnamoy10, 2019-09-09:
Hello;

I am trying to send the frames received from a HikVision camera to kinesis video stream.  I am successfully able to prepare the frames received from the camera into buffers (of type `uint8_t`) that can be written to both a JPEG image and MP4 file using `fwrite`

Could you please tell me how should I prepare such buffers into a frame that can be sent to kinesis video stream?  Any pointers will be much appreciated.

Thank you so much.

unicornss, 2019-09-09:
Hi @arnamoy10 ,

Please refer to the sample https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c#L124 for sending frames to KVS.

unicornss, 2019-10-31:
Closing this issue as it has been answered. Please feel free to reopen if needed. Thanks

closed 2019-10-31 by unicornss