awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #266: Is there any way to implement Kinesis Video Stream in java without using KinesisVideoProducerJNI.dll file?

prakashraj007, 2019-09-13:


MushMal, 2019-09-13:
@prakashraj007 the Kinesis Video Streams Service has public service APIs described here https://docs.aws.amazon.com/en_pv/kinesisvideostreams/latest/dg/API_Reference.html

Once can simply use the service APIs and send the KVS specific MKV packaged encoded media to the service. 

The SDK makes this simpler.

prakashraj007, 2019-09-13:
@MushMal. Thanks for your reply. I could not find any SDK for java. Can you please share the SDK link to download.
Thanks in advance

MushMal, 2019-09-13:
@prakashraj007 we have https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-java which is Java code that still has JNI interface to this source. Alternatively, as I mentioned, you could use the service API directly using any client, including AWS SDK.

prakashraj007, 2019-09-13:
sorry, i can't understand what you mean "Service API".

MushMal, 2019-09-13:
Kinesis Video Stream is a service with well documented service-side API. A developer who has her own HTTP client, can use it directly against the backend services.

MushMal, 2019-09-17:
Closing for now. Please open a separate issue if needed

closed 2019-09-17 by MushMal