awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #267: Build fails for Windows 10 32 bit platform. Following build rules exactly.

Ulyssez, 2019-09-25:
Following rules you've provided exactly. Build fails with errors below ... Could you please advise what am I doing wrong? Building 32 bit version, it uses c:\program files (x86)\windows kits\10\include\10.0.17134.0\

     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(320): error C2371: 'PSSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk
   -cpp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(319): error C2371: 'SIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-c
   pp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(319): error C2371: 'PSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
   cpp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(320): error C2371: 'SSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
   cpp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(320): error C2371: 'PSSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk
   -cpp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(319): error C2371: 'SIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-c
   pp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(319): error C2371: 'PSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
   cpp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(320): error C2371: 'SSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk-
   cpp\kinesis-video-native-build\pic_test.vcxproj]
     c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-pic\src\common\include\com\amazonaws\kinesis\video\common\c
   ommondefs.h(320): error C2371: 'PSSIZE_T': redefinition; different basic types [C:\amazon-kinesis-video-streams-producer-sdk
   -cpp\kinesis-video-native-build\pic_test.vcxproj]

249 Warning(s)
176 Error(s)
Time Elapsed 00:00:48.24
Build error occurred! Please delete related directories in downloads\ to start a clean build.

Ulyssez, 2019-09-25:
Commented out typedefs below in CommonDefs.h 

#ifndef _SIZE_T_DEFINED_IN_COMMON
    #if defined (__MINGW32__)
        typedef ULONG_PTR SIZE_T, *PSIZE_T;
        typedef LONG_PTR SSIZE_T, *PSSIZE_T;
    #else
 //       typedef UINT_PTR SIZE_T, *PSIZE_T;
 //       typedef INT_PTR SSIZE_T, *PSSIZE_T;
    #endif
    #define _SIZE_T_DEFINED_IN_COMMON
#endif

build continued, now I face completely different set of errors though much less:

Severity	Code	Description	Project	File	Line	Suppression State
Error	LNK1120	1 unresolved externals	cproducer	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\cproducer.dll	1	
Error	LNK1120	1 unresolved externals	pic_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\pic_test.exe	1	
Error	LNK1120	2 unresolved externals	cproducer_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\cproducer_test.exe	1	
Error	C1083	Cannot open include file: 'jni.h': No such file or directory	KinesisVideoProducerJNI	c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer-jni\src\include\com\amazonaws\kinesis\video\producer\jni\JNICommon.h	8	
Error	C1083	Cannot open include file: 'jni.h': No such file or directory	KinesisVideoProducerJNI	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer-jni\src\include\com\amazonaws\kinesis\video\producer\jni\com_amazonaws_kinesisvideo_internal_producer_jni_NativeKinesisVideoProducerJni.h	1	
Error	C1083	Cannot open include file: 'jni.h': No such file or directory	KinesisVideoProducerJNI	c:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-producer-jni\src\include\com\amazonaws\kinesis\video\producer\jni\JNICommon.h	8	
Error	LNK1257	code generation failed	kinesis_video_gstreamer_sample_app	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\LINK	1	
Error	LNK1257	code generation failed	gstkvssink	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\LINK	1	
Error	LNK1257	code generation failed	producer_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\LINK	1	
Error	C1007	unrecognized flag '-Ot' in 'p2'	kinesis_video_gstreamer_sample_app	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\LINK	1	
Error	C1007	unrecognized flag '-Ot' in 'p2'	gstkvssink	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\LINK	1	
Error	C1007	unrecognized flag '-Ot' in 'p2'	producer_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\LINK	1	
Error	LNK2019	unresolved external symbol __except_handler4_common referenced in function __except_handler4	cproducer	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\MSVCRT.lib(_chandler4gs_.obj)	1	
Error	LNK2019	unresolved external symbol __except_handler4_common referenced in function __except_handler4	cproducer_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\MSVCRT.lib(_chandler4gs_.obj)	1	
Error	LNK2019	unresolved external symbol ___std_reverse_trivially_swappable_4 referenced in function "class std::vector<enum testing::internal::edit_distance::EditType,class std::allocator<enum testing::internal::edit_distance::EditType> > __cdecl testing::internal::edit_distance::CalculateOptimalEdits(class std::vector<unsigned int,class std::allocator<unsigned int> > const &,class std::vector<unsigned int,class std::allocator<unsigned int> > const &)" (?CalculateOptimalEdits@edit_distance@internal@testing@@YA?AV?$vector@W4EditType@edit_distance@internal@testing@@V?$allocator@W4EditType@edit_distance@internal@testing@@@std@@@std@@ABV?$vector@IV?$allocator@I@std@@@5@0@Z)	cproducer_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\gtest.lib(gtest-all.obj)	1	
Error	LNK2019	unresolved external symbol ___std_reverse_trivially_swappable_4 referenced in function "class std::vector<enum testing::internal::edit_distance::EditType,class std::allocator<enum testing::internal::edit_distance::EditType> > __cdecl testing::internal::edit_distance::CalculateOptimalEdits(class std::vector<unsigned int,class std::allocator<unsigned int> > const &,class std::vector<unsigned int,class std::allocator<unsigned int> > const &)" (?CalculateOptimalEdits@edit_distance@internal@testing@@YA?AV?$vector@W4EditType@edit_distance@internal@testing@@V?$allocator@W4EditType@edit_distance@internal@testing@@@std@@@std@@ABV?$vector@IV?$allocator@I@std@@@5@0@Z)	pic_test	C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\gtest.lib(gtest-all.obj)	1

Ulyssez, 2019-09-25:
Missing jni - should I also run windows-java-install.bat prior to windows-install.bat?

MushMal, 2019-09-25:
@Ulyssez seems you are trying to use msvc compiler tools with CMD/Powershell. The instructions are listed here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md

Did these fail to install the prerequisites and then build?

zhiyua-git, 2019-09-25:
Hi @Ulyssez , you will need to install JDK and add JAVA_HOME to your environment variable to find the jni.h. 

You don't need java to run windows-install.bat though. Are you hitting this when you run windows-install.bat?

Ulyssez, 2019-09-26:
Thanks zhiyua-git.  Yes, using "windows-install.bat 32" I hit the first batch of errors related with PSIZE_T': redefinition; and 'SSIZE_T': redefinition; - THis is the problem I would like to get rid of first.
I manually commented out definitions in CommonDefs.h, but I guess these errors should not happen in the first place.
Then build resumed but did not complete requiring jhi.h ... Are you saying that windows-install.bat ld automatically install JDK if was not interrupted by "redefinition" errors?

Ulyssez, 2019-09-26:
@MushMal - Thanks for comment - I am using instructions at location pointed by your link exactly. It fails using these instructions.

MushMal, 2019-09-26:
@Ulyssez can you clarify your scenario. Are you using clean GitHub repository on Windows 32 bit with no modifications, running windows-install.bat 32 from the native build directory and you are hitting redefinition of SIZE_T types? After commenting it you are getting missing jni.h?

Ulyssez, 2019-09-26:
@MushMal: I am on Windows 10 64 bit platform. Application where I'm planning to use Kinesis video streaming is 32 bit. I need 32 bit version of Kinesis producer SDK.
In this particular case, I have followed instructions exactly - cloning repository, installing vs tools, rebooting laptop, running windows-install.bat 32. It gives redefinition error after the step installing GStreamer, while building pic-test. 
Then I'm opening pic-test.vcxproj in Visual Studio 2015, commenting out mentioned lines. Rebuilding pic-test solution and getting missing jni.h and other linker errors.

Ulyssez, 2019-09-26:
Below are errors when I'm running windows-java-install.bat
Build started 9/26/2019 11:09:15 PM.
     1>Project "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\KinesisVideoProducerJNI.vcx
       proj" on node 1 (default targets).
     1>Project "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\KinesisVideoProducerJNI.vcx
       proj" (1) is building "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\ZERO_CHECK.vc
       xproj" (2) on node 1 (default targets).
     2>C:\BuildTools\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(65,5): error MSB8020: The build tools for
       Visual Studio 2010 (Platform Toolset = 'v100') cannot be found. To build using the v100 build tools, please inst
       all Visual Studio 2010 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by select
       ing the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\amazon-kinesis-vid
       eo-streams-producer-sdk-cpp\kinesis-video-native-build\ZERO_CHECK.vcxproj]
     2>Done Building Project "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\ZERO_CHECK.vc
       xproj" (default targets) -- FAILED.
     1>Done Building Project "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\KinesisVideoP
       roducerJNI.vcxproj" (default targets) -- FAILED.

Build FAILED.

       "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\KinesisVideoProducerJNI.vcxproj" (d
       efault target) (1) ->
       "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\ZERO_CHECK.vcxproj" (default target
       ) (2) ->
       (PlatformPrepareForBuild target) ->
         C:\BuildTools\Common7\IDE\VC\VCTargets\Microsoft.Cpp.Platform.targets(65,5): error MSB8020: The build tools fo
       r Visual Studio 2010 (Platform Toolset = 'v100') cannot be found. To build using the v100 build tools, please in
       stall Visual Studio 2010 build tools.  Alternatively, you may upgrade to the current Visual Studio tools by sele
       cting the Project menu or right-click the solution, and then selecting "Retarget solution". [C:\amazon-kinesis-v
       ideo-streams-producer-sdk-cpp\kinesis-video-native-build\ZERO_CHECK.vcxproj]

    0 Warning(s)
    1 Error(s)

Time Elapsed 00:00:00.89
Build error occurred

Ulyssez, 2019-09-26:
I got pic_test compiled ... Now stuck at linker error on cproducer ... 
Adding more info on my environment - I have installed VS 2015 and VS build tools installed by vs-buildtools-install.bat. 
Should I uninstall visual studio 2015 and install vs2017 ?

  1>LINK : warning LNK4098: defaultlib 'MSVCRT' conflicts with use of other libs; use /NODEFAULTLIB:library [C:\amazon-kinesis-v
       ideo-streams-producer-sdk-cpp\kinesis-video-native-build\cproducer.vcxproj]
     1>MSVCRT.lib(chandler4gs.obj) : error LNK2019: unresolved external symbol __except_handler4_common referenced in function __ex
       cept_handler4 [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\cproducer.vcxproj]
     1>C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\cproducer.dll : fatal error LNK1120: 1 u
       nresolved externals [C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\cproducer.vcxproj]
     1>Done Building Project "C:\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\cproducer.vcxproj" (defau
       lt targets) -- FAILED.

Build FAILED.

zhiyua-git, 2019-09-26:
Hi @Ulyssez ,

Can you try to delete the *.sln and *.vcxproj files in kinesis-video-native-build and rerun the windows-install.bat again? Visual Studio doesn't work well with the generated project file from CMakeLists.txt. 

If you want to build the project in Visual Studio, I would recommend deleting those files and import the CMakeLists.txt in kinesis-video-native-build. You would need to resolve issues like deleting JNI part from the CMakeLists.txt and mark dependency on CProducer to make project links on the symbols.

kamocio, 2019-09-26:
I'm having a similar issue with windows-install.bat 32. Win10, MSVC 2015 and 2019 installed on that box. Followed instructions. Gives me a bunch of errors like:

c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: negative subscript [C:\Dev\amazon-kinesis-video-streams-producer-sdk-cpp\ki
       nesis-video-native-build\pic_test.vcxproj]

This build is so complicated ... Thing is - I dont actually need to build this at all, just need the binaries to run the samples/develop against. Is it possible to get pre-built bins from somewhere?

zhiyua-git, 2019-09-27:
Hi @Ulyssez @kamocio ,

We are aware of the regression for 32 bit build on msvc, will get your posted when it got fixed. Sorry for the inconvenience.

kamocio, 2019-09-27:
Hi @zhiyua-git,

Thanks for heads up. In the meantime, is it possible to get the pre-built binaries from "shomewhere"? We are trying to evaluate this for a very date-sensitive project. Thanks!

zhiyua-git, 2019-09-28:
Hi @kamocio ,

If you don't need C version producer, for now you can do
git clean -dfx
git reset --hard a8bbb9be541eafe57d93ccadef65ab3583cc6cc0
Then run windows-install.bat 32 with administrator permission on CMD. We don't have pre-built 32 bit windows binary currently.

kamocio, 2019-09-28:
Thanks, @zhiyua-git. well, we are trying to stream from a 32 bit C++ app (built with VC2015), so I guess we need the producer?

Anyway, after the reset I get a bunch of those errors, is it safe to just define this const for all offending *vcxproj files to make it go?

c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2482): error C2338: Windows headers require the default packing option. Changing this can lead
       to memory corruption. This diagnostic can be disabled by building with WINDOWS_IGNORE_PACKING_MISMATCH defined. [C:\Dev\amazon-kinesis-video-streams-producer-sdk-cpp\ki
       nesis-video-native-build\producer.vcxproj]

zhiyua-git, 2019-09-28:
Hi @Ulyssez @kamocio ,

The issue was caused by mixing up static and dynamic runtime libraries from windows.

Please replace [this file](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-native-build/projectFiles/jsmn/CMakeLists.txt) with [CMakeLists.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3664973/CMakeLists.txt). Then delete kinesis-video-native-build/downloads/jsmn-1.0.0 and rerun windows-install.bat 32. This should work.

kamocio, 2019-09-28:
Great! Thanks so much!

I'm now down to half a dozen of those:

1>c:\program files (x86)\windows kits\10\include\10.0.18362.0\um\winnt.h(2487): error C2118: negative subscript [C
       :\Dev\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\pic_test.vcxproj]

Was your change intended to address that too, or is it a different issue? My understanding it happens when there is a mix-up between 32 and 64 bits, though not sure what I do wrong, i'm just running windows-install.bat 32. I have VS2015 and VS 2019 installed on this box, could that be an issue, or is this build entirely self-contained?

zhiyua-git, 2019-09-30:
@kamocio ,

It is not clear to me how it failed in that way on your PC. I tried the same fix for building 32 bits on docker and it worked. 

You can get the 32-bits binary from docker container if you need.

Create a directory with short path in your disk (I used C:\docker). Then do following and you will get the 32-bits binaries built in your docker container.
1. cd C:\docker
2. copy /path/to/Dockerfile.txt Dockerfile
3. git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp.git
4. cd amazon-kinesis-video-streams-producer-sdk-cpp
5. git apply /path/to/fixDiffForWin32.txt
6. copy amazon-kinesis-video-streams-producer-sdk-cpp\docker_native_scripts\windows\start_rtsp_in_docker.bat ..\
7. docker build .

[fixDiffForWin32.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3673686/fixDiffForWin32.txt)

[Dockerfile.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3673688/Dockerfile.txt)

Ulyssez, 2019-10-03:
@zhiyua-git I have followed your  advice below and seems got the build, thanks much.

git clean -dfx
git reset --hard a8bbb9b
Then run windows-install.bat 32 with administrator permission on CMD. We don't have pre-built 32 bit windows binary currently.

Could you please let me know what's special about this change-list? Still could not figure out what exactly helped...

zhiyua-git, 2019-10-03:
Hi @Ulyssez,

One issue was the redefinition of PSIZE_T, the other was jsmn by default it links with dynamic windows runtime library without the changes which mixes dynamic and static runtime library and failed the build at linking time.

Ulyssez, 2019-10-07:
Thanks for support. I'm good for now.

closed 2019-10-07 by Ulyssez