awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #268: Audi and video from IP camera to KVS [own java application with GStreamer kvssink]

Th0ru5, 2019-10-01:
Hello,
I'm trying to create Java application with https://github.com/gstreamer-java/gst1-java-core,
to send audio and video from RTSP source (IP camera) to KVS.
But i'm getting this in my console:
`[WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 37884632 presentation timestamp: 37884632 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 39591084 presentation timestamp: 39591084 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 39804418 presentation timestamp: 39804418 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 39824184 presentation timestamp: 40624116 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40017751 presentation timestamp: 40017751 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40224150 presentation timestamp: 42223960 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40231084 presentation timestamp: 40231084 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 40657624 presentation timestamp: 40657624 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42364169 presentation timestamp: 42364169 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42577385 presentation timestamp: 42577385 [WARN ][2019-09-30 23:44:59] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42623919 presentation timestamp: 42623919 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 42790718 presentation timestamp: 42790718 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 43857272 presentation timestamp: 43857272 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44070606 presentation timestamp: 44070606 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44223761 presentation timestamp: 45023684 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44283939 presentation timestamp: 44283939 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44497272 presentation timestamp: 44497272 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 44710497 presentation timestamp: 44710497 [WARN ][2019-09-30 23:45:00] putKinesisVideoFrame(): Failed to submit frame to Kinesis Video client. status: 0x30000005 decoding timestamp: 45137164 presentation timestamp: 45137164 [DEBUG][2019-09-30 23:45:00] postReadCallback(): Wrote 9270 bytes to Kinesis Video. Upload stream handle: 0 [DEBUG][2019-09-30 23:45:00] postReadCallback(): Pausing CURL read for upload handle: 0`

Once again i have problem with Audio and Video, when i'm trying to send only Video all is working fine.

In Your documentation on https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html is nothing about audio.

The pipeline for KVS looks like this:
`appsrc name=videoAppSrc "+ "! rawvideoparse name=videoparser use-sink-caps=false format=8 "+ "! videoconvert ! video/x-raw,format=I420 "+ "! x264enc ! h264parse ! video/x-h264,stream-format=avc,alignment=au "+ "! kvssink name=kvs stream-name=new_straming_system storage-size=512 access-key=xxx secret-key=xxx aws-region=eu-central-1 "+ "appsrc name=audioAppSrc "+ "! rawaudioparse use-sink-caps=true "+ "! audioconvert ! avenc_aac ! kvs.`
Caps for audio looks like this:
`audio/x-raw, format=(string)F32LE, layout=(string)interleaved, rate=(int)48000, channels=(int)1`

I was trying to implement simple function to check if DTS(n) < DTS(n-1) + Duration(n-1) and i was eliminating those audio samples but this didn't help.

When I'm trying to send the stream with video and audio to YouTube from the same application but using different pipeline all is working.
I was trying to deal with it alone but after 6 hours i give up.

chehefen, 2019-10-01:
can you export DEBUG_DUMP_FRAME_INFO=TRUE and send us the log? we want to know what your audio and video frame timestamps look like.

Th0ru5, 2019-10-02:
The logs you are asking are here:
[kvs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3680100/kvs.log)

chehefen, 2019-10-04:
can you print all the values in this line: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/75087f5a90a02a47191c9278cfec329e09535e98/kinesis-video-pic/src/view/src/ContentView.c#L460? The error shouldnt be happening based on the logs. There may be some data corruption.

Th0ru5, 2019-10-07:
I added those line above 460
`if(pHead){
		printf("pHead->timestamp: %d pHead->duration: %d  timestamp:  %d \n", pHead->timestamp, pHead->duration, timestamp);
	}`
and here is the log file:
[kvs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3696034/kvs.log)

chehefen, 2019-10-07:
can you try comment out this line: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/StreamDefinition.cpp#L102? Thanks.

Th0ru5, 2019-10-08:
I did what you wanted.
It almost works.
Now i can see some frames after one minute of streaming and GStreamer is shutting down with this communicat:
`GStreamer has encountered a general stream error.`
Logs from kvssink:
[kvs.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3701563/kvs.log)

MushMal, 2019-10-08:
@Th0ru5 there is a number of issues in the log:

* MAX_FRAGMENT_DURATION_REACHED - this indicates that you have over 10 seconds long GoP sizes
* FRAMES_MISSING_FOR_TRACK - This is an indication that your stream (which is multi-track stream) has no frames of a particular type of track within the fragment. This is the current limitation of KVS which requires every fragment to have at least one frame of each track

Can you check your configuration? Start by simply streaming the video only.

wlinInspire, 2020-12-16:
@Th0ru5 adding `key-int-max=60` to `x264enc`? if yes, seems this has nothing to do with the audio, right?