awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #269: No-dependencies gstreamer plugin

felipealmeida, 2019-10-02:
Hello,

I'd like to use the gstreamer plugin for a use-case in security. However, the amount of dependencies it pulls is just not possible to use it. I have a S3 plugin I created for image upload which is written in C and should get accepted to gst-plugins-bad. It would be _greatly_ benefical if kinesis plug-in could be upstream to gst-plugins-bad would really help users.

MushMal, 2019-10-02:
@felipealmeida thanks for your suggestion. We are constantly looking for ways to make it easier for the customers to adapt and integrate our solutions. Publishing the plugin into the standard plugin bundle is something that we are looking at.

In the meanwhile, the KVS gstreamer plugin itself doesn't really have much dependencies on the gstreamer - only on the KVS SDK. The install script builds a number of components but the kvs plugin itself doesn't rely on other plugins.

felipealmeida, 2019-10-02:
Thanks. What I mean by dependencies is on the SDK, which brings a lot of things not very interesting for native code. Unfortunately that means duplicating a lot of code, but that's why people choose one language over the others, their runtime matters. I own a company specialized in open source development, and these are the requirements we get.

MushMal, 2019-10-02:
@felipealmeida as I mentioned, KVS plugin is a simple wrapper over the C++ SDK. The SDK itself for C++ layer has a dependency on log4cplus. The C++ layer sits on top of C SDK which has dependency on JSMN, libCurl and OpenSSL (with libCrypto). If your solution has those dependencies then you don't need to bring them in but simply link against them.

chehefen, 2019-10-02:
@felipealmeida if you already have all the dependencies, you can try use the min-install-script which will not build everything from scratch but instead search for the libraries in the system and use it.

MushMal, 2019-12-04:
Resolving for now. We will look into making kvs gst plugin utilizing C producer to lessen the dependencies

closed 2019-12-04 by MushMal