awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #270: kinesis_video_gstreamer_audio_video_sample_app gets stuck

mistersingh179, 2019-10-03:
I am running this command:

`GST_DEBUG=4 AWS_ACCESS_KEY_ID=xx AWS_SECRET_ACCESS_KEY=yy AWS_KVS_AUDIO_DEVICE=hw:1,0 ./kinesis_video_gstreamer_audio_video_sample_app test`

And it gets stuck with the last message being:
> Setting video_encode state from Idle to Executing

I do see these 2 warn messages
> audiosrc gstaudiosrc.c:197:audioringbuffer_thread_func:<audiosrc> failed to set thread priority
>audio-resampler audio-resampler.c:274:convert_taps_gint16_c: can't find exact taps

Also I can confirm that running the sample app which records just video, works fine - `kinesis_video_gstreamer_sample_app`

I can also confirm that i do have the audio recording device at hw:1,0 by doing
```
arecord -l
**** List of CAPTURE Hardware Devices ****
card 1: Device [USB PnP Audio Device], device 0: USB Audio [USB Audio]
  Subdevices: 0/1
  Subdevice #0: subdevice #0
```

I can also confirm that the microphone works by recording and playing the audio with command:
```
arecord -D plughw:1,0 -d 3 test.wav
```

Attached are the logs.
[kinesis_audio_video_sample_app_logs.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3687876/kinesis_audio_video_sample_app_logs.txt)

chehefen, 2019-10-15:
can you try export AWS_KVS_VIDEO_DEVICE="/dev/video0" too?

mistersingh179, 2019-10-15:
@chehefen same results. it gets stuck at 

> Setting video_encode state from Idle to Executing

You have asked me to set the video device variable. But i think the issue is with the audio device because:
1. video sample app works fine.
2. gst-device-monitor-1.0 does not see any audio device, even though arecord -l shows it
3. logs have audio warnings

warn messages from the logs:

> 0:00:01.441722980  1308  0x1fcacf0 WARN         audio-resampler audio-resampler.c:274:convert_taps_gint16_c: can't find exact taps
> 0:00:01.438756888  1308 0xa0602830 WARN                audiosrc gstaudiosrc.c:197:audioringbuffer_thread_func:<audiosrc> failed to set thread priority

chehefen, 2019-10-15:
can you try running this command?
```
gst-launch-1.0 -v v4l2src ! videoconvert ! video/x-raw,format=I420 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! queue ! matroskamux name=mux ! fakesink name=sink silent=FALSE alsasrc device=hw:1,0 ! avenc_aac ! aacparse ! audio/mpeg,stream-format=raw ! queue ! mux.
```

mistersingh179, 2019-10-16:
@chehefen i get this:

```
Setting pipeline to PAUSED ...
Pipeline is live and does not need PREROLL ...
ERROR: from element /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0: Internal data stream error.
Additional debug info:
gstbasesrc.c(3055): gst_base_src_loop (): /GstPipeline:pipeline0/GstAlsaSrc:alsasrc0:
streaming stopped, reason not-negotiated (-4)
ERROR: pipeline doesn't want to preroll.
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline ...
```

mistersingh179, 2019-11-15:
@chehefen do you have any ideas/hints I could pursue?

mistersingh179, 2019-11-18:
I was able to resolve this issue by replacing the microphone/camera setup I had with a Logitech C920e Webcam. This indicates that perhaps there is a compatibility issue with the old microphone I was using. 

Hopefully this helps someone else in the future.

closed 2019-11-18 by mistersingh179