awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #271: Cmake error on raspberry pi 3b+ environment

hsh07260, 2019-10-15:
when I execute " ./install-script " in /kinesis-video-native-build/, there was a problem,
following error code is problem.
how to Cmake successfully?

Setting up for KVS Producer SDK installation
======= make sure the following libraries have been installed =======
* pkg-config
* libssl-dev
* cmake
* libgstreamer1.0-dev
* libgstreamer-plugins-base1.0-dev
* gstreamer1.0-plugins-base-apps
* gstreamer1.0-plugins-bad
* gstreamer1.0-plugins-good
* gstreamer1.0-plugins-ugly
* gstreamer1.0-omx (if on raspberry pi)
* gstreamer1.0-tools
* libcurl4-openssl-dev
* liblog4cplus-1.1-9
* liblog4cplus-dev
Checking jsmn at /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads/local/lib/libjsmn.a
Clean up artifacts
done Clean up artifacts
-- The C compiler identification is GNU 6.3.0
-- The CXX compiler identification is GNU 6.3.0
-- Check for working C compiler: /usr/bin/cc
-- Check for working C compiler: /usr/bin/cc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /usr/bin/c++
-- Check for working CXX compiler: /usr/bin/c++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- System name is Linux
-- Kinesis Video PIC path is ../kinesis-video-pic
-- Kinesis Video Producer path is ../kinesis-video-producer
-- Kinesis Video JNI path is ../kinesis-video-producer-jni
-- Kinesis Video Gstreamer demp app path is ../kinesis-video-gst-demo
-- Kinesis Video Open Source path is /home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/downloads
-- Configuration Unix Makefiles
-- Bitness 32 bits
-- Looking for pthread.h
-- Looking for pthread.h - found
-- Looking for pthread_create
-- Looking for pthread_create - not found
-- Check if compiler accepts -pthread
-- Check if compiler accepts -pthread - yes
-- Found Threads: TRUE  
-- exe libraries Threads::Threads;dl;rt
-- Found PkgConfig: /usr/bin/pkg-config (found version "0.29") 
-- Could NOT find Log4cplus (missing:  LOG4CPLUS_LIBRARIES LOG4CPLUS_INCLUDE_DIR) 
Log4cplus is not found. Only c producer will be built. 
CMake Error at /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:138 (message):
  Could NOT find OpenSSL, try to set the path to OpenSSL root folder in the
  system variable OPENSSL_ROOT_DIR (missing: OPENSSL_LIBRARIES
  OPENSSL_INCLUDE_DIR)
Call Stack (most recent call first):
  /usr/share/cmake-3.7/Modules/FindPackageHandleStandardArgs.cmake:378 (_FPHSA_FAILURE_MESSAGE)
  /usr/share/cmake-3.7/Modules/FindOpenSSL.cmake:385 (find_package_handle_standard_args)
  CMakeLists.txt:155 (find_package)

— Configuring incomplete, errors occurred!
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeOutput.log".
See also "/home/pi/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/CMakeFiles/CMakeError.log".

chehefen, 2019-10-15:
can you follow the instructions here: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-steps-for-ubuntu-17x-and-raspbian-stretch-using-apt-get to get all the dependencies.

unicornss, 2019-10-31:
Closing the issue for now as it has been answered. Please reopen if needed.

closed 2019-10-31 by unicornss