awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #272: Timecodes are not monotonically increasing?

TheDeathV2, 2019-10-15:
Hello,
I am a new user of AWS, and I am trying to make a personal project using Raspberry Pi and cloud. In this project I stream my Pi Camera to Kinesis Video Stream.

At the begin of my project, I had this error status : 0x52000060

Which means "Timecodes are not monotonically increasing", but after a while of trying to debug this sdk, it had been fix without doing anything ...

And now this error is back, and I don't know how to handle it.

Thanks for your help, or if you have anything that can help me, I'll take it :)
Best regards.

chehefen, 2019-10-15:
0x52000060 means the timestamp of your key frames are not monotonically increasing. you can export DEBUG_DUMP_FRAME_INFO=TRUE to see the frame timestamps.

MushMal, 2019-12-04:
@TheDeathV2 do you still have this issue?

disa6302, 2020-02-25:
Closing this issue for now. Feel free to reopen/create a new issue in the future.

closed 2020-02-25 by disa6302