awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #273: Save live video fragments to S3

poojakittur, 2019-10-15:
I am trying to stream a live video from an ip camera to AWS kinesis video stream. Is there any way by which I could save video fragments between two timestamps to S3?

zhiyua-git, 2019-10-15:
Hi @poojakittur ,

You can refer to the sample here to retrieve the fragments and then upload file to S3. 
https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/master/src/test/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExampleTest.java#L49

You might need to modify the [sample](https://github.com/aws/amazon-kinesis-video-streams-parser-library/blob/57063da12cbfbfa95abbf04bb8d472fe559e7497/src/main/java/com/amazonaws/kinesisvideo/parser/examples/KinesisVideoExample.java#L110) to call GetMediaForFragmentList and provide proper Selector based on your fragment between two timestamps requirement.

unicornss, 2019-11-05:
Closing this issue as the question has been answered. Please reopen or create a new one if support is needed.

closed 2019-11-05 by unicornss