awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #277: it is wrong to based on file path contains substring "pem"  to  determine it is a file or directory

ek9852, 2019-10-27:
In https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/source/Response.c#L88
It uses strstr to check if a file path contains 'pem' substring to determine it is a file or  directory.
This is very bad, as  the root ca cert file is not necessary with pem and it can be a path but contain pem substring.
It should be change to as follows to use stat to determine it is a file or directory :
-            if (STRSTR(pCurlRequest->certPath, CA_CERT_FILE_SUFFIX) != NULL) {
-                curl_easy_setopt(pCurlResponse->pCurl, CURLOPT_CAINFO, pCurlRequest->certPath);
-            } else {
-                curl_easy_setopt(pCurlResponse->pCurl, CURLOPT_CAPATH, pCurlRequest->certPath);
+            struct stat statbuf;
+            if (stat(pCurlRequest->certPath, &statbuf) != -1){
+                if (S_ISDIR(statbuf.st_mode)) {
+                    curl_easy_setopt(pCurlResponse->pCurl, CURLOPT_CAPATH, pCurlRequest->certPath);
+                } else {
+                    curl_easy_setopt(pCurlResponse->pCurl, CURLOPT_CAINFO, pCurlRequest->certPath);^M
+                }

MushMal, 2019-10-28:
@ek9852  thanks for opening the issue. We agree, the check for PEM file is rather rudimentary and we will fix this in the upcoming releases. In the meanwhile, we are accepting PRs :)

MushMal, 2019-12-04:
This should be fixed in the latest release

closed 2019-12-04 by MushMal

ek9852, 2019-12-04:
The latest  2.1.0 release still checking  on the extension and assume it is ending with pem.
pem file can be ending with .crt /  .cert. It is not necessary to check the extension on that. 
It will just failing for client not using .pem as the extension.  (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/source/Response.c#L181)

ek9852, 2019-12-10:
@MarkAtwood PR created: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/306

MushMal, 2019-12-11:
Sorry for the delay. The current implementation handles PEM files only - hence it checks the extension. We are planning on adding support for other types of certs too (the wiring is done in the code) but haven't got around due to lack to requests from the customers. Please provide more info whether PEM file can be obtained for your platform.

hassanctech, 2020-08-25:
You can use this to convert your CRT format files to PEM:

`openssl x509 -in cert.crt -out cert.pem`