awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #278: createDefaultCallbacksProviderWithIotCertificate root ca cert not optional

ek9852, 2019-10-27:
In createDefaultCallbacksProviderWithIotCertificate,
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/include/com/amazonaws/kinesis/video/cproducer/Include.h#L413
It mentioned that CA Cert path is "IN/OPT" which is optional.
But it is not.
See  https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/source/IotAuthCallback.c#L25, it will return STATUS_NULL_ARG if that is not set.

unicornss, 2019-10-28:
Hi @ek9852 

Thanks for notifying; we will update the documentation block as applicable to the type signature.

MushMal, 2019-10-28:
On the second thought, this should really be an OPT param. We will fix this in the upcoming releases

ek9852, 2019-10-28:
Yep, I agree.

MushMal, 2019-12-04:
Fixed! THANK YOU! The latest release v2.1.0

closed 2019-12-04 by MushMal

ek9852, 2019-12-04:
@MushMal It is not fixed. Pls see :
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/2.1.0/kinesis-video-c-producer/src/include/com/amazonaws/kinesis/video/cproducer/Include.h#L281

MushMal, 2019-12-04:
Just a NOTE: the cert path is a mandatory param (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/include/com/amazonaws/kinesis/video/cproducer/Include.h#L279) whereas the caCertPath is not (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/src/include/com/amazonaws/kinesis/video/cproducer/Include.h#L281)