awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #279: Does not support B frame h264 stream

ek9852, 2019-10-27:
The mkv generator  (https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/mkvgen/src/MkvGenerator.c#L806)
creates new cluster based on key frame.
But it wrongly assume that any frames follows it have pts greater than it :
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/mkvgen/src/MkvGenerator.c#L772
That assumption is incorrect and is a design fault, since the h264 stream is in dts order, and b frame  can be followed by I frame with the B frame has a smaller pts than the I frame.
i.e. kinesis video sdk can only supports baseline h264 stream (without B frame)

ek9852, 2019-10-28:
The h264 encoder will output DTS in sequential order (decoding order), i.e. the PTS will be out of order . So it does not help. Another approach  is to let caller to set the segment boundary instead by using a flag. Since we can restricted the number of B-frame in h264 encoder, and we need how many b-frame which PTS is less than the key frame that follows.

MushMal, 2019-10-28:
Yes indeed, we need to be careful. There are requirements imposed by MKV (which doesn't have a notion of DTS/PTS) and multiple streams and pipelines don't have that notion either. 
PTS being out of order should be OK. The main issue is start the Fragment (Cluster in MKV terms) based on a B-frame which might break multiple downstream dependencies. 
This hasn't been a serious issue as most of the streaming encoders will not produce B-frames in general (due to latency).

MushMal, 2019-12-04:
I am going to resolve this for now. It's not immediately clear how we can support initial b-frame

closed 2019-12-04 by MushMal

farleylai, 2020-04-14:
@MushMal 
Are B frames in H.264 officially supported by KVS SDK?
There are use cases we need to forward a previous encoded stream/video such the B frame thing may not be changed.
Any update would be appreciated.

MushMal, 2020-04-14:
@farleylai the SDK does support B-frames which are not the first frame in the fragment and are referencing frames from within the fragment only (due to MKV validity and streamability restrictions). However, the backend might impose further restrictions when attempting to generate MPEG-DASH or HLS sessions.

I guess the best would be to try the given content and see whether the b-frames that are in the content can break the SDK or your end-to-end