awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #281: GStreamer sink suddenly starts dropping all frames on RPI

MathiasKoch, 2019-10-28:
Hi AWS.

I am having an issue with the Gstreamer `kvssink`.

My chain on my raspberry pi 3 B+ looks like this: 
```
gst-launch-1.0 -v v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc control-rate=1 target-bitrate=10240000 ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="$BALENA_DEVICE_UUID" framerate=30 absolute-fragment-times=true retention-period=730 iot-certificate="iot-certificate,endpoint=$AWS_ENDPOINT,cert-path=$CERTSDIR/cert.pem,key-path=$CERTSDIR/privkey.pem,ca-path=$CERTSDIR/root-CA.crt,role-aliases=$ROLE_ALIAS"
```

It works great for about a day or so, and then all of a sudden it starts logging:

```
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431368660000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431369340000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431370020000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431370660000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431371340000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431372020000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431372660000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431373340000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431374020000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
[2019-10-27 23:04:59]-WARN-Reported droppedFrame callback for stream handle 21004285. Dropped frame timecode in 100ns: 431374660000
[2019-10-27 23:04:59]-WARN-viewItemRemoved(): Reporting a dropped frame/fragment.
```

And it just continues in the same manner, with no video in the stream. On the cloud side of Kinesis Video, it just reports `No Fragments Found` after the time the dropped frames began.

There is nothing notably before the the dropped frame log entries:

```
[2019-10-27 22:20:22]-INFO-putStreamResultEvent(): Put stream result event.
[2019-10-27 22:20:22]-INFO-writeHeaderCallback(): RequestId: fcda07b6-19bc-4f4e-aaea-1dd0d6f544b9
[2019-10-27 22:20:25]-INFO-getStreamData(): Handle 32 waiting for last persisted ack with ts 405791040000
[2019-10-27 22:20:25]-INFO-getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 32
[2019-10-27 22:20:25]-INFO-postReadCallback(): Reported end-of-stream for stream 9db91c1afed068db944c5b459ac9d030. Upload handle: 32
[2019-10-27 22:23:21]-INFO-writeHeaderCallback(): RequestId: 0a30fdf8-80dc-4b58-bdb4-342c6ea72140
[2019-10-27 22:23:22]-INFO-getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[2019-10-27 22:23:22]-INFO-getStreamingTokenResultEvent(): Get streaming token result event.
[2019-10-27 22:23:22]-INFO-putStreamResultEvent(): Put stream result event.
[2019-10-27 22:23:22]-INFO-writeHeaderCallback(): RequestId: d83ceb96-a76e-c250-8e0c-f08c70c9c122
[2019-10-27 22:23:26]-INFO-getStreamData(): Handle 33 waiting for last persisted ack with ts 407592400000
[2019-10-27 22:23:26]-INFO-getStreamData(): Indicating an EOS after last persisted ACK is received for stream upload handle 33
[2019-10-27 22:23:26]-INFO-postReadCallback(): Reported end-of-stream for stream 9db91c1afed068db944c5b459ac9d030. Upload handle: 33
```

I have been trying to debug this issue for quite a while, but without a way to replicate the issue, other than wait approximately a day or so, it is getting a bit tedious tbh.

I hope you guys have some tricks that i might have overlooked..?

In the interrest of full info, it is built as:
```
RUN git clone https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp /opt/amazon-kinesis-video-streams-producer-sdk-cpp
WORKDIR /opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build/
RUN ./min-install-script

ENV LD_LIBRARY_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$LD_LIBRARY_PATH
ENV GST_PLUGIN_PATH=/opt/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-native-build:$GST_PLUGIN_PATH
```

Running in a docker container on Balena Engine

unicornss, 2019-10-28:
@MathiasKoch 

>It works great for about a day or so, and then all of a sudden it starts logging: 
**WARN**- droppedFrame callback for stream handle 

It happens when network throughput (TCP) is not fast enough (or loss of connectivity) which could cause the storage buffer become full. SDK provides these callbacks to enable customization of client behaviors based on their use case.
Addl. Ref: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-callbacks.html

MathiasKoch, 2019-10-29:
@unicornss 

Hmm..
Okay. It sounds a bit strange though. I am running on a wired, rock solid office internet connection?

I have now tried to drastically reduce the framerate, to see how that affects the issue.

That aside, is there any way to automatically recover from this, while still just using the gstreamer kvssink? (not the Cpp interface)

ek9852, 2019-10-30:
i am also facing same issue with a solid rock internet connection, it works for 2 hours, then its outputs:
viewItemRemoved(): Reporting a dropped frame/fragment and never recover by itself.
It is definitely not related to bitrate, since in our case the bit rate is only 17kbps.
BTW, i am writing own application called to kinesis c++  library.

This happens randomly. And most of the time, even i disconnect internet for a while and connect back , the kinesis library can recover by itself. But not when the above case.

ek9852, 2019-10-30:
Using netstat shows that when this happens,, the https connection was dropped.
But kinesis sdk  is not noticed nor trying to do a reconnection.

MathiasKoch, 2019-10-30:
@ek9852 Your observations look just like mine. I have also tried to play around with bitrates and framerates, but can confirm that it does not look like it is related to that, nor to the storage-size assigned to the kinesis sink.

I am glad that it doesn't look like it is hardware related, nor environment (docker), as that was one of the things i have not yet tried to change out. I am also using the recoverOnError feature, but with the same result. It works if i disconnect the internet manually, it works if i disconnect the camera, and a lot of other things trying to provoke the SDK, but it does not recover from these `viewItemRemoved(): Reporting a dropped frame/fragment`..

@unicornss Any thing we can try out?

MushMal, 2019-10-30:
@ek9852 , @MathiasKoch Thanks for the detailed info.

Some background info: KVS SDK is designed in a layered fashion. At the core, there is PIC (Platform Independent Codebase) which handles most of the business logic which includes the recovery, restreaming. On top of that, we have the C-SDK which now handles the networking, adds more functionality and defaults and some behaviors like continuous streaming. C SDK utilizes libCurl for networking. 
The C++ SDK is a thin layer on top of C SDK which adds a few helper functionality and makes it OO.

The PIC itself has a retry logic with progressive fallback. If, however, after MAX_RETRY_COUNT iterations it can't recover, it will give up and return an error. In this case, the C SDK layer will run through state machinery to "continuously retry". 

From what I can see, there has been something that makes the upper retry state machinery to stall. Perhaps, it's the way the Curl does the connection termination detection. It's not immediately clear what's going on from the snippet of logs. What I can tell is that there are two back-to-back sessions gracefully terminate but the stat machinery is not "primed" to continue. Perhaps, there is an issue prior to that with token rotation and consequent error that gets the state machinery awry.

Would you be able to reproduce the scenario and send the complete logs to kinesis-video-support@amazon.com?

P.S. There are multiple callbacks that can get fired - for example stream pressure callbacks which the custom application can react to. The C SDK provides it's own implementation to "react" on these notifications.

ek9852, 2019-10-30:
I enabled LOG_STREAMING this morning, and more ENTERS and LEAVED in all functions  continousreties file.  I am waiting for it to failed which will takes few hours or half day. Seems to me it happens on hour boundary during token rotation I guess.

MushMal, 2019-10-30:
Sounds good! LOG_STREAMING will produce lots of spew but perhaps will help. The token rotation is an interesting part - it doesn't let the streaming session to terminate first and then start a new session as it tries to maximize the bandwidth and handle high bandwidth high latency network scenarios by preparing the session before hand and switching over. There are numerous unit tests which validate the correctness in these scenarios both based on PIC (emulating the network) and with real networking. 

Let's see what the logs show. We have long-haul tests which are run for multiple days

ek9852, 2019-10-30:
@MathiasKoch  I am not sure, I use aws IoT certification (createDefaultCallbacksProviderWithIotCertificate) , seems the library use it to get a token, and the expiration time is one hour after current time.

unicornss, 2019-10-30:
@MathiasKoch 

- Could you please confirm if you are using  [#define DEFAULT_RESTART_ON_ERROR TRUE](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_app.cpp#L39)
- If you have tried it with "FALSE", does it behave differently?
Thanks

MushMal, 2019-10-30:
C SDK sets the default logging level via AWS_KVS_LOG_LEVEL env variable. Set it to 1 to log verbose. LOG_STREAMING might produce just too much spew which will be mostly irrelevant.

Regarding the token rotation.. KVS streaming sessions will be limited to max 40 minutes by the SDK (45 minutes enforced by the backend) regardless of the type of auth integration. True, some session tokens might have shorter lifespan but max is 40 minutes even with access key/secret key.

ek9852, 2019-10-30:
My expiration is 3600 s  BTW, probably due to I am using Iot:
 convertTimestampToEpoch(): Difference between current time and iot expiration is 3600
Trying to  reproduce with loglevel = debug and log_streaming off.

MushMal, 2019-10-30:
@ek9852 yes, indeed, IoT vends shorter term credentials. Also, try to set the logLevel as verbose to get more details.

ek9852, 2019-10-31:
@MushMal  Verbose log is sent to kinesis-video-support@amazon.com and Siva. 
BTW, the log files is >40M, gmail auto create a google drive link for it and  link is attached in the email.

ek9852, 2019-10-31:
Is seems the state machine had problem in :
2019-10-31 03:33:02 defaultStreamLatencyPressureCallback(): Reported streamLatencyPressure callback for stream handle 596597. Current buffer duration in 100ns: 510160000
2019-10-31 03:33:02 streamLatencyStateMachineHandleStreamLatency(): Stream Latency State Machine starting from NORMAL_STATE
2019-10-31 03:33:02 streamLatencyStateMachineToResetConnectionState(): Stream Latency State Machine move to RESET_CONNECTION_STATE
Then it never recover......

MathiasKoch, 2019-11-01:
Hmm.. Is there a way to set these verbose flags using the gstreamer sink? I tried setting the log4cpp loglevel to trace, but it seems to only log entries from the sink itself, not the underlying C lib? I also tried with `AWS_KVS_LOG_LEVEL=1 gst-launch-1.0 ...` but it doesn't really change anything?

MushMal, 2019-11-04:
Just to provide an update here... We have identified the root cause of the issue and will fix this in the upcoming release

MathiasKoch, 2019-11-04:
Hey @MushMal.

That sounds fantastic! Thank you! Given that i build from source anyways, is there a way to test your fix now? Master or a branch? Or is it still being developed?

MushMal, 2019-11-04:
@MathiasKoch, the internal codebase has diverged somewhat from the public codebase making the fix slightly less self-contained. Let us try to expedite this public release.

MathiasKoch, 2019-11-04:
Sounds good. Thank you very much :+1:

MathiasKoch, 2019-11-11:
@MushMal Any update on when we can expect a release fixing this?

MushMal, 2019-11-11:
@MathiasKoch very sorry for the delay. We are prioritizing this release currently. It will be up shortly.

MathiasKoch, 2019-11-22:
@MushMal Do you think you could maybe give a rough estimate of "shortly"? Just so we can relay the information to our costumers..?

MushMal, 2019-11-22:
@MathiasKoch our sincere apologies this is taking longer than anticipated. There were a few set backs which we have been handling. We are looking at most 2 weeks.

MushMal, 2019-12-04:
@MathiasKoch @ek9852 Apologies for the delay. The newest release v2.1.0 has the fixes. Could I ask you to give it a try. Thanks a lot for reporting!

Resolving this issue for now. Please open separate issues if anything is not as expected

closed 2019-12-04 by MushMal

MathiasKoch, 2019-12-04:
@MushMal Thank you! That sounds great! 

I will definitely give it a go, as soon as I am back from re:Invent. Will open a new issue if it is still a problem. Thank you!