awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #282: m3u8 resolution is wrong

ek9852, 2019-10-28:
I am sure we get correctly resolution in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/mkvgen/src/SpsParser.c#L213
Which is 160x120 in my case. Also the resolution showing correctly in the Amazon kinesis video streaming console.
Codec avc1.640028 (H.264 High@4.0)
Bitrate 12.6 kbps
Video Resolution 160x120
=============
But when i get the HLS url, the m3u8 playlist contains wrong resolution  as follow:
O0kszuq3JDl7iIgwZNtZaGwF5HQ7TO3HsD4kV0KIu-gCzt9lWxADaH74_o~  
#EXTM3U
#EXT-X-VERSION:1
#EXT-X-STREAM-INF:CODECS="avc1.640028",RESOLUTION=**160x16**,FRAME-RATE=30.02,BANDWIDTH=11153

This might be a problem in the server side btw.

ek9852, 2019-10-28:
That's wrong resolution does not affect playback the video using ffplay which it trusts the SPS/PPS information inside the codec instead. But creates a problem when playback in Apple quicktime player. which it trusts the m3u8 and shows the video as 160x16 size.

MushMal, 2019-10-28:
@ek9852 Thanks for the heads up, we will get back to you on this shortly.

MushMal, 2019-10-28:
@ek9852 could you please send the stream name and the AWS account ID to kinesis-video-support@amazon.com so we can debug this issue closely

ek9852, 2019-10-28:
Thx , writing up now.

ek9852, 2019-10-29:
@MushMal wonder did you received that, or that is taking care of by another team ?

zhiyua-git, 2019-10-29:
@ek9852 We have received the stream information, a team member is looking into it. Will get back to you once we found the root cause. Thanks.

MixMasterMitch, 2019-10-30:
@ek9852 Yes, similar to #255. I have found the bug that is causing the incorrect width and height parsing. It has been fixed and will be deployed soon.

ek9852, 2019-11-07:
@MixMasterMitch This have been one week. Is there any schedule when the fix will be released . Thanks.

MixMasterMitch, 2019-11-20:
This is released now. Can you please confirm that it is now working as expected?

ek9852, 2019-11-20:
The HLS video url can be playback now with Quicktime player with correct resolution.

closed 2019-11-20 by ek9852