awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #283: WARNING: erroneous pipeline: no element "kvssink" - Kinesis Video Streams Producer SDK on Windows using MSVC (Microsoft Visual C++)

jtheanalytica, 2019-10-31:
Hi everyone.
I followed the instructions for installing Kinesis Video Streams Producer SDK on Windows using MSVC (Microsoft Visual C++) in:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md

**When I run the following command:**
gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name="stream-name" access-key="YourAccessKeyId" secret-key="YourSecretAccessKey"

**I get the following error:**
(gst-launch-1.0:7588): GStreamer-WARNING **: Failed to load plugin 'C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll': 'C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll': The specified module could not be found.
WARNING: erroneous pipeline: no element "kvssink"

**Note: I'm on Virtual machine where both the host and the guest are Windows 10 Ent.**

**Note: running command "gst-device-monitor-1.0", gives the following:**

(gst-device-monitor-1.0:7324): GStreamer-WARNING **: Failed to load plugin 'C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll': 'C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll': The specified module could not be found.
Probing devices...

Device found:

        name  : Primary Sound Driver
        class : Audio/Sink
        caps  : audio/x-raw, format=(string)S16LE, layout=(string)interleaved, rate=(int)[ 1, 2147483647 ], channels=(int)[ 1, 2 ];
                audio/x-raw, format=(string)U8, layout=(string)interleaved, rate=(int)[ 1, 2147483647 ], channels=(int)[ 1, 2 ];
                audio/x-ac3, framed=(boolean)true;
                audio/x-dts, framed=(boolean)true;
        properties:
                device.api = directsound
                device.guid = "\(NULL\)"
                directsound.device.driver = ""
                directsound.device.description = "Primary\ Sound\ Driver"
        gst-launch-1.0 ... ! directsoundsink

Device found:

        name  : Primary Sound Capture Driver
        class : Audio/Source
        caps  : audio/x-raw, format=(string){ S16LE, S8 }, layout=(string)interleaved, rate=(int)[ 1, 2147483647 ], channels=(int)[ 1, 2 ];
        properties:
                device.api = directsound
                device.guid = "\(NULL\)"
                directsound.device.driver = ""
                directsound.device.description = "Primary\ Sound\ Capture\ Driver"
        gst-launch-1.0 directsoundsrc ! ...

Device found:

        name  : VirtualBox Webcam - Integrated Camera
        class : Video/Source
        caps  : image/jpeg, width=(int)640, height=(int)480, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)320, height=(int)180, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)320, height=(int)240, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)352, height=(int)288, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)12/11;
                image/jpeg, width=(int)424, height=(int)240, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)640, height=(int)360, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)848, height=(int)480, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)960, height=(int)540, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)1280, height=(int)720, framerate=(fraction)[ 5/1, 30/1 ], pixel-aspect-ratio=(fraction)1/1;
        gst-launch-1.0 ksvideosrc device-path="\\\\\?\\usb\#vid_80ee\&pid_0030\&mi_00\#6\&18ea6372\&0\&0000\#\{6994ad05-93ef-11d0-a3cc-00a0c9223196\}\\global" ! ...

jtheanalytica, 2019-10-31:
**Also, when I try to execute the command:** 
kinesis_video_gstreamer_sample_app.exe my-test-stream

**I get the following:**

(kinesis_video_gstreamer_sample_app.exe:8016): GLib-GObject-WARNING **: g_object_set_is_valid_property: object class 'GstKsVideoSrc' has no property named 'device'
Default list of resolutions (1920x1080, 1280x720, 640x480) are not supported by video source

**Full output:**
C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release>kinesis_video_gstreamer_sample_app.exe MyKinesisVideoStream
[INFO ][2019-10-31 14:53:25] Using region: us-west-2
[INFO ][2019-10-31 14:53:25] Using aws credentials for Kinesis Video Streams
[INFO ][2019-10-31 14:53:25] No session token was detected.
[INFO ][2019-10-31 14:53:25] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-10-31 14:53:25] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-10-31 14:53:25] heapInitialize(): Creating AIV heap.
[INFO ][2019-10-31 14:53:25] heapInitialize(): Heap is initialized OK
[DEBUG][2019-10-31 14:53:25] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-10-31 14:53:25] getSecurityTokenHandler invoked
[DEBUG][2019-10-31 14:53:25] Refreshing credentials. Force refreshing: 0 Now time is: 15725336052176467 Expiration: 0
[INFO ][2019-10-31 14:53:25] New credentials expiration is 1572536005
[DEBUG][2019-10-31 14:53:25] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-10-31 14:53:25] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-10-31 14:53:25] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-10-31 14:53:25] clientReadyHandler invoked
[DEBUG][2019-10-31 14:53:25] Client is ready
[INFO ][2019-10-31 14:53:25] Creating Kinesis Video Stream MyKinesisVideoStream
[INFO ][2019-10-31 14:53:25] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-10-31 14:53:25] stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-10-31 14:53:25] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-10-31 14:53:26] writeHeaderCallback(): RequestId: e4dcd84f-a1ba-4fa7-a0ee-03df7bdb5bc5
[DEBUG][2019-10-31 14:53:26] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.572504870372E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:368013261494:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:368013261494:stream/MyKinesisVideoStream/1572504870372","StreamName":"MyKinesisVideoStream","Version":"6FFxvSsmIRaej8cpGBP7"}}
[INFO ][2019-10-31 14:53:26] describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-10-31 14:53:26] stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
[DEBUG][2019-10-31 14:53:26] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-10-31 14:53:27] writeHeaderCallback(): RequestId: 298ad05f-16f3-4be2-895b-5a3e7c4640ad
[DEBUG][2019-10-31 14:53:27] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
[INFO ][2019-10-31 14:53:27] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-10-31 14:53:27] stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
[DEBUG][2019-10-31 14:53:27] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[DEBUG][2019-10-31 14:53:27] getStreamingTokenHandler invoked
[DEBUG][2019-10-31 14:53:27] Refreshing credentials. Force refreshing: 1 Now time is: 15725336077664995 Expiration: 1572536005
[INFO ][2019-10-31 14:53:27] New credentials expiration is 1572536007
[INFO ][2019-10-31 14:53:27] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-10-31 14:53:27] stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-10-31 14:53:27] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 1948218831033
[DEBUG][2019-10-31 14:53:27] streamReadyHandler invoked
[DEBUG][2019-10-31 14:53:27] Stream is ready
[INFO ][2019-10-31 14:53:27] Streaming from live source
[DEBUG][2019-10-31 14:53:27] Streaming with live source and width: 0, height: 0, fps: 25, bitrateInKBPS512

(kinesis_video_gstreamer_sample_app.exe:8016): GLib-GObject-WARNING **: g_object_set_is_valid_property: object class 'GstKsVideoSrc' has no property named 'device'
Default list of resolutions (1920x1080, 1280x720, 640x480) are not supported by video source
[INFO ][2019-10-31 14:53:27] stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream 000001c59ae430b9.
[DEBUG][2019-10-31 14:53:27] defaultStreamClosedCallback(): Reported streamClosed callback for stream handle 1948218831033. Upload handle 18446744073709551615
[DEBUG][2019-10-31 14:53:27] streamClosedHandler invoked for upload handle: 18446744073709551615
[INFO ][2019-10-31 14:53:27] Freeing Kinesis Video Stream MyKinesisVideoStream
[INFO ][2019-10-31 14:53:27] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-10-31 14:53:28] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 1948218831033
[INFO ][2019-10-31 14:53:28] Freeing Kinesis Video Stream MyKinesisVideoStream
[INFO ][2019-10-31 14:53:28] freeKinesisVideoClient(): Freeing Kinesis Video Client

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release>

unicornss, 2019-10-31:
Hi @jtheanalytica 

Could you please clarify how  are you  enabling webcam attached to the Guest OS?

zhiyua-git, 2019-10-31:
Hi @jtheanalytica ,

Can you try to run gst-launch-1.0 ksvideosrc ! decodebin ! videoconvert ! autovideosink to check if you are able to use the webcam? You might need to add your gstreamer path (kinesis-video-native-build/downloads/gstreamer/1.0/x86_64/bin) to your GST_PLUGIN_SYSTEM_PATH. 

For gst-launch-1.0, did you set the PATH and GST_PLUGIN_PATH as https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md? If you did, can you also try **set PATH=%PATH%;C:\Users\<myuser>\Downloads\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release** and rerun the command?

Thanks,
-ZH

jtheanalytica, 2019-10-31:
Hi @zhiyua-git. Thank you for the reply. 

1) Running **gst-launch-1.0 ksvideosrc ! decodebin ! videoconvert ! autovideosink** invokes the webcam.
Following is what I got before and after closing the webcam:
Setting pipeline to PAUSED ...

(gst-launch-1.0:7900): GStreamer-WARNING **: Failed to load plugin 'C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll': 'C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release\gstkvssink.dll': The specified module could not be found.
Pipeline is live and does not need PREROLL ...
Setting pipeline to PLAYING ...
New clock: GstSystemClock
ERROR: from element /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstD3DVideoSink:autovideosink0-actual-sink-d3dvideo: Output window was closed
Additional debug info:
d3dhelpers.c(1865): d3d_render_buffer (): /GstPipeline:pipeline0/GstAutoVideoSink:autovideosink0/GstD3DVideoSink:autovideosink0-actual-sink-d3dvideo
Execution ended after 0:00:06.978603300
Setting pipeline to PAUSED ...
Setting pipeline to READY ...
Setting pipeline to NULL ...
Freeing pipeline …

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>

2) I did  **set PATH=%PATH%;C:\Users<myuser>\Downloads\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build\Release**
I ran the command. Now I get the following:

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>gst-launch-1.0.exe ksvideosrc do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name=MyKinesisVideoStream access-key="" secret-key=""
**Setting pipeline to PAUSED ...
[INFO ][2019-10-31 18:55:53] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-10-31 18:55:53] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-10-31 18:55:53] heapInitialize(): Creating AIV heap.
[INFO ][2019-10-31 18:55:53] heapInitialize(): Heap is initialized OK
[DEBUG][2019-10-31 18:55:53] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-10-31 18:55:53] getSecurityTokenHandler invoked
[DEBUG][2019-10-31 18:55:53] Refreshing credentials. Force refreshing: 0 Now time is: 15725481539796083 Expiration: 0
[DEBUG][2019-10-31 18:55:53] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-10-31 18:55:53] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-10-31 18:55:53] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-10-31 18:55:53] clientReadyHandler invoked

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>**

jtheanalytica, 2019-10-31:
Hi @unicornss. Thank you for the inquiry. 
From the VM Manager, go to settings of the VM Machine, then USB. Enable USB Controller. Select USB 2.0 Controller. 
Hope that helps!

zhiyua-git, 2019-10-31:
Hi @jtheanalytica ，

Seems you were not able to get the playback with **gst-launch-1.0 ksvideosrc ! decodebin ! videoconvert ! autovideosink**, then I would recommend checking your exposure of camera into virtual machine.

Before you do that, can you try **kinesis_video_gstreamer_sample_app.exe my-test-stream rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov**? If this works then KVS sdk is working and you will only need to fix the camera issue.

Thanks,
-ZH

jtheanalytica, 2019-10-31:
Hi @zhiyua-git.
1) **kinesis_video_gstreamer_sample_app.exe my-test-stream rtsp://wowzaec2demo.streamlock.net/vod/mp4:BigBuckBunny_115k.mov**  worked!

2) running gst-launch-1.0 ksvideosrc ! decodebin ! videoconvert ! autovideosink opens my camera. Is this what you meant by "playback"? If not, please suggest how I can fix the camera issue? 

Thanks

zhiyua-git, 2019-10-31:
Hi @jtheanalytica ,

Nice to know rtsp worked for you! By playback I mean you will actually see a playback windows showing what your camera captures with the command. If only your camera is opened, but no video data flows into the pipeline, then you might not be able to see any playback. Then you will need to check if you exposed your camera properly, one potential way to check that is you can open your camera app in windows in your virtual machine to check if your camera is exposed.

Thanks,
-ZH

jtheanalytica, 2019-10-31:
Hi @zhiyua-git.
Thank you.
If by "exposed" you mean like seeing myself in the camera - then yes I can! If not please explain otherwise. 

Also, please let me know how I can solve the other issue I pointed out earlier - pertaining to the command: **kinesis_video_gstreamer_sample_app.exe my-test-stream**
I get the following:
**(kinesis_video_gstreamer_sample_app.exe:8016): GLib-GObject-WARNING **: g_object_set_is_valid_property: object class 'GstKsVideoSrc' has no property named 'device'
Default list of resolutions (1920x1080, 1280x720, 640x480) are not supported by video source**

zhiyua-git, 2019-10-31:
Hi @jtheanalytica ,

Seems the kinesis_video_gstreamer_sample_app.exe my-test-stream is not getting your device correctly.

Can you run this? If this doesn't work for you, what was the error?
gst-launch-1.0 ksvideosrc device-path="\\\\\?\\usb\#vid_80ee\&pid_0030\&mi_00\#6\&18ea6372\&0\&0000\#\{6994ad05-93ef-11d0-a3cc-00a0c9223196\}\\global" do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name=MyKinesisVideoStream access-key="" secret-key=""

Thanks,
-ZH

unicornss, 2019-10-31:
Hi @jtheanalytica 

The sample GStreamer application and the pipelines demonstrate ingesting video source through GStreamer. Please refer to the media format of the Webcam from the Windows (physical). 

```
Device found:

        name  : HP Wide Vision HD
        class : Video/Source
        caps  : video/x-raw, format=(string)YUY2, width=(int)640, height=(int)480, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
                video/x-raw, format=(string)YUY2, width=(int)176, height=(int)144, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)12/11;
                video/x-raw, format=(string)YUY2, width=(int)320, height=(int)240, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
                video/x-raw, format=(string)YUY2, width=(int)352, height=(int)288, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)12/11;
                video/x-raw, format=(string)YUY2, width=(int)640, height=(int)360, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
                video/x-raw, format=(string)YUY2, width=(int)1280, height=(int)720, framerate=(fraction)10/1, pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)640, height=(int)480, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)176, height=(int)144, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)12/11;
                image/jpeg, width=(int)320, height=(int)240, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)352, height=(int)288, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)12/11;
                image/jpeg, width=(int)640, height=(int)360, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
                image/jpeg, width=(int)1280, height=(int)720, framerate=(fraction)30/1, pixel-aspect-ratio=(fraction)1/1;
        gst-launch-1.0 ksvideosrc device-path="\\\\\?\\usb\#vid_05c8\&pid_038e\&mi_00\#6\&20a8b444\&0\&0000\#\{6994ad05-93ef-11d0-a3cc-00a0c9223196\}\\global" ! ...
```

In comparison, the Virtualbox reports ` image/jpeg, width=(int)1280,` from the webcam source.  

Given that the decodebin ! videoconvert works for you, you just need to construct GStreamer pipeline to convert the source media type similar to the  example _2.2 Use gst-launch-1.0 to send audio and raw video in the [instructions]( https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-windows-msvc.md#running-the-sample-demo-applications-to-stream-video-to-kinesis-video-streams )_ and [GStreamer developer documentation](https://gstreamer.freedesktop.org/dev/).

jtheanalytica, 2019-11-01:
Hello @zhiyua-git 
running the command, produces the following:

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>gst-launch-1.0 ksvideosrc device-path="\\?\usb#vid_80ee&pid_0030&mi_00#6&18ea6372&0&0000#{6994ad05-93ef-11d0-a3cc-00a0c9223196}\global" do-timestamp=TRUE ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name=MyKinesisVideoStream access-key=''' secret-key='''

Setting pipeline to PAUSED ...
[INFO ][2019-11-01 04:37:50] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-11-01 04:37:50] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-11-01 04:37:50] heapInitialize(): Creating AIV heap.
[INFO ][2019-11-01 04:37:50] heapInitialize(): Heap is initialized OK
[DEBUG][2019-11-01 04:37:50] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-11-01 04:37:50] getSecurityTokenHandler invoked
[DEBUG][2019-11-01 04:37:50] Refreshing credentials. Force refreshing: 0 Now time is: 15725830706532486 Expiration: 0
[DEBUG][2019-11-01 04:37:50] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-11-01 04:37:50] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-11-01 04:37:50] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-11-01 04:37:50] clientReadyHandler invoked

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>

jtheanalytica, 2019-11-01:
Hi @unicorns.

jtheanalytica, 2019-11-01:
Hi @unicorns.
Thank you.
When I run the command you suggested, I get the following:

C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>gst-launch-1.0 -v ksvideosrc ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 tune=zerolatency ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink name=sink stream-name=MyKinesisVideoStream access-key=''' secret-key=''' wasapisrc device="\{0.0.1.00000000\}.\{f1245929-0c97-4389-9c28-1ca9cb01576b\}" ! audioconvert ! avenc_aac ! queue ! sink.
Setting pipeline to PAUSED ...
[INFO ][2019-11-01 05:10:19] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-11-01 05:10:19] heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
[INFO ][2019-11-01 05:10:19] heapInitialize(): Creating AIV heap.
[INFO ][2019-11-01 05:10:19] heapInitialize(): Heap is initialized OK
[DEBUG][2019-11-01 05:10:19] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-11-01 05:10:19] getSecurityTokenHandler invoked
[DEBUG][2019-11-01 05:10:19] Refreshing credentials. Force refreshing: 0 Now time is: 15725850199651894 Expiration: 0
[DEBUG][2019-11-01 05:10:19] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-11-01 05:10:19] createDeviceResultEvent(): Create device result event.

I do not plan to run it from the host machine.
[DEBUG][2019-11-01 05:10:19] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-11-01 05:10:19] clientReadyHandler invoked
C:\Users\IEUser\amazon-kinesis-video-streams-producer-sdk-cpp\kinesis-video-native-build>

It's not possible to run it from the host OS.

Thanks

unicornss, 2019-11-01:
Hi @jtheanalytica 

> access-key=''' secret-key=''' 

Please don't post your credentials here, but confirm if you are using your AWS_ACCESS_KEY_ID and AWS_SECRET_ACCESS_KEY when running the gst-launch-1.0 command.

jtheanalytica, 2019-11-01:
Hi @unicornss. 
Yes, I confirm. I just leave them blanks between "" when posting in here. Thanks

zhiyua-git, 2019-11-02:
Hi @jtheanalytica ,

Due to some issues, the plugin was not working. Can you comment out/remove these [two lines](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-gstreamer-plugin/plugin-src/gstkvssink.cpp#L1324-L1325) and rebuild by running windows-install.bat 64, then run with this pipeline?

gst-launch-1.0 ksvideosrc do-timestamp=TRUE ! image/jpeg,width=640,height=480 ! jpegdec ! video/x-raw,width=640,height=480,framerate=30/1 ! videoconvert ! x264enc bframes=0 key-int-max=45 bitrate=512 ! video/x-h264,profile=baseline,stream-format=avc,alignment=au ! kvssink stream-name="stream-name" access-key="" secret-key=""

Your previous one will not work as it try to filter video/x-raw but your camera only outputs image/jpeg.

Thanks,
-ZH

jtheanalytica, 2019-11-02:
Hi @zhiyua-git .

That really did the trick! It worked!
Thank you a lot.

jtheanalytica, 2019-11-11:
Hi @zhiyua-git 
Hi @unicornss .
I've been trying to build up on the SDK mentioned in this thread by the following the steps mentioned in the following link "Recognizing Faces in a Streaming Video": 
https://docs.aws.amazon.com/rekognition/latest/dg/recognize-faces-in-a-video-stream.html
I'm stuck at steps 8 and 9. Any idea how to put streamed video into the Kinesis data stream and then reading the analysis - using this SDK (Kinesis Video Streams C++ Producer Library)? 

Reason being is that if you delve inside the "Sending Data to Kinesis Video Streams Using the PutMedia API " (https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/examples-putmedia.html#examples-putmedia-write)
as indicated by step 8 in  https://docs.aws.amazon.com/rekognition/latest/dg/recognize-faces-in-a-video-stream.html
you would notice that it overlaps with C++ producer library.

Thank you!

zhiyua-git, 2019-11-11:
Hi @jtheanalytica ,

You can refer to https://github.com/aws/amazon-kinesis-video-streams-parser-library#kinesisvideo---rekognition-examples. This sample will capture the result and draw a bounding box over detected faces on the video.

Thanks,
-ZH

jtheanalytica, 2019-11-12:
Hi @zhiyua-git. Thank you.
Does this mean that I need to first run the "amazon-kinesis-video-streams-producer-sdk-cpp", then stream the video, and then to run the example code you provided in your previous comment? 
Thanks

zhiyua-git, 2019-11-12:
@jtheanalytica  Yes.

jtheanalytica, 2019-11-14:
Hi @zhiyua-git . Thank you.
I built the project by running the command: mvn clean package. A target folder was created with four jar files. I got the following error trying to run the the "KinesisVideoRekognitionIntegrationExampleTest" class: 

C:\Users\IEUser\amazon-kinesis-video-streams-parser-library>java -cp target\amazon-kinesis-video-streams-parser-library-1.0.14.jar com.amazonaws.kinesisvideo.parser.examples.KinesisVideoRekognitionIntegrationExampleTest

Error: Could not find or load main class com.amazonaws.kinesisvideo.parser.examples.KinesisVideoRekognitionIntegrationExampleTest

C:\Users\IEUser\awsProjects\amazon-kinesis-video-streams-parser-library>

Thanks

zhiyua-git, 2019-11-14:
Hi @jtheanalytica ,

As your pipeline issue on Windows using MSVC is resolved, can you close this issue and create a new issue for your rekognition sample questions in https://github.com/aws/amazon-kinesis-video-streams-parser-library/issues to make it clear?

Thanks,
-ZH

jtheanalytica, 2019-11-14:
Hi @zhiyua-git. 
Sure thing. 
Thanks

closed 2019-11-15 by jtheanalytica

jtheanalytica, 2019-11-15:
@zhiyua-git.
I've opened the issue #75. Link: https://github.com/aws/amazon-kinesis-video-streams-parser-library/issues/75

Thanks

zhiyua-git, 2019-11-15:
Hi @jtheanalytica , a colleague is looking into and will reply to you when he find something, at the same time, do you mind trying it in Intellij first? Thanks for your patience.