awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #285: Samples not Working on GM8136

hsinghaowang, 2019-11-01:
**Chip: GM8136
RAM: 128MB
OS: arm-linux-3.3
toolchain_gnueabi-4.8.x_ARM926
KVS SDK version: 2.02 commit 116b9041e111705c056**

I am trying to make KVS sample working on our platform.
It works normally in Ubuntu 18.04, but not on our platform.

I wonder if it can work on a pure 32bits Linux?

I was testing kvsVideoOnlyStreamingSample in kinesis-video-c-producer
here is the output:
```
[root@GM]#  ./kvsVideoOnlyStreamingSample test 10 samples/
1970-01-01 05:43:04 heapInitialize(): Initializing native heap with limit size 10485760, spill ratio 0% and flags 0x00000001
1970-01-01 05:43:04 heapInitialize(): Creating AIV heap.
1970-01-01 05:43:04 heapInitialize(): Heap is initialized OK
1970-01-01 05:43:04 stepStateMachine(): PIC Client State Machine - Current state: 0x2dfc8, Next state: 0x1
1970-01-01 05:43:04 stepStateMachine(): PIC Client State Machine - Current state: 0x76f76ce0, Next state: 0x2
1970-01-01 05:43:04 createDeviceResultEvent(): Create device result event.
1970-01-01 05:43:04 stepStateMachine(): PIC Client State Machine - Current state: 0x76f5a17c, Next state: 0x10
1970-01-01 05:43:04 createKinesisVideoStream(): Creating Kinesis Video Stream.
1970-01-01 05:43:04 stepStateMachine(): PIC Stream State Machine - Current state: 0x76b41630, Next state: 0x1
Segmentation fault
```

So I tried the unittest in PIC libs, there are failures in kvsgen & client. The others work fine.

Here are the logs:
[mkvgenTest.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3797219/mkvgenTest.log)
[clientTest.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3797222/clientTest.log)

37920091b02da6b63256b

MushMal, 2019-11-01:
@hsinghaowang thanks for the detailed information. Seems that you might have two issues with the crash and the unit tests failing. You are using fairly old toolchain - 4.8 for the C++ assets (which are the unit tests). Could you try to update the toolchain to something more recent? If not, could you please set the explicit array sizes in the test and see if it works? https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/mkvgen/tst/AnnexBCpdNalAdapterTest.cpp#L39 and LN 44, 45 and 47 - instead of empty brackets like BYTE cpds[][21] = ... set it to 2 like BYTE cpds[2][21] = ...

Regarding the crash in the response. Could you please let us know whether your platform supports unaligned access? Looking at the crash stack I can't really tell what's going on as the line numbers seem to not match.

hsinghaowang, 2019-11-04:
*I tried clang 9.0, and it still has failures.
*After changed to  BYTE cpds[2][21], it still has failures. [AnnexBCpdNalAdapterTest.log](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3802571/AnnexBCpdNalAdapterTest.log)
*For unaligned access, I wrote a test:
```
    char buf[8];
    int* pi = (int*)(buf+1);
    *pi = 9;
    cout << "pi=" << *pi << endl;
```
the result is:
```
[root@GM]# ./unalinged_access 
pi=150994944
```
with aligned access:
```
    char buf[8];
    int* pi = (int*)(buf);
    *pi = 9;
    cout << "pi=" << *pi << endl;
```
the result is:
```
[root@GM]# ./alinged_access 
pi=9
```
So I don't think it support unaligned access.

MushMal, 2019-11-04:
It is odd that the platform doesn't crash on your unaligned sample - I would expect that if the platform doesn't support alignment and the compiler can't emulate it then it would bus error.

The KVS SDK is designed for tight packing and as such we have no padding in the structures (pragma pack 1), including auxiliary structures in the internal heap implementation.

Is there a way you guys can turn on unaligned access support for the SOC you have? The KVS SDK itself wouldn't suffer from the perf degradation. Hope the encoder will not be impacted.

hsinghaowang, 2019-11-04:
I found a compile option "-fmax-type-align=1" to make my sample works.
And the AnnexBCpdNalAdapterTest passed.
```
[root@GM]# ./mkvgenTest --gtest_filter=AnnexBCpdNalAdapterTest.nalAdapter_ValidZerosInStream:AnnexBCpdNalAdapterTest/*.nalAdapter_ValidZerosInStream:*/AnnexBCpdNa
lAdapterTest.nalAdapter_ValidZerosInStream/*:*/AnnexBCpdNalAdapterTest/*.nalAdapter_ValidZerosInStream --gtest_color=no
Note: Google Test filter = AnnexBCpdNalAdapterTest.nalAdapter_ValidZerosInStream:AnnexBCpdNalAdapterTest/*.nalAdapter_ValidZerosInStream:*/AnnexBCpdNalAdapterTest.nalAdapter_ValidZerosInStream/*:*/AnnexBCpdNalAdapterTest/*.nalAdapter_ValidZerosInStream
[==========] Running 1 test from 1 test case.
[----------] Global test environment set-up.
[----------] 1 test from AnnexBCpdNalAdapterTest
[ RUN      ] AnnexBCpdNalAdapterTest.nalAdapter_ValidZerosInStream
[       OK ] AnnexBCpdNalAdapterTest.nalAdapter_ValidZerosInStream (1 ms)
[----------] 1 test from AnnexBCpdNalAdapterTest (3 ms total)

[----------] Global test environment tear-down
[==========] 1 test from 1 test case ran. (9 ms total)
[  PASSED  ] 1 test.
```
But MkvgenApiFunctionalityTest still has failures:
[mkvgenTest_2019-11-04.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3803050/mkvgenTest_2019-11-04.txt)

And clientTest is still bad:
```
[root@GM]# ./clientTest 
[==========] Running 1235 tests from 20 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from ClientApiFunctionalityTest
[ RUN      ] ClientApiFunctionalityTest.createKinesisVideoClient_AuthIntegration

[0x000d15dd00000003] [level 852703] (null) 
[0x000d15dd00000003] [level 852703] (null) 
[0x000dd0bb00000003] [level 902350] t 
[0x000dd0bb00000003] [level 902350] t 
Segmentation fault
[root@GM]# 
```
BTW:
__attribute__((aligned(1))) works:
```
  typedef int unaligned_int __attribute__((aligned(1)));

  char buf[8];
  unaligned_int *pi = (unaligned_int*)(buf + 1);
  *pi = 9;
  cout << "pi=" << *pi << endl;
```
__packed   doesn't work.
```
Building CXX object sandbox.cpp.o
sandbox.cpp:15:3: error: unknown type name '__packed'
  __packed int* pi = (__packed int*)(buf+1);
  ^
sandbox.cpp:15:23: error: use of undeclared identifier '__packed'
  __packed int* pi = (__packed int*)(buf+1);
```

--pointer_alignment=1  doesn't work.
```
Building CXX object sandbox.cpp.o
clang-9: error: unsupported option '--pointer_alignment=1'
```

MushMal, 2019-11-04:
Not sure if fmax-type-align can help as it will simply align the pointers but will not change the structure layout. The failures in the unit tests are stemming from the bad code gen.

Can you please investigate whether the platform can have unaligned memory access?

chenek, 2019-11-04:
@MushMal As I know, GM8136 uses ARM926 which doesn't support unaligned memory access. Can you provide advise how to revise KVS sample code to make it work with aligned memory access?

MushMal, 2019-12-03:
@chenek the KVS SDK does not support unaligned memory access currently. We will evaluate the unaligned access support and see how we can support it easier.

MushMal, 2019-12-03:
Unfortunately, as we have no support for unaligned memory, I am going to resolve this as "won't fix" for now

closed 2019-12-03 by MushMal

hsinghaowang, 2019-12-19:
I pulled the latest code today, and build tests on our platform.
It had segmentation fault. Any idea?

```
[root@GM]# ./kvspic_test
[==========] Running 1825 tests from 55 test cases.
[----------] Global test environment set-up.
[----------] 2 tests from ClientApiFunctionalityTest
[ RUN      ] ClientApiFunctionalityTest.createKinesisVideoClient_AuthIntegration

[0x0000000076f0e000] [level 3] KinesisVideoClient createKinesisVideoClient(): Creating Kinesis Video Client
[0x0000000076f0e000] [level 3] Heap heapInitialize(): Initializing native heap with limit size 10485760, spill ratio 0% and flags 0x00000001
[0x0000000076f0e000] [level 3] Heap heapInitialize(): Creating AIV heap.
[0x0000000076f0e000] [level 3] Heap heapInitialize(): Heap is initialized OK
[0x0000000076f0e000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
[0x0000000076f0e000] [level 2] State stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
[0x0000000076f0e000] [level 3] platform-utils CreateClient(): Create client returned status code is 00000000

Segmentation fault

```

MushMal, 2019-12-19:
@hsinghaowang could you please open a separate issue and include more details like the stack trace, etc..