awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #286: Timeout  or thread sleep in the library is based on real time clock not monotonic timer

ek9852, 2019-11-06:
Real time clock should never be used  for calc the timeout.
IoT device might not have RTC  backup battery and  real time clock   will sync to ntp server after connected to  internet. This will mess up the timeout in the library when it is launched and when ntp is get sync.

Pls use clock_gettime(CLOCK_MONOTONIC, &time) instead of clock_gettime(CLOCK_REALTIME, &time)  when calc timeout for tasks in the library.

FYI, there is a lot of open source example  of how to use clock_gettime(CLOCK_MONOTONIC, &time) i.
Such as libcurl which already using monotonic timer for timeout calculation.
e.g.
https://github.com/curl/curl/blob/46c89348b973b1088a4310e740def9859ef05408/src/tool_util.c#L67

For pthread timeout in Mutex.c pls use
pthread_condattr_t attr;
pthread_cond_t cond;
/* ... */
pthread_condattr_init(&attr);
pthread_condattr_setclock(&attr, CLOCK_MONOTONIC);
pthread_cond_init(&cond, &attr); To init the mutex.
for  use monotonic timer for pthread_cond_timedwait

MushMal, 2019-11-06:
@ek9852 Thanks a lot for the suggestion. The time functionality in PIC is defined in https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/utils/src/Time.c#L30

and C++ layer uses https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-producer/src/GetTime.cpp#L4

Aside from this, there is a clock definition in the Java producer library which uses sys time.

One has to be very careful with these in order to not break other platforms - we in fact were using monotonic initially but after some breaks and issues with the backend decided to use system time (for example, AWS operates with real time for credentials, API calls).

Let us look into this first but I would like to understand your mentioned scenario with the realtime and IoT. The device should do NTP at the early boot time as most of the security/certs/kerberos won't work anyway. In this essence, it would be prudent to rely on the fact that by the time the KVS based application starts (a daemon for example) the time would be already adjusted?

ek9852, 2019-11-06:
We shalln't assume the NTP is sync before the program is ran,  and even the NTP server time might be incorrect either. We are seeing company using  its own  intranet NTP server  and is 2 min. behind the actual time.

Using monotonic timer will not break platform does not support it . ( and i doubt if any platform does not support it nowadays). See libcurl for the  implementation as an example. We can use #define or whatever ways to fallback into gettimeofday of CLOCK_REALTIME if CLOCK_MONOTONiC is not available. In Windows , we should use  GetTickCount or GetTickCount64. See libcurl tvnow implementation:
https://github.com/curl/curl/blob/master/src/tool_util.c#L30

ek9852, 2019-11-06:
Prepares introduce a new function GETTICK/GETMONOTONICTIME  instead of GETTIME  on the library . And changes those use the monotonic timer function for timeout  and keep the remaining the same.

MushMal, 2019-11-06:
These are valid points, however, in the context of the KVS and overall AWS these will not be valid anyway. We use time functionality for two purposes - 1) calculating time intervals and 2) using for absolute time.

True, the monotonic timer will work for cases which fall into category #1 but will not help with category #2. Curl implementation use cases fall into category #1 when time intervals are calculated.

The main assumption for the KVS SDK is that the NTP issue is solved on the devices - this is also important from the perspective of absolute frame timestamps that many of the customers are requiring.

Separating time interval functionality to use one clock (Monotonic) and the wall clock functionality to use another (Real time) will not be the main issue - it's a lot of plumbing and we need to handle cases where the customer passes in clocks and the Java/iOS/Android scenarios but it is not sufficient due to clock drift issues in Auth for example.

You can try to change the implementation of the GETTIME macro in PIC Utils and see how that works for your platform.

MushMal, 2019-11-06:
BTW, Using KVS C Producer, one can supply his/her own get time functionality in the Platform Callbacks

MushMal, 2019-12-04:
Resolving this issue

closed 2019-12-04 by MushMal

ek9852, 2019-12-27:
One PR submit to fix the issue:
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/pull/316
There is other codebase in the project  that use realtime clock as timeout also which should use monotonic clock).