awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #288: [ 88%] Built target kvs_producer_plugin_demo make: *** [Makefile:84: all] Error 2

gabrielarcangelbol, 2019-11-07:
Hi guys,

I´m trying to install Kinesis Video Streams Producer SDK on Windows using MSVC (Microsoft Visual C++), i followed the guidelines, when i type:  ./min-install-script, i get the following error:

[ 88%] Linking CXX executable kvs_producer_plugin_demo.exe
Assembler messages:
Fatal error: can't create CMakeFiles/kinesis_video_gstreamer_sample_multistream_app.dir/C_/msys64/home/gabri_7a484pu/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_multistream_app.cpp.obj: No such file or directory
make[2]: *** [CMakeFiles/kinesis_video_gstreamer_sample_multistream_app.dir/build.make:63: CMakeFiles/kinesis_video_gstreamer_sample_multistream_app.dir/C_/msys64/home/gabri_7a484pu/amazon-kinesis-video-streams-producer-sdk-cpp/kinesis-video-gst-demo/kinesis_video_gstreamer_sample_multistream_app.cpp.obj] Error 1
make[1]: *** [CMakeFiles/Makefile2:150: CMakeFiles/kinesis_video_gstreamer_sample_multistream_app.dir/all] Error 2
make[1]: *** Waiting for unfinished jobs....
[ 88%] Built target kvs_producer_plugin_demo
make: *** [Makefile:84: all] Error 2

I´m using windows 10 home, i read that the version supported is windows 10 enterprise, this could be the cause?

If is not, could you guide me through the steps i need to take to get this done, please?

Thanks!

chehefen, 2019-12-02:
you can try rename amazon-kinesis-video-streams-producer-sdk-cpp directory to something shorter? liker "kvs"

MushMal, 2019-12-04:
Resolving as answered

closed 2019-12-04 by MushMal