awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #289: Unable to pause gstreamer pipeline containing rtspsrc and kvssink due to MAX_FRAGMENT_DURATION_REACHED error

ManojBahuguna, 2019-11-09:
I have a use-case where I need to
- keep the video stream paused ideally (i.e drop the frames)
- then on some event start stream for about a few seconds
- and pause again.

I am using the following pipeline for this:
`rtspsrc location=rtsp://rtsp-url protocols=tcp ! valve name=valve drop=1 ! rtph264depay ! h264parse ! kvssink name=stream_sink stream-name=stream-name`

I change the `drop` property of `valve` to pause/resume stream

While this works fine when I use `filesink`, this is not working with `kvssink`.

The first time it plays, It works fine. Then when it is paused (set valve property `drop=1`), and resumed again `valve drop=0`, I receive **`MAX_FRAGMENT_DURATION_REACHED`** error.

I also tried to dynamically unlink kvssink and link a fakesink to the pipeline, but although that too works with filesink, with kvssink I get the same error.

This actually happens when the stream is paused for more than 10 seconds (which is the max fragment duration according to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/limits.html). So can't I just drop that faulty fragment/buffer and continue with the next one?

**What would be the most efficient way to pause and resume the stream seamlessly with minimum delay?**

js1972, 2019-11-09:
I’m trying to do this as well and have found out that kvssink simply isn’t designed to handle the intermittent producer scenario that your speaking of. 
Instead we have to use the c++ producer api directly. Maybe you could use a probe on one of your elements and in there start a stream; put frames and then close the stream.

ManojBahuguna, 2019-11-09:
> I’m trying to do this as well and have found out that kvssink simply isn’t designed to handle the intermittent producer scenario that your speaking of.  

Thanks for this information.

> Instead we have to use the c++ producer api directly. Maybe you could use a probe on one of your elements and in there start a stream; put frames and then close the stream.  

I was actually working with python bindings of gstreamer till now for this. But since kvssink element seems not to be an option, will try using the c++ producer api.

Would you suggest any good resources that could help in this scenario?

js1972, 2019-11-09:
Some of the basic example code on the aws website is the best I’ve managed
to find.

ManojBahuguna, 2019-11-27:
Currently I am
- removing kvssink element from pipeline to pause
- recreating and adding the element to play.
Working fine for my usecase as of now.

Resources:
- https://gstreamer.freedesktop.org/documentation/application-development/advanced/pipeline-manipulation.html?gi-language=python
- https://github.com/ViGLug/libre-streaming/blob/master/libre-streaming.py

closed 2019-11-27 by ManojBahuguna

js1972, 2019-11-28:
Is there much time delay when you add the kvssink element back in?

ManojBahuguna, 2019-11-28:
Yes there's 1-2 seconds of delay in the stream uploaded to kinesis.

js1972, 2019-12-09:
Any chance you could share the code how you did it,or even just pseudo code?

ManojBahuguna, 2019-12-10:
I took most of the code from here:
https://github.com/ViGLug/libre-streaming/blob/master/libre-streaming.py

shout2sendDown and shout2sendReconnect functions to be precise to pause and resume respectively.

Also, If you just want to just pause and play I think it's ok to destroy and recreate the whole pipeline each time.
But if you want to process the stream before sending (like I am buffering to add delay/latency to stream), then the above method will work.

I am also exploring some other ways like using appsink/appsrc or udpsink/udpsrc where I can just keep streaming at one pipeline and destroy/recreate the other one for this play/pause effect.

js1972, 2019-12-10:
Yeah the one to two seconds delay when you reconnect the kvssink element is
a bit annoying.

I’ve been testing with gstd (gstreamer daemon). It provides a simplified
api over gstreamer but has handy interpipesrc and interpipesink elements.
These are built on top of app src and app sink and allow dynamic switching
of the pipeline instantly.
But I’m using Nvidia Jenson hardware and they don’t support buffers on the
gpu so I have to use convert elements to “downgrade” the performance to
cross the interpipe bridge.
I’ve been testing with file sink and not kvssink and I expect kvssink will
still have the 1-2 second lag as it starts up.

lummish, 2020-08-06:
@js1972 Curious how you were able to get the probe solution to work. I'm currently doing the following and running into issues (I'll specify what these issues are in a few short moments).

- Define the element and pad that we want to attach our probe to. In my case, I've selected the sink pad of a `fakesink` element.
- Upon some event (let's say it's a simple timeout), create a new stream definition and stream. Start the stream using codec data acquired from pad caps, then immediately thereafter attach a buffer probe to the pad that we've selected.
- In our probe callback, execute the exact same operations performed by the `kvssink` element when a buffer is received (I've essentially copied and pasted these operations).

I haven't even gotten to the point of stopping and restarting the stream. After ~2 buffers, this probe will cause the pipeline to fail. If the probe is not attached but the pipeline otherwise remains the same, no issues whatsoever.

Here are the logs that are produced:

```
DEBUG - Creating IoT auth callbacks.
2020-08-06 13:44:00 [548493684752] INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2020-08-06 13:44:00 [548493684752] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2020-08-06 13:44:00 [548493684752] INFO - heapInitialize(): Creating AIV heap.
2020-08-06 13:44:00 [548493684752] INFO - heapInitialize(): Heap is initialized OK
2020-08-06 13:44:00 [548493684752] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-08-06 13:44:00 [548493684752] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000002, Next state: 0x0000000000000010
2020-08-06 13:44:00 [548493684752] INFO - createDeviceResultEvent(): Create device result event.
2020-08-06 13:44:00 [548493684752] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000010, Next state: 0x0000000000000040
2020-08-06 13:44:00 [548493684752] DEBUG - clientReadyHandler invoked
Now playing
(my-app:25966): GLib-CRITICAL **: 13:44:00.557: g_strrstr: assertion 'haystack != NULL' failed
Opening in BLOCKING MODE
Opening in BLOCKING MODE
Creating LL OSD context new
Playing at 1
Playing at 2
Playing at 3
Playing at 4
Playing at 1
Playing at 2
Playing at 3
Playing at 4
gstnvtracker: Loading low-level lib at /opt/nvidia/deepstream/deepstream-4.0/lib/libnvds_nvdcf.so
gstnvtracker: Optional NvMOT_RemoveStreams not implemented
gstnvtracker: Batch processing is ON
[NvDCF] Initialized
Running...
** INFO: <bus_callback:135>: Pipeline ready

NvMMLiteOpen : Block : BlockType = 261
NVMEDIA: Reading vendor.tegra.display-size : status: 6
NvMMLiteBlockCreate : Block : BlockType = 261
NvMMLiteOpen : Block : BlockType = 4
NvMMLiteOpen : Block : BlockType = 4
===== NVMEDIA: NVENC =====
===== NVMEDIA: NVENC =====
NvMMLiteBlockCreate : Block : BlockType = 4
NvMMLiteBlockCreate : Block : BlockType = 4
H264: Profile = 66, Level = 0
H264: Profile = 66, Level = 0
Opening in BLOCKING MODE
Creating LL OSD context new
** INFO: <bus_callback:121>: Pipeline running

INFO - Creating Kinesis Video Stream MyStream
2020-08-06 13:44:05 [548493684752] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): Kinesis Video Stream Info
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Stream name: MyStream
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Streaming type: STREAMING_TYPE_REALTIME
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Content type: video/h264
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Max latency (100ns): 600000000
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Fragment duration (100ns): 20000000
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Key frame fragmentation: Yes
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Use frame timecode: Yes
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Absolute frame timecode: Yes
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Nal adaptation flags: 0
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Average bandwith (bps): 4194304
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Framerate: 25
[... 11 lines cut ...]
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Track type: TRACK_INFO_TYPE_VIDEO
2020-08-06 13:44:05 [548493684752] DEBUG - logStreamInfo(): 	Track cpd: NULL
2020-08-06 13:44:05 [548493684752] DEBUG - stepStateMachine(): State Machine - Current state: 0x0000000000000001, Next state: 0x0000000000000002
2020-08-06 13:44:05 [548493684752] INFO - writeHeaderCallback(): RequestId: 9e8598d7-8e43-48eb-a3e5-2806971a8e13
2020-08-06 13:44:05 [544442130800] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.591916769633E9,"DataRetentionInHours":24,"DeviceName":null,"IngestionConfiguration":null,"KmsKeyId":"arn:aws:kms:***:***:alias/aws/kinesisvideo","MediaType":null,"Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:***:***:stream/MyStream/***","StreamName":"MyStream","Version":"IoErbt3EpS5RSQ9jiYCA"}}
2020-08-06 13:44:05 [544442130800] INFO - describeStreamResultEvent(): Describe stream result event.
2020-08-06 13:44:05 [544442130800] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
[... cut ...]
(my-app:25966): GStreamer-CRITICAL **: 13:44:05.968: gst_mini_object_unref: assertion 'GST_MINI_OBJECT_REFCOUNT_VALUE (mini_object) > 0' failed

(my-app:25966): GStreamer-CRITICAL **: 13:44:05.968: gst_mini_object_unlock: assertion 'GST_MINI_OBJECT_IS_LOCKABLE (object)' failed

(my-app:25966): GStreamer-CRITICAL **: 13:44:05.968: gst_mini_object_unref: assertion 'GST_MINI_OBJECT_REFCOUNT_VALUE (mini_object) > 0' failed
```

Attempts to backtrace the error have not proven fruitful. Any ideas what might be causing the issue? Were you able to get the producer SDK to work as a probe? What did you do differently?