awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #290: Clarification of DEFAULT_VIDEO_AUDIO_FRAME_RATE = 120

farleylai, 2019-11-14:
In the sample in `KvsVideoOnlyStreamingSample.c`, the `pStreamInfo.streamCaps` is set to the defaults by `createRealtimeVideoStreamInfoProvider(...)` and the resulting `pStreamInfo.streamCaps.frameRate` is `120` defined by `DEFAULT_VIDEO_AUDIO_FRAME_RATE`. 

Could you clarify the time scale of the frame rate number? 120 seems to high for video.
What if the actual encoder FPS is set to `15`, how do we set `pStreamInfo.streamCaps.frameRate` accordingly?

zhiyua-git, 2019-11-14:
Hi @farleylai ,

You can directly set the pStreamInfo.streamCaps.frameRate = 15; after calling createRealtimeVideoStreamInfoProvider(...).

MushMal, 2019-11-14:
@farleylai this is a bit of a misnomer really. The FPS value in stream caps primarily used for optimal content view buffer allocations - not much else. You can certainly leave these defaults and produce at 15 fps. You might end-up with 10kb more memory allocated for content view than needed.

Please close this issue if you are satisfied with the answer. You can open a separate one to discuss other topics though.

farleylai, 2019-11-14:
@MushMal
I think you mean the frame rate here has little to do with the encoder frame rate but more to do with the stream buffering or so, right?

MushMal, 2019-11-14:
Correct

closed 2019-11-14 by farleylai