awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #291: Failed to put a frame into the stream. StatusCode: 0x3200000c

Iorek73, 2019-11-15:
[log.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3849611/log.txt)

Hello, 

I'm trying to setup android to Kinesis Video Stream. I cloned the github for [android ](https://github.com/awslabs/aws-sdk-android-samples), followed [Android Producer Library setup instructions](https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-android.html), and double checked with [this instruction guide](https://console.aws.amazon.com/kinesisvideo/producerSetup/libraryType/android/downloadAndConfigure?region=us-east-1).

awsconfiguration.json is setup correctly (quintuple checked) and the KINESIS_VIDEO_REGION set to US_EAST_1, where both I am and everything else was setup.

The emulator (also tried on a physical android device) and app both boot correctly, and cognito user and identity pools run fine. After initiating the stream, the app registers a new stream with Kinesis Video Streams, but fails to transfer any data (error x3200000c ).

From the docs...
> x3200000c | STATUS_MKV_LARGE_FRAME_TIMECODE | The max frame timecode was reached. The MKV format uses signed 16 bits to represent the relative timecode of the frame to the beginning of the cluster. The error is generated if the frame timecode cannot be represented. This error indicates either a bad timecode scale selection or the cluster duration is too long, so representing the frame timecode overflows the signed 16-bit space.

Meanwhile, in Kinesis Video Stream, after waiting about 20s, the following pops up
```
No fragments found
No fragments were found. Verify that media is currently being streamed and that the fragment timecodes are correct.
```

What's happening here and how is this fixable? #210 touched on this, but I didn't understand the answer. Logs also attached.

zhiyua-git, 2019-11-15:
Hi @Iorek73 ,

The encoder behavior in emulator is different from actual device. To make it work in emulator you will need to add 
`if (mFrameIndex % mMediaSourceConfiguration.getFrameRate() == 0) {`
`            Bundle b = new Bundle();`
`            b.putInt(MediaCodec.PARAMETER_KEY_REQUEST_SYNC_FRAME, 0);`
`            mEncoder.setParameters(b);`
`}`
into https://github.com/aws-amplify/aws-sdk-android/blob/master/aws-android-sdk-kinesisvideo/src/main/java/com/amazonaws/mobileconnectors/kinesisvideo/encoding/EncoderWrapper.java#L138

You will need to follow the 9 steps in [this comment](https://github.com/aws-amplify/aws-sdk-android/issues/909#issuecomment-495007538) to build the changes.

Alternatively you can use actual android device, and it should work.

Thanks,
-ZH

Iorek73, 2019-11-15:
Thank you ZH! I just got it working on a physical device and was editing the post while you changed it. Greatly appreciate the fast response!

closed 2019-11-15 by Iorek73