awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #293: stopKinesisVideoStreamSync(): Synchronously stopping Kinesis Video Stream

saurav1404, 2019-11-22:
I am using jetson nano with realsense camera to stream the video. I am getting this error

[image]

zhiyua-git, 2019-11-22:
Hi @saurav1404 ,

According to the [guide](https://developer.download.nvidia.com/embedded/L4T/r32_Release_v1.0/Docs/Accelerated_GStreamer_User_Guide.pdf?QOntBL6v7uhyv7WNEjwRQEs5x36s7H6IIFEQHCyE_gXuaTn4qTmEegHvmFznWAk8vH6ouoqCXQ7MO4VH0I9HvnkD3hMsLor_1z1vkTHNkLv9_bhMdaoJNVlJYvm4ZRNWXPbU16iAD9JH48P7njosNwuCql4laoh6rdxvj3MlUyD8wB7JVn4) you will need to use this pipeline to run

gst-launch-1.0 nvarguscamerasrc ! \
 'video/x-raw(memory:NVMM), width=(int)1920, height=(int)1080, \
format=(string)NV12, framerate=(fraction)30/1' ! nvv4l2h264enc ! \
bitrate-8000000 ! h264parse ! kvssink stream-name=YourStreamName storage-size=128 access-key="YourAccessKey" secret-key="YourSecretKey"

Related post: https://devtalk.nvidia.com/default/topic/1050950/jetson-nano/h-264-h-265-encoding-using-jetson-nano-gpu/

If that doesn't work, you will need to look into how to get h264 frames from your device to pipe in kvssink.

saurav1404, 2019-11-26:
Thanks, Let me try this. I will get back to you.

zhiyua-git, 2020-01-12:
Closing the issue as it is gstreamer pipeline not providing h264 frame instead of producer SDK issue. Please open new issue about Producer SDK if any.

closed 2020-01-12 by zhiyua-git