awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #295: Media preview screen is black for Kinesis stream, Vivotek FD9187-HT camera

ahmedha, 2019-11-25:
Hello,

I setup Kinesis stream for the video from Vivotek FD9187-HT camera.  I am not seeing any errors from the "Teleport" package installed in the camera that is sending the video to the Kinesis stream.  In AWS console I see the "Media preview" window is black with the caption "No fragments found. No fragments were found. Verify that media is currently being streamed and that the fragment timecodes are correct."

Stream2 (which is streamed to Kinesis) setting in the camera:
- H.264
- Frame size - 640x480
- Maximum frame rate - 15 fps
- Intra frame period - 1s
- Smart Stream III
  - Smart codec / Mode - Auto Tracking
- Bit rate control
  - Constrained bit rate: Target Quality - Detailed, Maximum bit rate - 3 Mbps, Policy - Frame rate piority, Smart Q - On

- Can someone please help me find out if there is indeed data/video coming into the Kinesis stream?  Like some kind of ingest rate or such, or increasing stream size (based on frames stored in the backing S3)?  (I see activity in the latency graph, but would like additional confirmation, if possible).
- What additional debugging can I perform to figure out why the Media preview is not working.  Encoding issues, etc?

Would appreciate any help.

Thank you,
Ahmed.

MushMal, 2019-11-25:
@ahmedha Could you please check whether you are streaming into the same region as you are attempting to view? In the console, there is a region selection drop down

ahmedha, 2019-11-26:
> @ahmedha Could you please check whether you are streaming into the same region as you are attempting to view? In the console, there is a region selection drop down

Yes I verified that I am producing to region us-west-2, and attempting to view the media in us-west-2 also.
Since the latency graph is showing activity that would mean I am looking at the right stream, is that right?

MushMal, 2019-11-26:
Generally speaking, you have a customer side Cloud Watch metrics you can see whether it successfully ingests fragments. You could also check the device logs if you have access to it.

Could you please send the account id and the stream name to kinesis-video-support@amazon.com so we can look whether we can find any request ids. Please attempt to stream again and then send so we can see the latest

ahmedha, 2019-11-27:
After changing camera settings, prefixing the "IoT Credentials provider" field with "https://", I can now see the video in "Media preview" window. 

Thank you for helping out.  Please close this Issue.

closed 2019-11-27 by unicornss

tschoepflin, 2021-10-18:
Hi Mike
Does this work for you?
https://code.platform.aero/resources/ai/blob/8fa9e3184602e53e16461f4b1e6469d9ae0f9f29/teleport-v3.0.0.1.tar.gz