awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #297: How to add CPD information for kinesis-video-c-producer ?

laoyouji007, 2019-11-28:
I use KvsVideoOnlyStreamingSample.c with adding AAC audio track data, but there is an error "Missing codec private data error" when displaying on browser. Maybe I need to add CPD information, but I have no idea on how to add CPD, Can someone know how to do this ?

zhiyua-git, 2019-11-29:
Hi @laoyouji007 ,

We don’t have a sample for sending audio video in C producer, because it is roughly same as the video frame sample.
For sending audio video you will need to
1. Replace https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c#L93 with one of https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/424580b800087f05f80000d30b43b61e5a964c29/kinesis-video-c-producer/src/source/StreamInfoProvider.c#L25-L40 depending on your use case.
2. Add your audio codec private data after https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-c-producer/samples/KvsVideoOnlyStreamingSample.c#L94
pStreamInfo->streamCaps.trackInfoList[1].codecPrivateDataSize = your_cpd_size;
pStreamInfo->streamCaps.trackInfoList[1].codecPrivateData = your_codec_private_data;
You can also add your video track codec private data in similar way. The code created both tracks is https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/424580b800087f05f80000d30b43b61e5a964c29/kinesis-video-c-producer/src/source/StreamInfoProvider.c#L195-L198
3. When you do putFrame(), send video frames to DEFAULT_VIDEO_TRACK_ID and audio frames to DEFAULT_AUDIO_TRACK_ID.
 
One thing to remind beforehand is backend will validate if one fragment sent contains both tracks, so if you want to stop sending video or audio but still send the other track, you will need to stop and free the stream and recreate one stream as video only sample to do the video only streaming.

Hope it helps.

laoyouji007, 2019-11-30:
@zhiyua-git 
Thank you very much ! 
I added the relevant code according to your requirements above，but the codecPrivateData of video and audio may not be correct，so I get the error `Failed with status 0x32000010` . Could you thell me where can I get the right cpd of audio and video ? 

My code is as follows：
const char audio_cpd_s[] = {"**1588**"};
const char video_cpd_s[] = {"**014d4016ffe10014674d40169da8280bfe58400000030040000007a101000468ee3c80**"};
char *cpd_video = NULL;
char *cpd_audio = NULL;

    //set video CPD for stream;
    CHK_STATUS(hexDecode((PCHAR)video_cpd_s, NULL, &size));
    cpd_video = malloc(size);
    if (cpd_video == NULL) {
        printf ("error: malloc for hexDecode video_cpd fail...\n");
        goto CleanUp;
    }
    CHK_STATUS(hexDecode((PCHAR)video_cpd_s, cpd_video, &size));
    for (i = 0; i < 2; i++) {
        if (pStreamInfo->streamCaps.trackInfoList[i].trackId == DEFAULT_VIDEO_TRACK_ID) {
            printf ("set video cpd ...\n");
            pStreamInfo->streamCaps.trackInfoList[i].codecPrivateDataSize = size;
            pStreamInfo->streamCaps.trackInfoList[i].codecPrivateData = cpd_video;
            break;
        }
    }

    //set audio CPD for stream;
    CHK_STATUS(hexDecode((PCHAR)audio_cpd_s, NULL, &size));
    cpd_audio = malloc(size);
    if (cpd_audio == NULL) {
        printf ("error: malloc for hexDecode audio_cap fail...\n");
        goto CleanUp;
    }
    CHK_STATUS(hexDecode((PCHAR)audio_cpd_s, cpd_audio, &size));
    for (i = 0; i < 2; i++) {
        if (pStreamInfo->streamCaps.trackInfoList[i].trackId == DEFAULT_AUDIO_TRACK_ID) {
            printf ("set audio cpd ...\n");
            pStreamInfo->streamCaps.trackInfoList[i].codecPrivateDataSize = size;
            pStreamInfo->streamCaps.trackInfoList[i].codecPrivateData = cpd_audio;
            break;
        }
    }

zhiyua-git, 2019-11-30:
Hi,

As your video CPD is in avcc format, you will need 
pStreamInfo->streamCaps.nalAdaptationFlags = NAL_ADAPTATION_ANNEXB_NALS; instead of both here https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/424580b800087f05f80000d30b43b61e5a964c29/kinesis-video-c-producer/src/source/StreamInfoProvider.c#L115

For error details, you can refer to https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-sdk-errors.html#producer-sdk-errors-mkvgen

laoyouji007, 2019-11-30:
@zhiyua-git 
Hi, Thanks for helping me !

I change the `nalAdaptationFlags` , my program run, but it will get a "Segmentation fault" mistake,  What shall I do now ? Can you give me more suggestions ?

The log of my program:

2019-11-30 09:21:15 writeHeaderCallback(): RequestId: 0c3c551f-e925-4924-915a-cf7f314ba00c
2019-11-30 09:21:15 getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-w
est-2.amazonaws.com"}

2019-11-30 09:21:15 getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2019-11-30 09:21:15 stepStateMachine(): PIC Stream State Machine - Current state: 0x0, Next state: 0x20
2019-11-30 09:21:15 stepStateMachine(): PIC Client State Machine - Current state: 0x7785f254, Next state: 0x40
2019-11-30 09:21:15 getStreamingTokenResultEvent(): Get streaming token result event.
2019-11-30 09:21:15 stepStateMachine(): PIC Stream State Machine - Current state: 0xf0, Next state: 0x10
2019-11-30 09:21:15 **Segmentation fault**

laoyouji007, 2019-11-30:
@zhiyua-git 
Hi, Thank you very much ! My program works OK now !

laoyouji007, 2019-11-30:
@zhiyua-git 
Thank you ! I have got the correct video CPD, but in trouble with getting CPD of audio. My audio AAC format is: AACLC, sampling rate is 8kHz, 16 bit, mono, what kind of CPD array should this audio format do?

zhiyua-git, 2019-11-30:
Your CPD 0x1588 looks correct for a 8kHz 1 channel audio. Can you clarify what is wrong with it?

laoyouji007, 2019-12-01:
@zhiyua-git 
Hello ! Thank you !

When I display on browser after runnig my program, it got an error of "Fragment decoding error", just like below：

Fragment decoding error
There was an error decoding the media. Verify that the stream contains valid content.
RetryHide details
Codec ID:
avc1.4d4016 (H.264 Main@2.2)
mp4a.40.2 (AAC LC)

Fragment number: 91343852333181858249007336952685937813058863131
Producer timestamp: December 1, 2019 12:00 AM CST
Server timestamp: December 1, 2019 12:00 AM CST
Size: 61.0 KB
This may be an intermittent issue. Retry the same action, or try refreshing your browser tab.

MushMal, 2019-12-01:
@laoyouji007 are you able to stream video only from this source with that video CPD? Extracting your CPD it seems it's in AvCC format already so it should work. If the issue is the Audio then we need to check whether your browser supports AAC audio.

laoyouji007, 2019-12-01:
@MushMal 
Thanks. 
I'm able to stream video only with my video CPD, and display on google browser correctly. But if I add audio together, the google browser return the error above. Can google browser supports AAC audio ?

MushMal, 2019-12-01:
Not entirely sure why the Chrome browser fails decoding it. There seem to be pretty good support for it checking https://caniuse.com/#search=AAC

Could you please share some more info on what platform you are using it and what versions of the platform/browser you are using

laoyouji007, 2019-12-01:
@MushMal 
 Hello, Here is my platform information:
system: win10
Firefox Browser version: 70.0.1 (32 bit)
Chrome browser version: 76.0.3809.100 (64 bit)

chehefen, 2019-12-02:
Hi @laoyouji007 , can you make sure that you dont put any audio frames until you have put a video key frame?

laoyouji007, 2019-12-03:
@chehefen 
Hello !
I am sorry for replying late, I'm in BeiJing time which have a 16 hour time difference with you.
Yes, I'm sure that I send a key frame of video first, and then follow others audio/video BP frame.

chehefen, 2019-12-03:
can you do `export KVS_DEBUG_DUMP_DATA_FILE_DIR=/path/to/dir` and then start streaming audio video? It should then generate a mkv file at the specified directory. Can you send that mkv to kinesis-video-support@amazon.com for us to take a look? Thanks

laoyouji007, 2019-12-03:
@chehefen 
Hello, I just send an email to [kinesis-video-support@amazon.com](url) with a kvs_0.mkv file, Please check it for me, thank you !

chehefen, 2019-12-03:
The mkv looks fine to me, did you see any errors from the producer log? Can you send us the complete producer log when the decoding error happens?

laoyouji007, 2019-12-03:
@chehefen 
Hello, there isn't any error when running my application, the complete producer log has been sent as a attachment in email to [kinesis-video-support@amazon.com](url), please check it for me, Thanks !

chehefen, 2019-12-04:
do you mind sending us the test.mp4 file you use to test?

laoyouji007, 2019-12-05:
Hi, I have send an email to [kinesis-video-support@amazon.com](url) with mp4 file, please check for me, thank you !

chehefen, 2019-12-05:
@laoyouji007 if you open up the mp4 file in a text editor, you will see that there is a big block of "2d2d" sitting in place of the first audio frame. So the decoder tried to decode that block as an audio frame and failed.

chehefen, 2019-12-05:
is your camera android based?

laoyouji007, 2019-12-06:
@chehefen 
Hello, My video and audio are collected and encoded by arm based SoC, not in android based. I will check the audio data, Thank you !

zhiyua-git, 2019-12-09:
Hi @laoyouji007 ,

Can you check if you are adding ADTS header to the AAC audio frames? KVS only accepts and play raw AAC frames. Thanks.

laoyouji007, 2019-12-10:
@zhiyua-git 
Thank you very much ! I remove the ADTS header, and the audio can play now !

zhiyua-git, 2019-12-10:
Hi @laoyouji007 ,

Good to hear it is working for you! 

Closing the issue as it is resolved.

closed 2019-12-10 by zhiyua-git