awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #298: ./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0

JZamora-Sepulveda, 2019-12-02:
Hi, 

I am trying to stream from a RTSP camera to KVS but im getting the same error don't matter what i try. 

I'm using the commands provides by this guide : https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md

I try with two options: 

1) Running the GStreamer RTSP sample application

AWS_ACCESS_KEY_ID=@1 AWS_SECRET_ACCESS_KEY=@2 AWS_DEFAULT_REGION="us-east-2" gst-launch-1.0 rtspsrc location=rtsp://user:key@1.1.1.1:1 ! rtph264depay ! h264parse ! kvssink stream-name=@3 storage-size=512

With that i got the following error message: 

Estableciendo el conducto a PAUSA …
[INFO ][2019-11-29 13:51:47] createKinesisVideoClient(): Creating Kinesis Video Client
[INFO ][2019-11-29 13:51:47] heapInitialize(): Initializing native heap with limit size 536870912, spill ratio 0% and flags 0x00000001
[INFO ][2019-11-29 13:51:47] heapInitialize(): Creating AIV heap.
[INFO ][2019-11-29 13:51:47] heapInitialize(): Heap is initialized OK
[DEBUG][2019-11-29 13:51:47] stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-11-29 13:51:47] getSecurityTokenHandler invoked
[DEBUG][2019-11-29 13:51:47] Refreshing credentials. Force refreshing: 0 Now time is: 1575035507886574972 Expiration: 0
[DEBUG][2019-11-29 13:51:47] stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
[INFO ][2019-11-29 13:51:47] createDeviceResultEvent(): Create device result event.
[DEBUG][2019-11-29 13:51:47] stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-11-29 13:51:47] clientReadyHandler invoked
[INFO ][2019-11-29 13:51:47] try creating stream
[INFO ][2019-11-29 13:51:47] Creating Kinesis Video Stream ExampleStream
[INFO ][2019-11-29 13:51:47] createKinesisVideoStream(): Creating Kinesis Video Stream.
[DEBUG][2019-11-29 13:51:47] stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
[DEBUG][2019-11-29 13:51:47] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-11-29 13:51:48] writeHeaderCallback(): RequestId: c7f83b22-1c02-4f33-a65b-e2cfc42f71bd
[DEBUG][2019-11-29 13:51:48] describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.57497025829E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-east-2:453087806180:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-east-2:453087806180:stream/ExampleStream/1574970258290","StreamName":"ExampleStream","Version":"gnFNrz6trsjYBLO4hWGo"}}
[INFO ][2019-11-29 13:51:48] describeStreamResultEvent(): Describe stream result event.
[DEBUG][2019-11-29 13:51:48] stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
[DEBUG][2019-11-29 13:51:48] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[INFO ][2019-11-29 13:51:49] writeHeaderCallback(): RequestId: 6238c70c-599a-4137-aecf-d904b6be44de
[DEBUG][2019-11-29 13:51:49] getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-171c51c7.kinesisvideo.us-east-2.amazonaws.com"}
[INFO ][2019-11-29 13:51:49] getStreamingEndpointResultEvent(): Get streaming endpoint result event.
[DEBUG][2019-11-29 13:51:49] stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
[DEBUG][2019-11-29 13:51:49] stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
[DEBUG][2019-11-29 13:51:49] getStreamingTokenHandler invoked
[DEBUG][2019-11-29 13:51:49] Refreshing credentials. Force refreshing: 1 Now time is: 1575035509747355201 Expiration: 18446744073709551615
[INFO ][2019-11-29 13:51:49] getStreamingTokenResultEvent(): Get streaming token result event.
[DEBUG][2019-11-29 13:51:49] stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
[DEBUG][2019-11-29 13:51:49] defaultStreamReadyCallback(): Reported streamReady callback for stream handle 94718034576281
[DEBUG][2019-11-29 13:51:49] streamReadyHandler invoked
Stream is ready
El conducto está vivo y no necesita PREPARARSE …
Progreso: (open) Opening Stream
Progreso: (connect) Connecting to rtsp://user:pass@1.1.1.1:1
Progreso: (open) Retrieving server options
Progreso: (open) Retrieving media info
Progreso: (request) SETUP stream 0
Progreso: (open) Opened Stream
Estableciendo el conducto a REPRODUCIENDO …
New clock: GstSystemClock
Progreso: (request) Sending PLAY request
Progreso: (request) Sending PLAY request
Progreso: (request) Sent PLAY request
ADVERTENCIA: del elemento /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: El enlazado retardado falló.
Información adicional de depuración:
./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
ERROR: del elemento /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1: Internal data stream error.
Información adicional de depuración:
gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc1:
streaming stopped, reason not-linked (-1)
Execution ended after 0:00:01.315634968
Estableciendo el conducto a PAUSA …
[... 85 lines cut ...]
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager/GstRtpPtDemux:rtpptdemux0.GstPad:sink: caps = application/x-rtp, media=(string)video, payload=(int)96, clock-rate=(int)90000, encoding-name=(string)H265, sprop-sps=(string)"QgEBAWAAAAMAsAAAAwAAAwB7oAPAgBDlja5JMvTcBAQEAg\=\=", sprop-pps=(string)"RAHA8vA8kAA\=", a-recvonly=(string)"", x-dimensions=(string)"1920\,1080", a-Media_header=(string)"MEDIAINFO\=494D4B48010200000400050000000000000000000000000000000000000000000000000000000000\;", a-appversion=(string)1.0, ssrc=(uint)1345022211, clock-base=(uint)1204874172, seqnum-base=(uint)53217, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
ADVERTENCIA: del elemento /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: El enlazado retardado falló.
Información adicional de depuración:
./grammar.y(510): gst_parse_no_more_pads (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
failed delayed linking some pad of GstRTSPSrc named rtspsrc0 to some pad of GstRtpH264Depay named rtph264depay0
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0.GstGhostPad:recv_rtp_src_0_1345022211_96.GstProxyPad:proxypad4: caps = application/x-rtp, media=(string)video, payload=(int)96, clock-rate=(int)90000, encoding-name=(string)H265, sprop-sps=(string)"QgEBAWAAAAMAsAAAAwAAAwB7oAPAgBDlja5JMvTcBAQEAg\=\=", sprop-pps=(string)"RAHA8vA8kAA\=", a-recvonly=(string)"", x-dimensions=(string)"1920\,1080", a-Media_header=(string)"MEDIAINFO\=494D4B48010200000400050000000000000000000000000000000000000000000000000000000000\;", a-appversion=(string)1.0, ssrc=(uint)1345022211, clock-base=(uint)1204874172, seqnum-base=(uint)53217, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
/GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstRtpBin:manager.GstGhostPad:recv_rtp_src_0_1345022211_96.GstProxyPad:proxypad3: caps = application/x-rtp, media=(string)video, payload=(int)96, clock-rate=(int)90000, encoding-name=(string)H265, sprop-sps=(string)"QgEBAWAAAAMAsAAAAwAAAwB7oAPAgBDlja5JMvTcBAQEAg\=\=", sprop-pps=(string)"RAHA8vA8kAA\=", a-recvonly=(string)"", x-dimensions=(string)"1920\,1080", a-Media_header=(string)"MEDIAINFO\=494D4B48010200000400050000000000000000000000000000000000000000000000000000000000\;", a-appversion=(string)1.0, ssrc=(uint)1345022211, clock-base=(uint)1204874172, seqnum-base=(uint)53217, npt-start=(guint64)0, play-speed=(double)1, play-scale=(double)1
ERROR: del elemento /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0: Internal data stream error.
Información adicional de depuración:
gstbasesrc.c(2939): gst_base_src_loop (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0/GstUDPSrc:udpsrc0:
streaming stopped, reason not-linked (-1)
Execution ended after 0:00:00.647177103
Estableciendo el conducto a PAUSA …
Estableciando el conducto a PREPARADO …
Estableciendo el conducto a NULL …
Liberando la tubería…
[INFO ][2019-12-02 15:13:45] Freeing Kinesis Video Stream CameraStream
[INFO ][2019-12-02 15:13:45] freeKinesisVideoStream(): Freeing Kinesis Video stream.
[DEBUG][2019-12-02 15:13:46] defaultStreamShutdownCallback(): Reported streamShutdown callback for stream handle 94068061396521

Maybe i'm missing a step, quotes, something... i will appreciate if someone can help me with this, i'm new at this and i really don't know what may be missing... in some places i found that it may be a error of the Gstream version with RTSP conecction, but still don't know how to fix it.

The camera i'm trying to use is a Hikvision model DS-2CD2621G0-IZS.

Thank you!

chehefen, 2019-12-02:
You rtsp source uses h265 encoding. Use this pipeline instead:
```
gst-launch-1.0 -v rtspsrc location=rtsp://user:pass@1.1.1.1:1 short-header=TRUE ! rtph265depay ! h265parse ! kvssink stream-name=@1 storage-size=128 access-key=@2 secret-key=@3
```

JZamora-Sepulveda, 2019-12-03:
Thanks for the answer!

I changed the encoding of the RTSP source to h264 before i try the pipelines from the previous comment. Still try the pipeline you provide and here is the result: 

gst-launch-1.0 -v rtspsrc location=rtsp://user:pass@1.1.1.1:111 short-header=TRUE ! rtph265depay ! h265parse ! kvssink stream-name=@1 storage-size=128 access-key=@2 secret-key=@3

Estableciendo el conducto a PAUSA …
log4cplus:ERROR could not open file ./kvs_log_configuration
INFO - createKinesisVideoClient(): Creating Kinesis Video Client
2019-12-03 09:56:10 [139736858847040] INFO - heapInitialize(): Initializing native heap with limit size 134217728, spill ratio 0% and flags 0x00000001
2019-12-03 09:56:10 [139736858847040] INFO - heapInitialize(): Creating AIV heap.
2019-12-03 09:56:10 [139736858847040] INFO - heapInitialize(): Heap is initialized OK
2019-12-03 09:56:10 [139736858847040] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x1, Next state: 0x2
2019-12-03 09:56:10 [139736858847040] DEBUG - getSecurityTokenHandler invoked
2019-12-03 09:56:10 [139736858847040] DEBUG - Refreshing credentials. Force refreshing: 0 Now time is: 1575377770845490934 Expiration: 0
2019-12-03 09:56:10 [139736858847040] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x2, Next state: 0x10
2019-12-03 09:56:10 [139736858847040] INFO - createDeviceResultEvent(): Create device result event.
2019-12-03 09:56:10 [139736858847040] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x10, Next state: 0x40
2019-12-03 09:56:10 [139736858847040] DEBUG - clientReadyHandler invoked
2019-12-03 09:56:10 [139736858847040] INFO - try creating stream
2019-12-03 09:56:10 [139736858847040] INFO - Creating Kinesis Video Stream CameraStream
2019-12-03 09:56:10 [139736858847040] INFO - createKinesisVideoStream(): Creating Kinesis Video Stream.
2019-12-03 09:56:10 [139736858847040] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x1, Next state: 0x2
2019-12-03 09:56:10 [139736858847040] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-12-03 09:56:10 [139736858847040] INFO - writeHeaderCallback(): RequestId: 548c5e52-7276-4e3a-b2bc-edd176b1a091
2019-12-03 09:56:11 [139736604518144] DEBUG - describeStreamCurlHandler(): DescribeStream API response: {"StreamInfo":{"CreationTime":1.574695712286E9,"DataRetentionInHours":2,"DeviceName":"Kinesis_Video_Device","KmsKeyId":"arn:aws:kms:us-west-2:453087806180:alias/aws/kinesisvideo","MediaType":"video/h264","Status":"ACTIVE","StreamARN":"arn:aws:kinesisvideo:us-west-2:453087806180:stream/CameraStream/1574695712286","StreamName":"CameraStream","Version":"FyWq0I4wkpNCBkzSbIr6"}}
2019-12-03 09:56:11 [139736604518144] INFO - describeStreamResultEvent(): Describe stream result event.
2019-12-03 09:56:11 [139736604518144] WARN - describeStreamResult(): Content type returned from the DescribeStream call doesn't match the one specified in the StreamInfo
2019-12-03 09:56:11 [139736604518144] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x2, Next state: 0x20
2019-12-03 09:56:11 [139736604518144] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-12-03 09:56:11 [139736604518144] INFO - writeHeaderCallback(): RequestId: 7f5d5785-a64c-436f-91f7-028282132f8a
2019-12-03 09:56:12 [139736596125440] DEBUG - getStreamingEndpointCurlHandler(): GetStreamingEndpoint API response: {"DataEndpoint":"https://s-4010bf70.kinesisvideo.us-west-2.amazonaws.com"}
2019-12-03 09:56:12 [139736596125440] INFO - getStreamingEndpointResultEvent(): Get streaming endpoint result event.
2019-12-03 09:56:12 [139736596125440] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x20, Next state: 0x10
2019-12-03 09:56:12 [139736596125440] DEBUG - stepStateMachine(): PIC Client State Machine - Current state: 0x40, Next state: 0x40
2019-12-03 09:56:12 [139736596125440] DEBUG - getStreamingTokenHandler invoked
2019-12-03 09:56:12 [139736596125440] DEBUG - Refreshing credentials. Force refreshing: 1 Now time is: 1575377772828521120 Expiration: 18446744073709551615
2019-12-03 09:56:12 [139736596125440] INFO - getStreamingTokenResultEvent(): Get streaming token result event.
2019-12-03 09:56:12 [139736596125440] DEBUG - stepStateMachine(): PIC Stream State Machine - Current state: 0x10, Next state: 0x40
2019-12-03 09:56:12 [139736596125440] DEBUG - defaultStreamReadyCallback(): Reported streamReady callback for stream handle 93917669647385
2019-12-03 09:56:12 [139736596125440] DEBUG - streamReadyHandler invoked
2019-12-03 09:56:12 [139736596125440] Stream is ready
El conducto está vivo y no necesita PREPARARSE …
Progreso: (open) Opening Stream
Progreso: (connect) Connecting to rtsp://user:pass@111.111.1.111:111
Progreso: (open) Retrieving server options
Progreso: (open) Retrieving media info
ERROR: del elemento /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0: Unauthorized
Información adicional de depuración:
gstrtspsrc.c(5580): gst_rtspsrc_send (): /GstPipeline:pipeline0/GstRTSPSrc:rtspsrc0:
Unauthorized (401)
ERROR: el conducto no quiere prepararse.
Estableciendo el conducto a PAUSA …
Estableciando el conducto a PREPARADO …
Estableciendo el conducto a NULL …
Liberando la tubería…

It still having pro

MushMal, 2019-12-03:
As far as my Spanish allows me to understand it complains about your credentials. Could you check your rtsp credentials. You should also try GStreamer forums for more help on non-KVS related issues. I am going to mark this as resolved. Please open a separate ticket for any KVS related issues

closed 2019-12-03 by MushMal

git-hamza, 2023-08-23:
> You rtsp source uses h265 encoding. Use this pipeline instead:
> 
> ```
> gst-launch-1.0 -v rtspsrc location=rtsp://user:pass@1.1.1.1:1 short-header=TRUE ! rtph265depay ! h265parse ! kvssink stream-name=@1 storage-size=128 access-key=@2 secret-key=@3
> ```

I changed the rstp video compression to h264 and it started working. Thankyou