awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #301: SEGV alignment trap in 2.1.0 when freeKinesisVideoStream is called

ek9852, 2019-12-05:
SEGV with alignment trap in ARM 32bits platform when calling  freeKinesisVideoStream
==1404==ERROR: AddressSanitizer: SEGV on unknown address 0xafa07a41 (pc 0xb6976028 bp 0xafa00200 sp 0xa79fcaf0 T5)
    #0 0xb6976027 in semaphoreFree (/root/n/libcproducer.so+0x6f027)
    #1 0xb6960d73 in freeStream (/root/n/libcproducer.so+0x59d73)
    #2 0xb69534c1 in freeKinesisVideoStream (/root/n/libcproducer.so+0x4c4c1)

Turned on alignment fix in kernel (/proc/cpu/alignment) does not help.
This does not happen/crash in previous release.

Why we need to turn on "#pragma pack(push, include_i, 1) // for byte alignment"
for internal structures. There are many alignment trap fix in kernel, but kernel will not fix all kinds of  alignment  trap.

Tried to remove #pragma pack(push, include_i, 1) in kinesis-video-pic/src/client/src/Include_i.h.
But got   stack-overflow error.

MushMal, 2019-12-05:
@ek9852 the SDK currently doesn't support aligned memory access. It's not a simple fix to remove the tight packing. Many of the SDK operations like MKV generator, heap management, tight storage packing are designed to work with unaligned memory access.

We are looking in the future to support aligned memory access only.

ek9852, 2019-12-05:
There is nothing wrong about using  unaligned struct  for data storage or communication.
But the  problem is that nearly all Include_i.h had the #pragma pack(push, include_i, 1) line , which  makes all structures to be on byte  alignment even with internal struct which are not used for storage or communication. This creates problem when passing a pointer member variable from packed struct to another function, since the compiler will not know that pointer is unaligned  And Linux kernel in arm32 bits does not fixes all kinds of alignment issues in user space.

Got  workaround by wrap the struct __KinesisVideoBase  with:
#pragma pack(pop, include_i)  and 
#pragma pack(push, include_i, 1)

This makes  __KinesisVideoBase  unpacked and temporary solve  the problem.
It is unnecessary to make __KinesisVideoBase packed.
Not sure any other code path will trigger another alignment trap though.

MushMal, 2019-12-05:
@ek9852 you are correct. There are few structures which are public which need to be aligned right now but as I said, simply removing the pragma packs will not help. We need to properly address this issue. The devices that can't support unaligned memory access even with the kernel trap will still be broken due to MKV generator, heap, etc.. We will prioritize this work

MushMal, 2020-03-17:
The KVS SDK can be run in aligned-only memory mode.

closed 2020-03-17 by MushMal

sharadrj, 2020-03-17:
@MushMal, can you share instructions here for building/running in aligned-only memory mode?

MushMal, 2020-03-17:
Sure, generally, we removed any tight packing in favor of natural packing. The only exception is the content store in PIC. For the devices that have aligned only access without a kernel trap/fix-up, the customer should specify ALIGNED_MEMORY_MODEL CMAKE variable as described in the readme: https://github.com/awslabs/amazon-kinesis-video-streams-pic/blob/master/README.md#note-on-alignment