awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #302: If "KinesisVideoProducer :: createSync" is called continuously, the process does not end.

HigashidaTomoyuki, 2019-12-06:
I'm using python ctypes to call "kinesis_video_gstreamer_audio_video_sample_app.cpp" with some modifications. CustomData is defined as global.
If putFrame fails, it terminates with an error. If freeStream is executed at that time, it will not end forever, so it will end without freeStream.
If you then call the same process starting with “KinesisVideoProducer :: createSync”, createSync will not finish.

I am using the following environment.
raspberry pi
SDK 2.0.2

Please help.

MushMal, 2019-12-06:
@HigashidaTomoyuki by default, the KVS stream is configured with continuous retry logic. Most of the errors will be retried by default. 

Could you please provide some more information about what the error is if you are simply trying the default behavior and why does it not re-stream?

sankyo-toshio, 2019-12-07:
Thank you @MushMal. I'm co-worker of @HigashidaTomoyuki.
I will answer instead of @HigashidaTomoyuki.
We send only audio data stream which encoded to flac without video stream to KVS.
We need to notify to the user in the cases like the network environment affects to audio stream.

And only when we got the callback of `streamErrorReportHandler`,  it happens that `data.kinesis_video_stream` can't close. (because of the sdk retrying?)
Our products sometimes need to create stream and sometimes doesn't need to create stream. 
So, when the users need to do next actions (like starting another stream), we think we should close the previous stream and start another stream.

And we have to notify the users when we can't provide audio stream, and we thought that `Invalid mkv data` was not recoverable, so at that time, we close the stream.

MushMal, 2019-12-07:
@sankyo-toshio thanks for sharing. Allow me to provide some thoughts. From what I understand, you have a scenario which is known as an "intermittent" producer scenario. In this case, the application gets a trigger (say a user presses a button or some event happens). The application then creates the stream, streams, stops the stream and frees it. This then happens again on the next event. You can still have a single client object, though, creating and freeing the client is really fast (in case you only have to deal with a single stream).

Next, you can opt to use the sync or async API. If you use sync api like createSync, stopSync your thread will block for createSync until the state machinery reaches to "ready" state which is when the stream object is done describe call, getting endpoint, etc... On stopSync, the SDK will attempt to send out the remaining frames in the buffer and receive the last ACK from the backend before returning. Both can time out - there is a default configuration which you can override in the DeviceInfo.ClientInfo.

Regarding the retries. The C++ SDK is a thin OO layer on top of C-SDK which provides the networking layer, credential providers, etc... together with continuous error handling. C-SDK is then layered on top of PIC (Platform Independent Codebase) which is the core business logic, storage, temporal logic, etc..  The PIC has a retry logic in itself but after MAX retries (some states are configured with progressive fallback with up-to 5 retries) it will return an error. The C-SDK layer then analyzes the error and re-tries on many of them. There are, however, errors which are not re-triable. For example, it makes no sense to retry authentication errors. In case of the "bad" fragments like INVALID_MKV_DATA, the SDK will skip over the fragment as it's deemed as "bad" by the backend. In other cases, the SDK will attempt to rollback and retry to ensure the frames are properly received.
Invalid MKV data is likely caused when the buffer is getting gradually filled and the frames are started getting dropped and partial frames get sent. This is what's known in signal processing as "po-go stick effect" where the streaming bounces off in the "tail" of the buffer without being able to catch up.
In general, we would recommend you to avoid dropping frames by reacting to the stream when on stream latency pressure is called. This is all configurable so before the frames start dropping you could trigger re-set of the stream (which will still keep the buffer) or entirely re-start the stream which would dump the content of the buffer.

To be able to help you more, I would request you to send detailed configuration you use (StreamInfo or in case of C++ StreamDefinition) and device info to ensure you have optimal settings. Also, send the captured logs where you see the errors (lots of logs are OK). You can package it and send it to kinesis-video-support@amazon.com

Looking forward to help.

HigashidaTomoyuki, 2019-12-09:
@MushMal Thank you for your answer.
> In general, we would recommend you to avoid dropping frames by reacting to the stream when on stream latency pressure is called. This is all configurable so before the frames start dropping you [...]

Does the above mean that when you receive a storageOverflowPressure callback, reconfigure the stream or restart the stream that dumps the contents of the buffer completely?

MushMal, 2019-12-09:
@HigashidaTomoyuki storasgeOverflowPressure is fired when the SDK is about to run out of the storage area. This can be also handled by re-setting or re-starting the stream but in this case we would recommend having a larger content store to accommodate the max buffer pressure. Here is a doc explaining how to calculate the content store size: https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-structures-producer.html

Size = NumberOfStreams * AverageFrameSize * FramesPerSecond * BufferDurationInSeconds * DefragmentationFactor

What I was referring to is the following callback: https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L1331

sankyo-toshio, 2019-12-12:
@MushMal 
Thank you. 
Your `storasgeOverflowPressure` is mean `getStorageOverflowPressureCallback` of this page?
`https://docs.aws.amazon.com/en_us/kinesisvideostreams/latest/dg/producer-reference-callbacks.html`

In this page,  `storasgeOverflowPressure` and `StreamLatencyPressure`  are seemed to be different.

If they are different, this your link is right?
https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/kinesis-video-pic/src/client/include/com/amazonaws/kinesis/video/client/Include.h#L1331

Or, they are same meanings?

We tried to stream audio file long time, but after 3 or 4 hours `defaultStreamLatencyPressureCallback()` must occur, and begun to drop the frames.

Can we handle this problem by setting appropriate parameters? (such as max_latency)

Or we are thinking that if we can't get the received ack which has timecode and `StreamLatencyPressure` or other errors occur, we reset and start the stream and retry to send the buffer from timecode which can't reived ack.
This error handling is bad? You have any ideas?

MushMal, 2019-12-12:
Those are different callbacks. The storage overflow happens when the is a pressure on the content store - aka, we are running out of storage space. The latency pressure happens when we are getting high latency in the stream - both can be an indication of throughput problems. 

Generally, you would want to ensure there is enough storage to store max buffer duration per https://docs.aws.amazon.com/kinesisvideostreams/latest/dg/producer-reference-structures-producer.html the size can be selected:
Size = NumberOfStreams * AverageFrameSize * FramesPerSecond * BufferDurationInSeconds * DefragmentationFactor

You need to understand why you are dropping the frames - is this due to temporary network conditions or your bandwidth in general does not allow the density of the stream which would be odd as we are talking about audio.

If it proves to be a temporary network connection hiccup then you can increase the buffer duration and max latency.

If you are controlling your media pipeline, you could try to lower the frame rate during the latency pressure to allow the buffer to clear. Or, you could reset the connection to allow a new connection to be faster or reset the entire stream which would dump the buffer and reconnect. This said, the default continuous stream callbacks should handle it.

You haven't mentioned which asset you are using.

sankyo-toshio, 2019-12-13:
@MushMal 
Thank you.

> You haven't mentioned which asset you are using
Sorry, I can't understand this meanings.
We use
```
raspberry pi
kinesis video producer SDK 2.0.2
```

<details><summary>our parameters</summary><div>

```
#define DEFAULT_RETENTION_PERIOD_HOURS 2
#define DEFAULT_KMS_KEY_ID ""
#define DEFAULT_STREAMING_TYPE STREAMING_TYPE_REALTIME
#define DEFAULT_CONTENT_TYPE "audio/flac"
#define DEFAULT_MAX_LATENCY_SECONDS 150
#define DEFAULT_FRAGMENT_DURATION_MILLISECONDS 4000
#define DEFAULT_TIMECODE_SCALE_MILLISECONDS 1
#define DEFAULT_KEY_FRAME_FRAGMENTATION TRUE
#define DEFAULT_FRAME_TIMECODES TRUE
#define DEFAULT_ABSOLUTE_FRAGMENT_TIMES FALSE
#define DEFAULT_FRAGMENT_ACKS TRUE
#define DEFAULT_RESTART_ON_ERROR TRUE
#define DEFAULT_RECALCULATE_METRICS TRUE
#define DEFAULT_STREAM_FRAMERATE 100
#define DEFAULT_AVG_BANDWIDTH_BPS (4 * 1024 * 1024)
#define DEFAULT_BUFFER_DURATION_SECONDS 300
#define DEFAULT_REPLAY_DURATION_SECONDS 40
#define DEFAULT_CONNECTION_STALENESS_SECONDS 60
#define DEFAULT_STORAGE_SIZE (256 * 1024 * 1024)
#define DEFAULT_CREDENTIAL_ROTATION_SECONDS 2400
#define DEFAULT_CREDENTIAL_EXPIRATION_SECONDS 180
#define DEFAULT_AUDIO_TRACK_NAME "audio"
#define DEFAULT_AUDIO_CODEC_ID "A_FLAC"
#define DEFAULT_AUDIO_TRACKID 1
```

</div></details>

We tried to update the sdk version to 2.1.0, and then, we kept streaming for 8 hours in high latency network environment. (about 200kbps)
(after that, credential refreshing errors occured )

In sdk version 2.0.2, we can't keep streaming for 4 hours longer because of stream latency pressure, but sdk version 2.1.0 can streaming for 8 hours, even if high latency network.
If we can keep streaming for 8 hours, it's enough to meet our requirements at now.
And we want to see how it goes for a while at this version.
Though dropping frame error happens yet, we will try to catch error and retry to send our remaining buffer. 
Thank you.

sankyo-toshio, 2019-12-13:
@MushMal 
And I found when stream latency pressure occured, our CPU usage of 1core is over 120%.
I think this high CPU usage affects stream latency.

MushMal, 2019-12-15:
@sankyo-toshio the high load on the CPU must be due to the continuous retry logic. I believe you might be hitting an issue in the previous SDK which causing IoT or file based credentials not refresh properly causing buffer to fill up. Are you using the IoT or File credential provider?

sankyo-toshio, 2019-12-16:
@MushMal 
Yes, we use `IotCertCredentialProvider` for credential refresh.

HigashidaTomoyuki, 2019-12-23:
@MushMal 
Thank you very much. Since the question has been resolved, I will close it.

closed 2019-12-23 by HigashidaTomoyuki