awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #303: MIPS platform test sample program running error   (mipsel running little endian)

henryChmn, 2019-12-06:
We got error when run kvsVideoOnlyStreamingSample to test video stream,
the error message as below:,
[image]
please advise,
thanks.

MushMal, 2019-12-06:
@henryChmn you are attempting to initialize around 13.4MB or memory but this seems to fail. How much available RAM does your device have?

henryChmn, 2019-12-09:
@MushMal We can run into next state as below,
but we fail to send video to KVS and got failed  as below,
[image]

Ubuntu can send video stream to KVS:
[image]

thanks.

MushMal, 2019-12-09:
@henryChmn From the two images it states explicitly why it fails

1) It complains about the ca cert missing. Please follow the explicit Readme instructions on how to install the ca certificate. https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/blob/master/install-instructions-linux.md#install-the-certificate-in-the-operating-system-certificate-store
2) It fails with error 0x00000009 which is if you look it up - can't open a file. This is likely when you are trying to run the video only demo that tries to load the frame files from the disk and send. make sure you run it from the native-build directory.

Resolving the issue. Please feel free to reopen if you need more help

closed 2019-12-09 by MushMal