awslabs/amazon-kinesis-video-streams-producer-sdk-cpp #305: Already download the PEM file from SFSRootCAG2.pem to /etc/ssl/cert.pem but got the error

henryChmn, 2019-12-10:
we are not build KVS under native build but under kinesis-video-c-producer.
[image]
[image]
[image]

MathiasKoch, 2019-12-10:
I seem to be getting the a related (maybe the same) error.

I am trying to use the kvssink gstreamer element with the IotCredentialProvider, and i am getting invalid_ca_path error, even though the CA is good, and it used to work.

gst chain:
```
gst-launch-1.0 -vvv v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="streamName" iot-certificate="iot-certificate,endpoint=$AWS_ENDPOINT,cert-path=$CERTSDIR/cert.pem,key-path=$CERTSDIR/privkey.pem,ca-path=$CERTSDIR/root-CA.crt,role-aliases=$ROLE_ALIAS"
```

And i am getting `Failed to init kvs producer. Error: Unable to create Iot Credential provider. Error status: 0x15000020`

henryChmn, 2019-12-10:
We had successful upload video stream to KVS under MIPS platform,but can't play video stream in AWS with error message as below,
[image]

also attached log when test program process.
[teraterm_kvs.txt](https://github.com/awslabs/amazon-kinesis-video-streams-producer-sdk-cpp/files/3944227/teraterm_kvs.txt)

zhiyua-git, 2019-12-10:
Hi @henryChmn ,

You are sending data for only 10s, but you are checking video playback after ~1 minute, so your live video is gone. You can either try to start your playback in console first then send video or use the button in left bottom to search back and find the video. Please note any data out of your data retention (by default 2 hours) would have been gone.

[image]

zhiyua-git, 2019-12-10:
Hi @MathiasKoch ,

We didn't support crt file before, but you can try to remove the file name to just use directory path, this should help you on the invalid ca cert path issue.

gst-launch-1.0 -vvv v4l2src do-timestamp=TRUE device=/dev/video0 ! videoconvert ! video/x-raw,format=I420,width=640,height=480,framerate=30/1 ! omxh264enc periodicty-idr=45 inline-header=FALSE ! h264parse ! video/x-h264,stream-format=avc,alignment=au ! kvssink stream-name="streamName" iot-certificate="iot-certificate,endpoint=$AWS_ENDPOINT,cert-path=$CERTSDIR/cert.pem,key-path=$CERTSDIR/privkey.pem,ca-path=$CERTSDIR/~~root-CA.crt~~,role-aliases=$ROLE_ALIAS"

henryChmn, 2019-12-11:
Thanks for assist to build KVS under MIPS platform, we finally watch the real time playback video stream in AWS,

and will keep going to integrate cproducer into our application.
[image]

zhiyua-git, 2019-12-11:
Hi @henryChmn , 

Good to know it worked for you! Looking forward to see the actual video/audio data from camera.

MathiasKoch, 2019-12-11:
@zhiyua-git Thank you! Turns out i actually had a .pem certificate, just named with a .crt extension.. Works perfectly if i just change the extension.. Guess the new implementation actually check extension, before opening the file? This used to work fine with the .crt extension. 

Anyways, works now! Thanks.

zhiyua-git, 2019-12-11:
@MathiasKoch We add the extension check in recent release. Will continue to improve so we can support all these certificate types.

Closing issue as all are resolved.

closed 2019-12-11 by zhiyua-git